polhenarejos/pico-hsm #1: pico-hsm-patch-vidpid.sh does not work as expected

rrottmann, 2022-09-23:
Right now, the tool pico-hsm-patch-vidpid.sh cannot be used to patch vid/pid and produces unmodified output:

```bash
d=`mktemp -d`
cd $d
wget https://github.com/polhenarejos/pico-hsm/releases/download/v2.6/pico_hsm_waveshare_rp2040_zero-2.6.uf2
wget https://github.com/polhenarejos/pico-hsm/raw/master/tools/pico-hsm-patch-vidpid.sh
chmod +x pico-hsm-patch-vidpid.sh
perl --version
#(...)This is perl 5, version 36, subversion 0 (v5.36.0) built for x86_64-linux-thread-multi(...)
/pico-hsm-patch-vidpid.sh abcd:1234 pico_hsm_waveshare_rp2040_zero-2.6.uf2 pico_hsm_waveshare_rp2040_zero-2.6.uf2.patched 
#----------------------------
#VID/PID patcher for Pico HSM
#----------------------------
#
#Patching pico_hsm_waveshare_rp2040_zero-2.6.uf2... Done!
#
#Patched file was saved in pico_hsm_waveshare_rp2040_zero-2.6.uf2.patched
md5sum pico_hsm_waveshare_rp2040_zero-2.6.uf2*
#2a595f9a4aa893e5dc200429dcdc3564  pico_hsm_waveshare_rp2040_zero-2.6.uf2
#2a595f9a4aa893e5dc200429dcdc3564  pico_hsm_waveshare_rp2040_zero-2.6.uf2.patched
cd
rm -rf $d
```

polhenarejos, 2022-09-25:
I fix it in a7396dc698278d8ce3a5791739e3e9f190a15c32. Can you check it?

rrottmann, 2022-10-03:
Sorry for the delay. I was not in town. I've tested the fix with same commands as above with pico_hsm_pico-2.6.uf2 and pico_hsm_waveshare_rp2040_zero-2.6.uf2 and md5 checksum still shows no difference.

polhenarejos, 2022-10-09:
Can you try with the new release 3.0? It now works.

rrottmann, 2022-10-09:
Could confirm that it works now. Thanks a lot!

closed 2022-10-09 by rrottmann