polhenarejos/pico-hsm #4: CKR_USER_PIN_NOT_INITIALIZED  / failed with Incorrect parameters in APDU

rrottmann, 2022-10-08:
Thanks for the latest code changes. With them, I could build a recent version for RP2040 Waveshare Zero.
After nuking the flash I have installed the uf2 file. I build the uf2 using the latest code in repo using the following commands on Debian 11:

```bash
export PICO_SDK_PATH=~/Devel/pico/pico-sdk
export USB_VID=0x234b
export USB_PID=0x0000
export PICO_BOARD=waveshare_rp2040_zero 
cmake .. -DENABLE_DELAYED_BOOT=1 -DPICO_BOARD=$PICO_BOARD -DUSB_VID=${USB_VID} -DUSB_PID=${USB_PID}
make -j4
```
I also compiled latest git versions of opensc and sc-hsm-embedded.

After flashing, unplugging and inserting the RP2040 it gets recognized:
```
$ sudo dmesg
[ 2671.810277] usb 2-2: new full-speed USB device number 8 using ohci-pci
[ 2672.163479] usb 2-2: New USB device found, idVendor=234b, idProduct=0000, bcdDevice= 3.04
[ 2672.163482] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2672.163483] usb 2-2: Product: Pico HSM CCID
[ 2672.163484] usb 2-2: Manufacturer: Pol Henarejos
[ 2672.163485] usb 2-2: SerialNumber: E66138935F5B1A28
```

Initialization seems to somewhat work but shows error message `Incorrect parameters in APDU`:

```bash
$ sc-hsm-tool 
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E66138935F5B1A28) 00 00
Version              : 2.6
SmartCard-HSM has never been initialized. Please use --initialize to set SO-PIN and user PIN.
sc_card_ctl(*, SC_CARDCTL_SC_HSM_IMPORT_DKEK_SHARE, *) failed with Incorrect parameters in APDU
DKEK shares          : 0
DKEK key check value : 0000000000000000

$  sc-hsm-tool --initialize --so-pin 3537363231383830 --pin 648219
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E66138935F5B1A28) 00 00

$ sc-hsm-tool 
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E66138935F5B1A28) 00 00
Version              : 2.6
Config options       :
  User PIN reset with SO-PIN enabled
SO-PIN tries left    : 15
User PIN tries left  : 3
sc_card_ctl(*, SC_CARDCTL_SC_HSM_IMPORT_DKEK_SHARE, *) failed with Incorrect parameters in APDU
DKEK shares          : 0
DKEK key check value : 0000000000000000

$ pkcs11-tool --login --login-type so --so-pin 3537363231383830 --change-pin --new-pin 0123456789012345
Using slot 0 with a present token (0x0)
error: PKCS11 function C_SetPIN failed: rv = CKR_USER_PIN_NOT_INITIALIZED (0x102)
Aborting.

$ pkcs11-tool -l --pin 648219 --keypairgen --key-type rsa:2048 --id 1 --label "RSA2K"
Using slot 0 with a present token (0x0)
error: PKCS11 function C_Login failed: rv = CKR_USER_PIN_NOT_INITIALIZED (0x102)
Aborting.

$ sc-tool -l --pin 648219 --keypairgen --key-type rsa:2048 --id 1 --label "RSA2K"
No slot with a token was found.

$ sc-tool --login --login-type so --so-pin 3537363231383830 --change-pin --new-pin 0123456789012345
No slot with a token was found.

$ sc-tool -I
Cryptoki version 2.20
Manufacturer     CardContact (www.cardcontact.de)
Library          SmartCard-HSM via PC/SC (ver 2.12)
No slot with a token was found.
```

Similar experience also with tools from `opensc` Debian package but I wanted to test potential fixes from upstream git version / CardContact tools.

Am I doing something wrong (e.g. wrong VID/PID) or are there some differences in the Waveshare board? I also notice that the LED is not working. Or is this only happening when running user APDUs to enable user confirmation?

It would be good to know some details about OS/tools versions that you use in order to build/test on your devel machine.
I will also source some Pimoroni Tiny2040 to compare with the expected behaviour.

CardContact/Nitrokey HSM works fine:

```bash
$ sc-tool -I
Cryptoki version 2.20
Manufacturer     CardContact (www.cardcontact.de)
Library          SmartCard-HSM via PC/SC (ver 2.12)
Using slot 0 with a present token (0x1)

$ sc-hsm-tool 
Using reader with a card: Nitrokey Nitrokey HSM (DE00000000         ) 00 00
Version              : 3.5
Config options       :
  User PIN reset with SO-PIN enabled
SO-PIN tries left    : 15
User PIN tries left  : 3

$ sc-tool -l --pin ** --keypairgen --key-type rsa:2048 --id 1 --label "RSA2K"
Using slot 0 with a present token (0x1)
Key pair generated:
Private Key Object; RSA 
  label:      RSA2K
  ID:         01
  Usage:      decrypt, sign
  Access:     sensitive, always sensitive, never extractable, local
Public Key Object; RSA 2048 bits
  label:      RSA2K
  ID:         01
  Usage:      encrypt, verify
  Access:     local
```

rrottmann, 2022-10-08:
Also tried after nuke and reflashing:

```bash
$ apt-get install -y pyscard pycvc
$ ./pico-hsm-tool.py initialize --pin 648219 --so-pin 3537363231383830
Pico HSM Tool v1.2
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues
********************************
*   PLEASE READ IT CAREFULLY   *
********************************
This tool will erase and reset your device. It will delete all private and secret keys.

Are you sure?

[Press enter to confirm]

Public Point: 04be520925425af40c83275c8365205c74111df1ea24776809e2b61bd8174310ba9797bd380a89465f22c711b07884da3d2da8028e8afd1ba42f1186047e250e3e
Traceback (most recent call last):
  File "/home/me/git/pico-hsm/tools/./pico-hsm-tool.py", line 301, in <module>
    run()
  File "/home/me/git/pico-hsm/tools/./pico-hsm-tool.py", line 298, in run
    main(args)
  File "/home/me/git/pico-hsm/tools/./pico-hsm-tool.py", line 286, in main
    initialize(card, args)
  File "/home/me/git/pico-hsm/tools/./pico-hsm-tool.py", line 169, in initialize
    j = get_pki_data('cvc', data=data)
  File "/home/me/git/pico-hsm/tools/./pico-hsm-tool.py", line 105, in get_pki_data
    response = urllib.request.urlopen(req)
  File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.9/urllib/request.py", line 523, in open
    response = meth(req, response)
  File "/usr/lib/python3.9/urllib/request.py", line 632, in http_response
    response = self.parent.error(
  File "/usr/lib/python3.9/urllib/request.py", line 561, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 641, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
```

With help of `pdb` the following url was not found: 

```
(Pdb) p req.full_url
'https://www.henarejos.me/pico-hsm/cvc/'
```

```
 wget https://www.henarejos.me/pico-hsm/cvc/
--2022-10-08 23:39:41--  https://www.henarejos.me/pico-hsm/cvc/
Resolving www.henarejos.me... 172.67.197.212, 104.21.68.187, 2606:4700:3031::ac43:c5d4, ...
Connecting to www.henarejos.me|172.67.197.212|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-10-08 23:39:42 ERROR 404: Not Found.
```

polhenarejos, 2022-10-09:
These days I made some changes in the backend, which should be finished now, as they work in the v3.0.

However, I can confirm the PIN bug, which returns a `CKR_DATA_INVALID (0x20)` error.

Other errors, such generating keypair can be caused by bad initialization due to a missing device certificate, since it was not upload (the URL was down). Could you try with the v3.0?

rrottmann, 2022-10-09:
Retried with your build of v3.0. No need for personal build anymore as it works out of the box on Waveshare RP2040 Zero. Just patched the VID:PID and initialized with the python based pico-hsm-tool. Init and RSA key generation worked. Awesome!

Issue is solved!

I would suggest to give some hints of how to properly setup dependencies for Python. Had to mix debian packages swig, python3-pyscard, install python3-pip and pip install pycvc to get it working.

(typed on mobile)

closed 2022-10-09 by rrottmann

rrottmann, 2022-10-09:
Also PIN change worked. However at least my device needs to be frequently plugged in again. Maybe due to USB power saving on battery.