polhenarejos/pico-hsm #6: Backend for pico-hsm-tool.py get_pki_data() fails with HTTP 500

rrottmann, 2022-10-17:
Tried yesterday to initialize a Pimoroni RP2040 Tiny with version 3 firmware. Backend for retrieving certificate in function get_pki_data() throws a server side HTTP error 500.

rrottmann, 2022-10-17:
Maybe this is due to Cloudflare?

My request in curl with following-redirect:

```
curl https://www.henarejos.me/pico/pico-hsm/cvc \
-H "User-Agent: Mozilla/5.0 (Windows; U; Windows NT 5.1; en-US;" \
-d "pubkey=BNM_P26OqBtNsAs5D4PMp8kpMXyNJ0EBBknwwuFheBPiPfzUIIdcIB5csMWLRQd6AD9FXAPfENeZurHyb4NdVro"
{}
```

polhenarejos, 2022-10-19:
Still happening? I cannot reproduce it.

rrottmann, 2022-10-21:
Just tested it again and it still fails. This time with 404:

```
$ python3 -m pdb pico-hsm-tool.py  initialize --so-pin 3537363231383830 --pin 648219
> /home/user/git/pico-hsm/tools/pico-hsm-tool.py(3)<module>()
-> """
(Pdb) b 105
Breakpoint 1 at /home/rottmrei/git/pico-hsm/tools/pico-hsm-tool.py:105
(Pdb) c
Pico HSM Tool v1.2
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues
********************************
*   PLEASE READ IT CAREFULLY   *
********************************
This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
[Press enter to confirm]
Public Point: 04d33f3f6e8ea81b4db00b390f83cca7c929317c8d2741010649f0c2e1617813e23dfcd420875c201e5cb0c58b45077a003f455c03df10d799bab1f26f835d56ba
> /home/user/git/pico-hsm/tools/pico-hsm-tool.py(105)get_pki_data()
-> response = urllib.request.urlopen(req)
(Pdb) p req.data
b'pubkey=BNM_P26OqBtNsAs5D4PMp8kpMXyNJ0EBBknwwuFheBPiPfzUIIdcIB5csMWLRQd6AD9FXAPfENeZurHyb4NdVro%3D'
(Pdb) p req.full_url
'https://www.henarejos.me/pico-hsm/cvc/'
(Pdb) c
Traceback (most recent call last):
  File "/usr/lib/python3.9/pdb.py", line 1705, in main
    pdb._runscript(mainpyfile)
  File "/usr/lib/python3.9/pdb.py", line 1573, in _runscript
    self.run(statement)
  File "/usr/lib/python3.9/bdb.py", line 580, in run
    exec(cmd, globals, locals)
  File "<string>", line 1, in <module>
  File "/home/user/git/pico-hsm/tools/pico-hsm-tool.py", line 301, in <module>
    run()
  File "/home/user/git/pico-hsm/tools/pico-hsm-tool.py", line 298, in run
    main(args)
  File "/home/user/git/pico-hsm/tools/pico-hsm-tool.py", line 286, in main
    initialize(card, args)
  File "/home/user/git/pico-hsm/tools/pico-hsm-tool.py", line 169, in initialize
    j = get_pki_data('cvc', data=data)
  File "/home/user/git/pico-hsm/tools/pico-hsm-tool.py", line 105, in get_pki_data
    response = urllib.request.urlopen(req)
  File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.9/urllib/request.py", line 523, in open
    response = meth(req, response)
  File "/usr/lib/python3.9/urllib/request.py", line 632, in http_response
    response = self.parent.error(
  File "/usr/lib/python3.9/urllib/request.py", line 561, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 641, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
Uncaught exception. Entering post mortem debugging
Running 'cont' or 'step' will restart the program
> /usr/lib/python3.9/urllib/request.py(641)http_error_default()
-> raise HTTPError(req.full_url, code, msg, hdrs, fp)
(Pdb) p code
404
```

polhenarejos, 2022-10-28:
I could reproduce it and I think it is now fixed. It was a problem with the backend that ran out of memory. 

Could you try it again?

rrottmann, 2022-10-29:
I tested it again but I receive the same 401 error as in my previous [debug output](https://github.com/polhenarejos/pico-hsm/issues/6#issuecomment-1287086442).

rrottmann, 2022-10-30:
Retried just now with complete new flash including nuke, fresh download of release and patch of VID:PID. Worked as expected.

closed 2022-10-30 by rrottmann