polhenarejos/pico-hsm #8: Failing to detect pico hsm

jayeye, 2022-10-26:
Environment:
* Fully-patched Ubuntu 20.04 LTS
* `pico-hsm` repo at commit `4536589e2c121c4798c86c6e3721fbd1282ee88b`
* All submodules fetched
* `pico-sdk` repo at commit `2e6142b15b8a75c1227dd3edbe839193b2bf9041`
* `open-sc` at release 0.22.0, built from source

Raspberry Pico, Adafruit p/n 4864

When first connected, the pico showed up as a drive (as expected). I copied the firmware, and it the led started blinking, but `open-sc` fails to detect it:
 
```
$ sudo opensc-tool -an
No smart card readers found.
Failed to connect to reader: No readers found
```

FWIW, it happily detects the smartcard part of my Yubikey (I had unplugged it to get the previous message):

```
$ sudo opensc-tool -an
Using reader with a card: Yubico YubiKey OTP+FIDO+CCID 00 00
3b:fd:13:moredigits...
Personal Identity Verification Card
```

So I know that opensc works. What could be going wrong? 

Here is how the pico shows up:

In `dmesg`:
```
[1185128.712613] usb 1-5.3.4.1: USB disconnect, device number 76
[1185152.278614] usb 1-5.3.4.1: new full-speed USB device number 77 using xhci_hcd
[1185152.428962] usb 1-5.3.4.1: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 3.04
[1185152.428976] usb 1-5.3.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1185152.428983] usb 1-5.3.4.1: Product: Pico HSM CCID
[1185152.428987] usb 1-5.3.4.1: Manufacturer: Pol Henarejos
[1185152.428991] usb 1-5.3.4.1: SerialNumber: E660C06213179335
```
In the output of `lsusb -v`:
```
Bus 001 Device 078: ID feff:fcfd  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0xfeff 
  idProduct          0xfcfd 
  bcdDevice            3.04
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0056
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 
      ChipCard Interface Descriptor:
        bLength                54
        bDescriptorType        33
        bcdCCID              1.10  (Warning: Only accurate for version 1.0)
        nMaxSlotIndex           0
        bVoltageSupport         1  5.0V 
        dwProtocols             3  T=0 T=1
        dwDefaultClock       3580
        dwMaxiumumClock      3580
        bNumClockSupported      0
        dwDataRate           9600 bps
        dwMaxDataRate        9600 bps
        bNumDataRatesSupp.      0
        dwMaxIFSD             254
        dwSyncProtocols  00000000 
        dwMechanical     00000000 
        dwFeatures       00040840
          Auto parameter negotiation made by CCID
          Short and extended APDU level exchange
        dwMaxCCIDMsgLen     65554
        bClassGetResponse    echo
        bClassEnvelope       echo
        wlcdLayout           none
        bPINSupport             0 
        bMaxCCIDBusySlots       1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

rrottmann, 2022-10-26:
This is due to the VID:PID. It needs to be patched to a value that is supported and understood by ccid. You could use the Test VID:PID of Free Software Initiative of Japan as described in the Readme. Unfortunately, VID:PID cannot be easily distributed without acquiring an own one and the testing values have to be used by the user as part of the usage contract. See https://github.com/polhenarejos/pico-hsm#download for how to change the VID:PID.

jayeye, 2022-10-26:
> This is due to the VID:PID. It needs to be patched to a value that is supported and understood by ccid. You could use the Test VID:PID of Free Software Initiative of Japan as described in the [...]

Thanks, just saw this after I edited my original message.

closed 2022-10-26 by jayeye