polhenarejos/pico-hsm #9: Pico HSM 'hangs' on running "opensc-tool -an" or "pico-hsm-tool.py initialize"

1oh1, 2022-10-27:
Hi, I downloaded [pico_hsm_pico-3.0.uf2](https://github.com/polhenarejos/pico-hsm/releases/download/v3.0/pico_hsm_pico-3.0.uf2) and patched it to change the VID/PID to `234b:0000` using `pico-hsm-patch-vidpid.sh` and the SHA-256 hash of the patched UF2 file is `359545acf49c9145e8d8b43ec92023586e0c8b0d42fdf03103a9cce7748b3b31`

I tried running `opensc-tool -an` and the Pico outputs the following and 'hangs':
```
$ opensc-tool -an
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
```
If I unplug the Pico at this point, the next line is printed:
```
SmartCard-HSM
```
I also ran `python3 pico-hsm-tool.py initialize --so-pin 3537363231383830 --pin 648219` after plugging the Pico back in and I see this before it 'hangs':
```
Pico HSM Tool v1.4
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

********************************
*   PLEASE READ IT CAREFULLY   *
********************************

This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
[Press enter to confirm]
```
The LED on the Pico is blinking 4 times a second when it 'hangs'. Only once, I was able to get the `opensc-tool -an` command to finish execution (without plugging out the Pico) and the output looked like this:
```
$ opensc-tool -an
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
SmartCard-HSM version 3.0
```

Any idea what's wrong here?

I checked `pcscd` logs and saw these:
```
Oct 27 21:28:09 ubuntu-box systemd[1]: Stopped PC/SC Smart Card Daemon.
Oct 27 21:28:12 ubuntu-box systemd[1]: Started PC/SC Smart Card Daemon.
Oct 27 21:28:12 ubuntu-box pcscd[3641]: 00000000 ccid_usb.c:993:ReadUSB() read failed (1/21): LIBUSB_ERROR_OVERFLOW
Oct 27 21:28:12 ubuntu-box pcscd[3641]: 00101227 ccid_usb.c:993:ReadUSB() read failed (1/21): LIBUSB_ERROR_TIMEOUT
Oct 27 21:28:12 ubuntu-box pcscd[3641]: 00002466 ccid_usb.c:993:ReadUSB() read failed (1/21): LIBUSB_ERROR_OVERFLOW
Oct 27 21:28:12 ubuntu-box pcscd[3641]: 00000024 ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
Oct 27 21:28:12 ubuntu-box pcscd[3641]: 00000006 eventhandler.c:336:EHStatusHandlerThread() Error communicating to: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00
Oct 27 21:28:44 ubuntu-box systemd[1]: Stopping PC/SC Smart Card Daemon...
Oct 27 21:28:44 ubuntu-box systemd[1]: pcscd.service: Deactivated successfully.
Oct 27 21:28:44 ubuntu-box systemd[1]: Stopped PC/SC Smart Card Daemon.
Oct 27 21:28:47 ubuntu-box systemd[1]: Started PC/SC Smart Card Daemon.
Oct 27 21:29:55 ubuntu-box pcscd[3668]: 00000000 ccid_usb.c:993:ReadUSB() read failed (1/21): LIBUSB_ERROR_NO_DEVICE
Oct 27 21:29:55 ubuntu-box pcscd[3668]: 00000055 ifdwrapper.c:543:IFDTransmit() Card not transacted: 617
Oct 27 21:29:55 ubuntu-box pcscd[3668]: 00000054 ccid_usb.c:886:WriteUSB() write failed (1/21): LIBUSB_ERROR_NO_DEVICE
Oct 27 21:29:56 ubuntu-box pcscd[3668]: 01000724 winscard.c:1618:SCardTransmit() Card not transacted: 0x80100017
Oct 27 21:30:57 ubuntu-box systemd[1]: pcscd.service: Deactivated successfully.
```

BTW, unrelated:
[pico-hsm-tool.py#L29](https://github.com/polhenarejos/pico-hsm/blob/4536589e2c121c4798c86c6e3721fbd1282ee88b/tools/pico-hsm-tool.py#L29) is a bit misleading as I had to also run `pip install pycvc` before I could get `pico-hsm-tool.py` to work

polhenarejos, 2022-10-28:
Which board do you use?

It seems a problem with the firmware you downloaded, which does not correspond to the one for your board. 

Can you try it with the generic one? https://github.com/polhenarejos/pico-hsm/releases/download/v3.0/pico_hsm_pico-3.0.uf2

It should work in all boards with +2MB of flash.

1oh1, 2022-10-28:
> Which board do you use?

I use a generic Raspberry Pi Pico 2020 from the Raspberry Pi Foundation

> Can you try it with the generic one? https://github.com/polhenarejos/pico-hsm/releases/download/v3.0/pico_hsm_pico-3.0.uf2

I tried using the generic UF2 file you linked (SHA-256: C49CD7017B820448412637C6387775CD829D32E6E14065A0D1A1BCE108A32B8A) and added the VID/PID/Product name to `/usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist` and run into the same issue. `opensc-tool -an` worked once but on the next run, it 'hangs' again. Before I flashed it, I made sure to flash [pico_nuke_pico_generic-1.0.uf2](https://github.com/polhenarejos/pico-nuke/releases/download/v1.0/pico_nuke_pico_generic-1.0.uf2) to ensure the flash was wiped clean.

```
$ opensc-tool -an
Using reader with a card: Pico HSM [Pico HSM Interface] (E6609103C32B7A2A) 00 00
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
SmartCard-HSM version 3.0
$ opensc-tool -an
Using reader with a card: Pico HSM [Pico HSM Interface] (E6609103C32B7A2A) 00 00
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
```
I'm running this on Ubuntu 22.04.1 LTS. Could this have something to do with how I installed/set up opensc? Maybe I missed something?

polhenarejos, 2022-10-28:
I fixed a problem with our backend during the initialization (related with #6).

Can you try again with `python3 pico-hsm-tool.py initialize --so-pin 3537363231383830 --pin 648219`?
You should receive a success message. After this, the board is ready so please try it.

1oh1, 2022-10-28:
Thank you! I tried the initialization again and it seems to have succeeded but the `opensc-tool -an` command still fails after the initialization.
```
1oh1@ubuntu-box:~/hsm/pico-hsm/tools$ python3 pico-hsm-tool.py initialize --so-pin 3537363231383830 --pin 648219
Pico HSM Tool v1.4
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

********************************
*   PLEASE READ IT CAREFULLY   *
********************************

This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
[Press enter to confirm]
Public Point: 04ed536cce3e986b472e3b5c76a68401bb31fd1caff77259c28681b2207c51ac00bc9080fc8da0912b2509df2276357f36f856baf595dad12c7a1fe7a38a51248b
Device name: ESPICOHSMTR2R4QB
Certificate uploaded successfully!

Note that the device is initialized with a default PIN and configuration.
Now you can initialize the device as usual with your chosen PIN and configuration options.
1oh1@ubuntu-box:~/hsm/pico-hsm/tools$ opensc-tool -an
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
Segmentation fault (core dumped)
1oh1@ubuntu-box:~/hsm/pico-hsm/tools$ opensc-tool -an
Using reader with a card: Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
```

`journalctl -u pcscd` returns this:
```
Oct 28 12:16:27 ubuntu-box systemd[1]: pcscd.service: Deactivated successfully.
Oct 28 16:57:11 ubuntu-box systemd[1]: Started PC/SC Smart Card Daemon.
Oct 28 16:58:21 ubuntu-box pcscd[9225]: 00000000 ccid_usb.c:993:ReadUSB() read failed (1/9): LIBUSB_ERROR_NO_DEVICE
Oct 28 16:58:21 ubuntu-box pcscd[9225]: 00000061 ifdwrapper.c:543:IFDTransmit() Card not transacted: 617
Oct 28 16:58:21 ubuntu-box pcscd[9225]: 00000082 ccid_usb.c:886:WriteUSB() write failed (1/9): LIBUSB_ERROR_NO_DEVICE
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 01000699 winscard.c:1618:SCardTransmit() Card not transacted: 0x80100017
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00201903 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000752 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000573 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000465 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000646 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000214 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000261 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000185 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000175 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000177 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000176 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000192 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000180 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000177 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000183 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000200 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000190 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 16:58:22 ubuntu-box pcscd[9225]: 00000191 winscard.c:264:SCardConnect() Reader Free Software Initiative of Japan Gnuk [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
[... cut ...]
Oct 28 17:11:01 ubuntu-box pcscd[9624]: 00000252 winscard.c:264:SCardConnect() Reader Pico HSM [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 17:11:01 ubuntu-box pcscd[9624]: 00000250 winscard.c:264:SCardConnect() Reader Pico HSM [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 17:11:01 ubuntu-box pcscd[9624]: 00000249 winscard.c:264:SCardConnect() Reader Pico HSM [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
Oct 28 17:11:01 ubuntu-box pcscd[9624]: 00000265 winscard.c:264:SCardConnect() Reader Pico HSM [Pico HSM Interface] (E6609103C32B7A2A) 00 00 Not Found
```

polhenarejos, 2022-10-28:
Paste the output of `OPENSC_DEBUG=9 opensc-tool -an`

If it's too long use [github gists](https://gist.github.com/)

Also please try `OPENSC_DEBUG=9 opensc-tool -an -c sc-hsm`

1oh1, 2022-10-29:
Output of `OPENSC_DEBUG=9 opensc-tool -an`:
PASS: [opensc-tool-an-success.log](https://gist.github.com/1oh1/ce8aa856f9b518f9b17ef5894b26a74e)
FAIL: [opensc-tool-an-failure.log](https://gist.github.com/1oh1/4eb27c603de1b0b993f9eb9afffef0ea)

Output of `OPENSC_DEBUG=9 opensc-tool -an -c sc-hsm`:
PASS: [opensc-tool-an-c-sc-hsm-success.log](https://gist.github.com/1oh1/02ea79af33382e4ad75fe5b7623e1030)
FAIL: [opensc-tool-an-c-sc-hsm-failure.log](https://gist.github.com/1oh1/42fdcb1da3dcd2522f0b3fac8bf9fd66)

In the FAIL cases, I had to unplug the Pico for the command to exit.

polhenarejos, 2022-10-29:
Which steps do you do previously to PASS and FAIL? Are you able to reproduce it or occurs randomly?

When it fails, is still blinking?

It seems that there is something that crashes internally in the second cpu, which breaks the communication between the usb thread and apdu. The usb thread waits for a response and finally the client timeouts as no response is given due to the internal crash.

1oh1, 2022-11-01:
> Which steps do you do previously to PASS and FAIL? Are you able to reproduce it or occurs randomly?

After some trial and error, I figured out a way to consistently reproduce the PASS and FAIL cases.

For the **PASS** case, this needs to happen:
1. Stop pcscd with `sudo service pcscd stop`
2. Unplug and connect (or plug in if not already done) the Pico and ensure LED is blinking 4 times a second once it's plugged in
[image]
3. `OPENSC_DEBUG=9 opensc-tool -an -c sc-hsm` or `OPENSC_DEBUG=9 opensc-tool -an` will work **once** at this point
4. After the opensc command is executed and has exited, the Pico LED starts blinking at a rate of about once per second
[image]
5. Any opensc command run at this point will fail unless steps 1-2 is repeated

For the **FAIL** case, any time the Pico is blinking slowly (about once per second), and an opensc command is executed, it will get stuck and won't finish unless the Pico is plugged out. If the pcscd service is not stopped and the Pico is plugged back in, it will still be blinking slowly (about once per second) and the opensc command will get stuck again. When the command is stuck, the Pico will be blinking 4 times a second.

Only if the Pico LED is blinking 4 times a second when plugged in (which seems to happen only if the pcscd daemon is not running) will an opensc command run successfully and after that the Pico LED starts blinking at a rate of once per second. Any subsequent opensc commands executed at this point will fail again (unless I run `sudo service pcscd stop` and unplug and re-connect the Pico)

1oh1, 2022-11-03:
Interesting idea. Let me try this again on a fresh install of Ubuntu and report back.

rrottmann, 2022-11-04:
Is your Linux box headless or with GUI? I noticed that browsers like Firefox tend to poll pkcs11 devices and can block a HSM.

polhenarejos, 2022-11-28:
Maybe related with a bug detected in pico-openpgp.

Can you try with latest rev 7ca96178?

1oh1, 2022-12-02:
I'm sorry about not being able to look into this. I've been busy with work recently. I'll look into the new revision this weekend and get back to you.

Thank you so much!

1oh1, 2022-12-04:
I can confirm that the issue is fixed when built using https://github.com/polhenarejos/pico-hsm/commit/7ca96178fb11b56c2c76573f890faffd084c56a8. The Pico HSM does not become unresponsive anymore. Thank you!

1oh1, 2022-12-04:
> Is your Linux box headless or with GUI? I noticed that browsers like Firefox tend to poll pkcs11 devices and can block a HSM.

It is not headless but I didn't have any web browsers running when I ran into the issue.