polhenarejos/pico-hsm #11: ModuleNotFoundError: No module named 'secure_key'

rrottmann, 2022-11-03:
A search on pypy.org does not yield any results and also my google fu does not show meaningful results of how to obtain the library "secure_key". Is this an official python package? Would it be possible to populate a `requirements.txt` or point to the package name?

1oh1, 2022-11-03:
I searched on GitHub and I think it's maybe this file (for the macos platform):
https://github.com/polhenarejos/pico-fido/blob/43cd8869f9534c90ff284c39ce6a45d4f6ad990e/tools/secure_key/macos.py

So it seems to me that "secure_key" is probably not a third party library.

polhenarejos, 2022-11-03:
I forgot to add it. But note that it is still in progress. At this moment, only macOS is supported.

polhenarejos, 2022-11-03:
I added a quick fix to import secure_key only when needed to avoid the block of the whole tool.

closed 2022-11-03 by polhenarejos

polhenarejos, 2022-11-04:
The reason is that I could not make work the DBus approach for linux under headless session. See https://pypi.org/project/keyring/#using-keyring-on-headless-linux-systems

Even we could manage to make it work, I think the overhead is too much for that simple feature. I am looking for alternatives and probably windows, macOS and linux will have different approaches.