polhenarejos/pico-hsm #12: Are there plans to add support for ed25519?

symbiont-ji, 2022-11-05:
not all commercial HSMs support it, of course, but the yubikey hsm does.

rrottmann, 2022-11-06:
Pol Henarejos is working on [PR in Mbed-TLS](https://github.com/Mbed-TLS/mbedtls/pull/5800#issuecomment-1120770071) to support EdDSA which is the software implementation of the crypto primitives that are used in the tokens.

polhenarejos, 2022-11-06:
As @rrottmann said, we sent a PR to MbedTLS team, which is under review. Since it is a complex PR, it will require lot of effort and time to be merged. It could happen by Q4 22 but IMHO that will be scheduled by Q2 23, if not later. It depends on their review queue.

Though, if some specific feature (like this) is popular, they can decide to prioritize it.

symbiont-ji, 2022-11-06:
I first saw that the PR you referenced was closed, then saw the comment about it having been split into five! Is it in a state where another pair of eyes and hands would speed up the process, or am I more likely to be a hindrance if I get involved?

polhenarejos, 2022-11-07:
You can try it, but likely you will be invited to review other unrelated more urgent PR.

Ed25519 should be the easiest to review, as it does not depend on SHA3.

polhenarejos, 2024-06-26:
There's a branch with support for Ed25519.

closed 2024-06-26 by polhenarejos