polhenarejos/pico-hsm #15: Unable to initialize a new Pico-HSM

lindenaar, 2023-01-07:
Hi,

I am trying this very interesting project but when trying to initialize a new Pico-HSM and bump into 2 issues:

1. The `--pin` option for the `initialise` subcommand of  `python pico-hsm-tool.py` does not seem to work as documented
2. When executing the initialize command without a `--pin` (or when I put it before the `initialise` keyword) results in an issue:

```
tools % python pico-hsm-tool.py initialize --so-pin 3537363231383830
********************************
*   PLEASE READ IT CAREFULLY   *
********************************

This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
Pico HSM Tool v1.8
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

[Press enter to confirm]
Public Point: 04e5e6192c215f560abf97e84c80317ce4ef9f47a939783189e0b86bd54019ec069a6425a59e1251cf343596e98ebed10856a508347a995649aed671548f24c45a
Traceback (most recent call last):
  File "/Users/me/projects/pico-hsm/tools/pico-hsm-tool.py", line 638, in <module>
    run()
  File "/Users/me/projects/pico-hsm/tools/pico-hsm-tool.py", line 635, in run
    main(args)
  File "/Users/me/projects/pico-hsm/tools/pico-hsm-tool.py", line 616, in main
    initialize(card, args)
  File "/Users/me/projects/pico-hsm/tools/pico-hsm-tool.py", line 275, in initialize
    j = get_pki_data('cvc', data=data)
  File "/Users/me/projects/pico-hsm/tools/pico-hsm-tool.py", line 199, in get_pki_data
    response = urllib.request.urlopen(req)
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 500: Internal Server Error
```

It looks like the initiallization calls an external web service that doesn't function as expected. Any clue what is going wrong and would it be possible to explain this process (i.e. why is it using an external service and for what?) so that the dependency is clear?

any help is appreciated!

Thanks!

Frederik

0xClandestine, 2023-01-07:
I'm getting similar issues

lindenaar, 2023-01-10:
@polhenarejos thanks! it indeed works now again.

However, what is this web service and what does it do? Is there a way to run this yourself?

I also noticed that the --pin parameter is no longer supported when initializing the HSM, am I using the wrong version of the script or is the documentation not up-to-date?

thank you again for your work and support!

regards,

Frederik

polhenarejos, 2023-01-12:
Usually SmartCards have a pair of device key and certificate, signed by a PKI owned by the vendor. The private key and the certificate are generated during the manufacturing and are embedded into the ROM of the device to avoid accidental deletions. This private key is used for attestation of all keys generated by the device. 

This web service emulates all this process. During the initialization, a private key is generated and store securely in the Pico HSM (the device key). Then, the public key (the public point) is extracted from the device and uploaded to my PKI, which signs the public key and returns a certificate tied to the device public-private keypair.

In the future I am planing to open this process to accept other PKI. However, this is not straightforward as the certificate must be CV compliant (a standard defined by BSI from Germany).

The `--pin` parameter is supported, as usual. Try to put `--pin` parameter before the `initialize` command.

shaffiqfiq88, 2024-11-10:
Hi.

I am getting the same output of urllib.error.HTTPError: HTTP Error 500: Internal Server Error. Is the service down in the mean time for supporting Pico HSM 5.0?

Thanks for any feedback.

polhenarejos, 2024-11-10:
Pico-hsm-tool.py?

shaffiqfiq88, 2024-11-10:
> Pico-hsm-tool.py?

I just downloaded a new pico-hsm-tool.py and use this command that bypass the certificate request.

`python3 tools/pico-hsm-tool.py initialize --no-dev-cert`

I get this response

`Pico HSM Tool v2.0
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

********************************
*   PLEASE READ IT CAREFULLY   *
********************************

This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
[Press enter to confirm]
Note that the device is initialized with a default PIN and configuration.
Now you can initialize the device as usual with your chosen PIN and configuration options.`

Does it mean that the device is initialized and a key pairs is generated?

polhenarejos, 2024-11-10:
Yes, should work properly if VIDPID are recognized.