polhenarejos/pico-hsm #16: Usage in Python

detherminal, 2023-01-30:
Hi! I want to use your project in [mine](https://github.com/detherminal/seruro), can you please write documents about usage in Python? Thanks.

polhenarejos, 2023-01-30:
Pico HSM (and others my projects) are written in C++ to run as firmware in uf2/elf files. There is no support for python/circuitpython.

closed 2023-01-30 by polhenarejos

rrottmann, 2023-01-31:
I think the idea was to have some ready examples for interacting with [python-pkcs11](https://python-pkcs11.readthedocs.io/en/latest/) library or how to use python to send APDUs on a low level.

Gnuk uses a[ test framework](https://salsa.debian.org/gnuk-team/gnuk/gnuk/-/tree/master/tests) for validating the OpenPGP implementation which gives some details of how to interact with the card using Python. 

However, even pico-openpgp cannot be used with this test framework as the lowlevel python CCID/libusb part is not able to detect the pico card reader. The APDUs should at least work for pico-openpgp as it is part of the standard that is implemented.

polhenarejos, 2023-01-31:
I am not sure about that. He's requesting the python files to be run on a Pico via CircuitPython, in replacement of .c files.

In any case, you can always interface with pkcs11 library. Why do you say that pico-openpgp cannot be interfaced as the lowlevel python CCID/libusb is not able to detect the reader? Is it a VID/PID discovery problem? 

For pico-hsm I have recently created an emulation mode to run as an app on a computer. Combined with a virtualsmartcard it will allow to test pico-hsm without the need of having a physical Pico. Tests are written in Python and all emulation and tests are integrated into github's CI. It is similar to SoftHSMv2 but interfacing via CCID.

rrottmann, 2023-01-31:
> Is it a VID/PID discovery problem?

Most likely. I reverse-engineered the not so well documented dependencies for the Gnuk test repository and initiated the contained tests. They should just use APDUs according to the OpenPGP spec. But I could not get so far.

The `conftest.py` fixture prepares the environment for each test and it failed somewhere around [`get_ccid_device()`](https://salsa.debian.org/gnuk-team/gnuk/gnuk/-/blob/master/tests/conftest.py#L13). 

I deleted the VM where I tested but it was related to [`card_reader.py`](https://salsa.debian.org/gnuk-team/gnuk/gnuk/-
/blob/master/tests/card_reader.py#L322) and [`usb.core`](https://github.com/walac/pyusb/blob/master/usb/core.py?L1245#L1245). The iterator was empty.

I think it is not just the VID:PID (if I remember correclty, I even tested a few) but some USB/CCID quirks are done for specific card reader models.

polhenarejos, 2023-02-02:
Can you open an issue at /pico-openpgp? I'll address that test in this repo.