polhenarejos/pico-hsm #17: AES Encrypt / Decrypt

B00148917, 2023-01-30:
Morning,

Testing Pico HSM with a number of different RP2040's and it works very well.

I'm having a problem with AES encryption process

I've built my Ubuntu 22.04 using the following

sudo apt-get install libusb-dev libusb++ -y
sudo apt-get install libccid -y
sudo apt-get install pcscd -y
sudo apt-get install libpcsclite1 -y
sudo apt-get install libpcsclite-dev -y
sudo apt-get install libpcsc-perl -y
sudo apt-get install pcsc-tools -y
sudo apt-get update -y
sudo apt install opensc -y
sudo apt install pkgconf libssl-dev -y
sudo apt-get install autoconf -y
sudo apt install libtool-bin -y

sudo apt install pip -y
sudo apt install swig -y
pip install pyscard
pip install pycvc

git clone https://github.com/polhenarejos/pico-hsm.git
git clone https://github.com/OpenSC/OpenSC.git
git clone https://github.com/OpenSC/libp11.git
cd libp11
./bootstrap
./configure && make
make check
sudo make install
cd ..

git clone https://github.com/CardContact/sc-hsm-embedded.git
cd sc-hsm-embedded
autoreconf -fi
./configure && make
sudo make install
cd ..

alias sc-tool='pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so'

ls /usr/lib/x86_64-linux-gnu/engines-3/pkcs11.so
ls /usr/lib/x86_64-linux-gnu/opensc-pkcs11.so 

All works fine, but when I use the AES example given in the doc's 

echo "This is a text." | sc-tool -l --pin 123456 --encrypt --id 12 --mechanism aes-cbc > crypted.aes

I get 

pkcs11-tool: unrecognized option '--encrypt'

Am I using the module for pkcs11?

M

B00148917, 2023-01-30:
mark@mark-desktop:~/M1/MyHSM$ alias sc-tool
alias sc-tool='pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so'

mark@mark-desktop:~/M1/MyHSM$ sc-tool -I
Cryptoki version 2.20
Manufacturer     CardContact (www.cardcontact.de)
Library          SmartCard-HSM via PC/SC (ver 2.12)mark@mark-desktop:~/M1/MyHSM$ sc-tool -M
Using slot 0 with a present token (0x1)

mark@mark-desktop:~/M1/MyHSM$ sc-tool -M
Using slot 0 with a present token (0x1)
Supported mechanisms:
  RSA-X-509, keySize={1024,4096}, hw, encrypt, decrypt, sign, verify
  RSA-PKCS, keySize={1024,4096}, hw, encrypt, decrypt, sign, verify
  RSA-PKCS-PSS, keySize={1024,4096}, hw, sign, verify
  SHA1-RSA-PKCS, keySize={1024,4096}, hw, sign, verify
  SHA256-RSA-PKCS, keySize={1024,4096}, hw, sign, verify
  SHA1-RSA-PKCS-PSS, keySize={1024,4096}, hw, sign, verify
  SHA256-RSA-PKCS-PSS, keySize={1024,4096}, hw, sign, verify
  ECDSA, keySize={192,521}, hw, sign, verify
  ECDSA-SHA1, keySize={192,521}, hw, sign, verify
  AES-CBC, keySize={16,32}, hw, encrypt, decrypt
  AES-CMAC, keySize={16,32}, hw, sign
  RSA-PKCS-OAEP, keySize={1024,4096}, hw, encrypt, decrypt
  SHA-1, digest
  SHA224, digest
  SHA256, digest
  SHA384, digest
  SHA512, digest
  ECDSA-KEY-PAIR-GEN, keySize={192,521}, hw, generate_key_pair
  RSA-PKCS-KEY-PAIR-GEN, keySize={1024,4096}, hw, generate_key_pair
  AES-KEY-GEN, keySize={16,32}, hw, generate
  mechtype-0x80000001, keySize={1024,4096}, hw, sign, verify
  mechtype-0x80000003, keySize={1024,4096}, hw, sign, verify
  mechtype-0x80000010, keySize={192,521}, hw, sign, verify
  mechtype-0x80000011, keySize={192,521}, hw, sign, verify

B00148917, 2023-01-30:
mark@mark-desktop:~/M1/MyHSM$ sc-tool -I
Cryptoki version 2.20
Manufacturer     CardContact (www.cardcontact.de)
Library          SmartCard-HSM via PC/SC (ver 2.12)
Using slot 0 with a present token (0x1)

polhenarejos, 2023-01-30:
Did you generate the AES key successfully?
Which version of OpenSC?

B00148917, 2023-01-30:
sc-tool -l --pin 123456 --list-object --type secrkey
Using slot 0 with a present token (0x1)
Secret Key Object; AES length 32
  label:      AES32
  ID:         12
  Usage:      encrypt, decrypt
  Access:     sensitive, always sensitive, never extractable, local

opensc-tool --i
OpenSC 0.22.0 [gcc  11.2.0]
Enabled features: locking zlib readline openssl pcsc(libpcsclite.so.1)

polhenarejos, 2023-01-30:
Really strange. Seems an opensc error but I cannot find where it triggered. I see you install opensc via apt but later you clone it. Missing something?

B00148917, 2023-01-30:
Mistake on my part. Can rebuild and take the git clone command out.

B00148917, 2023-01-30:
Rebuilt clean Ubuntu 22.04 and ran

sudo apt install git

cd ~

mkdir M1

cd M1

sudo apt-get update -y
sudo apt-get install libusb-dev libusb++ -y
sudo apt-get install libccid -y
sudo apt-get install pcscd -y
sudo apt-get install libpcsclite1 -y
sudo apt-get install libpcsclite-dev -y
sudo apt-get install libpcsc-perl -y
sudo apt-get install pcsc-tools -y
sudo apt install opensc -y
sudo apt install pkgconf libssl-dev -y
sudo apt-get install autoconf -y
sudo apt install libtool-bin -y

sudo apt install pip -y
sudo apt install swig -y
pip install pyscard
pip install pycvc

git clone https://github.com/polhenarejos/pico-hsm.git
#git clone https://github.com/OpenSC/OpenSC.git
git clone https://github.com/OpenSC/libp11.git
cd libp11
./bootstrap
./configure && make
make check
sudo make install
cd ..

git clone https://github.com/CardContact/sc-hsm-embedded.git
cd sc-hsm-embedded
autoreconf -fi
./configure && make
sudo make install
cd ..

alias sc-tool='pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so'

ls /usr/lib/x86_64-linux-gnu/engines-3/pkcs11.so
ls /usr/lib/x86_64-linux-gnu/opensc-pkcs11.so

#sudo nano /etc/ssl/openssl.cnf....

#engines=engine_section
#[engine_section]
#pkcs11 = pkcs11_section
#[pkcs11_section]
#engine_id = pkcs11
#dynamic_path = /usr/lib/x86_64-linux-gnu/engines-3/pkcs11.so
#MODULE_PATH = /usr/lib/x86_64-linux-gnu/opensc-pkcs11.so
#init=0
#PIN=123456

B00148917, 2023-01-30:
Initialised the Pico HSM and can do everything as before but get the same error echo 

"This is a text." | sc-tool -l --encrypt --pin 123456 --id 12 --mechanism aes-cbc > crypted.aes

pkcs11-tool: unrecognized option '--encrypt'

polhenarejos, 2023-01-30:
Seems the package `opensc` in ubuntu's repos is outdated.

Can you clone opensc and build it? It should work.

B00148917, 2023-01-31:
Cloned and built OpenSC and it works.

Did see a warning 

error: PKCS11 function C_EncryptUpdate failed: rv = CKR_FUNCTION_NOT_SUPPORTED (0x54)

using 

echo "This is a complex text." | pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --encrypt --pin 123456 --id 12 --mechanism aes-cbc > crypted.aes

B00148917, 2023-01-31:
Thank you for your help

B00148917, 2023-01-31:
Sorry, wrong message.

warning: PKCS11 function C_GetAttributeValue(ALWAYS_AUTHENTICATE) failed: rv = CKR_ATTRIBUTE_TYPE_INVALID (0x12)

closed 2023-02-02 by B00148917