polhenarejos/pico-hsm #23: EC Key Creation

B00148917, 2023-03-29:
Morning 

Different outcome based on the same key type?

      clear
      
      sudo service pcscd start
      
      ## Initialization
      echo "Initialize the HSM."
      
      time python3 ~/M1/pico-hsm/tools/pico-hsm-tool.py initialize --so-pin 3537363231383830 
      
      echo "Change the PIN"
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --login --pin 648219 --change-pin --new-pin 123456
      
      # DSA - Test # 17 - DSA Key Gen
      
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:secp192r1 --id 11 --label "DSA192"
      
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:secp256r1 --id 12 --label "DSA256"
      
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:secp384r1 --id 13 --label "DSA384"
      
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:secp521r1 --id 14 --label "DSA521"
      
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:secp192k1 --id 15 --label "DSA192K"
      
      time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:secp256k1 --id 16 --label "DSA256K"

All works as expected.

      for i in `seq 11 16`
      do
          echo ""
          echo ""
          echo -e "\e[0;31mCreation \e[0m of key $i"
          time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin 123456 --id $i --type pubkey > $i.der
          time openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub
          echo ""
       
          echo ""
          echo -e "\e[0;31mSigning \e[0m using key $i"
          time pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin 123456 --mechanism ECDSA -i data.file -o $i.sig --signature-format openssl
          echo ""
          echo -e "\e[0;32mVerifing \e[0m using key $i"
          time openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig
      done

All of the odd numbers done work? But the odd numbers don't with an EVP_PKEY issue?

Creation  of key 11
Using slot 0 with a present token (0x1)
error: cannot create EVP_PKEY
Aborting.

real    0m0.446s
user    0m0.004s
sys     0m0.005s
read EC key
Could not read public key from 11.der
unable to load Key

real    0m0.006s
user    0m0.006s
sys     0m0.000s

Signing  using key 11
Using slot 0 with a present token (0x1)
Using signature algorithm ECDSA

real    0m0.160s
user    0m0.000s
sys     0m0.006s

Verifing  using key 11
Could not read public key from 11.pub
pkeyutl: Error initializing context

real    0m0.006s
user    0m0.006s
sys     0m0.000s

Creation  of key 12
Using slot 0 with a present token (0x1)

real    0m0.191s
user    0m0.004s
sys     0m0.008s
read EC key
writing EC key

real    0m0.007s
user    0m0.007s
sys     0m0.000s

Signing  using key 12
Using slot 0 with a present token (0x1)
Using signature algorithm ECDSA

real    0m0.296s
user    0m0.007s
sys     0m0.000s

Verifing  using key 12
Signature Verified Successfully

real    0m0.007s
user    0m0.001s
sys     0m0.007s

Creation  of key 13
Using slot 0 with a present token (0x1)
error: cannot create EVP_PKEY
Aborting.

real    0m0.191s
user    0m0.006s
sys     0m0.005s
read EC key
Could not read public key from 13.der
unable to load Key

real    0m0.006s
user    0m0.006s
sys     0m0.001s

Signing  using key 13
Using slot 0 with a present token (0x1)
Using signature algorithm ECDSA

real    0m0.312s
user    0m0.000s
sys     0m0.006s

Verifing  using key 13
Could not read public key from 13.pub
pkeyutl: Error initializing context

real    0m0.005s
user    0m0.005s
sys     0m0.001s

Creation  of key 14
Using slot 0 with a present token (0x1)

real    0m0.192s
user    0m0.004s
sys     0m0.008s
read EC key
writing EC key

real    0m0.008s
user    0m0.000s
sys     0m0.008s

Signing  using key 14
Using slot 0 with a present token (0x1)
Using signature algorithm ECDSA

real    0m0.552s
user    0m0.000s
sys     0m0.007s

Verifing  using key 14
Signature Verified Successfully

real    0m0.009s
user    0m0.004s
sys     0m0.005s

Creation  of key 15
Using slot 0 with a present token (0x1)
error: cannot create EVP_PKEY
Aborting.

real    0m0.189s
user    0m0.003s
sys     0m0.007s
read EC key
Could not read public key from 15.der
unable to load Key

real    0m0.006s
user    0m0.005s
sys     0m0.001s

Signing  using key 15
Using slot 0 with a present token (0x1)
Using signature algorithm ECDSA

real    0m0.173s
user    0m0.000s
sys     0m0.006s

Verifing  using key 15
Could not read public key from 15.pub
pkeyutl: Error initializing context

real    0m0.006s
user    0m0.006s
sys     0m0.001s

Creation  of key 16
Using slot 0 with a present token (0x1)

real    0m0.198s
user    0m0.008s
sys     0m0.004s
read EC key
writing EC key

real    0m0.008s
user    0m0.007s
sys     0m0.001s

Signing  using key 16
Using slot 0 with a present token (0x1)
Using signature algorithm ECDSA

real    0m0.305s
user    0m0.006s
sys     0m0.001s

Verifing  using key 16
Signature Verified Successfully

real    0m0.009s
user    0m0.008s
sys     0m0.001s

Any suggestions?

Br,

markone [b00148917]

polhenarejos, 2023-03-31:
Really strange. I cannot reproduce it.
```
Creation  of key 11
Using slot 2 with a present token (0x9)
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin    0,00s user 0,01s system 0% cpu 2,151 total
read EC key
writing EC key
openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub  0,01s user 0,01s system 66% cpu 0,022 total

Signing  using key 11
Using slot 2 with a present token (0x9)
Using signature algorithm ECDSA
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin   0,01s user 0,01s system 0% cpu 2,489 total

Verifing  using key 11
Signature Verified Successfully
openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig  0,01s user 0,00s system 75% cpu 0,013 total

Creation  of key 12
Using slot 2 with a present token (0x9)
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin    0,01s user 0,01s system 0% cpu 2,329 total
read EC key
writing EC key
openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub  0,00s user 0,00s system 66% cpu 0,009 total

Signing  using key 12
Using slot 2 with a present token (0x9)
Using signature algorithm ECDSA
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin   0,00s user 0,01s system 0% cpu 2,572 total

Verifing  using key 12
Signature Verified Successfully
openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig  0,01s user 0,01s system 76% cpu 0,018 total

Creation  of key 13
Using slot 2 with a present token (0x9)
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin    0,01s user 0,01s system 0% cpu 2,339 total
read EC key
writing EC key
openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub  0,00s user 0,00s system 76% cpu 0,008 total

Signing  using key 13
Using slot 2 with a present token (0x9)
Using signature algorithm ECDSA
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin   0,00s user 0,01s system 0% cpu 2,824 total

Verifing  using key 13
Signature Verified Successfully
openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig  0,00s user 0,00s system 71% cpu 0,009 total

Creation  of key 14
Using slot 2 with a present token (0x9)
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin    0,00s user 0,01s system 0% cpu 2,318 total
read EC key
writing EC key
openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub  0,00s user 0,00s system 68% cpu 0,009 total

Signing  using key 14
Using slot 2 with a present token (0x9)
Using signature algorithm ECDSA
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin   0,00s user 0,00s system 0% cpu 3,076 total

Verifing  using key 14
Signature Verified Successfully
openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig  0,00s user 0,00s system 67% cpu 0,010 total

Creation  of key 15
Using slot 2 with a present token (0x9)
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin    0,00s user 0,00s system 0% cpu 2,316 total
read EC key
writing EC key
openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub  0,00s user 0,00s system 69% cpu 0,009 total

Signing  using key 15
Using slot 2 with a present token (0x9)
Using signature algorithm ECDSA
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin   0,00s user 0,00s system 0% cpu 2,509 total

Verifing  using key 15
Signature Verified Successfully
openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig  0,00s user 0,00s system 73% cpu 0,012 total

Creation  of key 16
Using slot 2 with a present token (0x9)
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin    0,00s user 0,00s system 0% cpu 2,315 total
read EC key
writing EC key
openssl ec -inform DER -outform PEM -in $i.der -pubin > $i.pub  0,00s user 0,00s system 84% cpu 0,010 total

Signing  using key 16
Using slot 2 with a present token (0x9)
Using signature algorithm ECDSA
pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --id $i --sign --pin   0,00s user 0,01s system 0% cpu 2,587 total

Verifing  using key 16
Signature Verified Successfully
openssl pkeyutl -verify -pubin -inkey $i.pub -in data.file -sigfile $i.sig  0,00s user 0,00s system 67% cpu 0,009 total
```

- Which version of openssl do you use?
- Which version of pkcs11 module do you use?

Assuming key 11 fails, can you paste the output of this command?
```
OPENSC_DEBUG=9 pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --read-object --pin 123456 --id 11 --type pubkey > 11.der
```

B00148917, 2023-04-05:
Apologies for the delay. Off for a few days.

OpenSSL 3.0.2 15 Mar 2022 (Library: OpenSSL 3.0.2 15 Mar 2022)

pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -I
Cryptoki version 2.20
Manufacturer     CardContact (www.cardcontact.de)
Library          SmartCard-HSM via PC/SC (ver 2.12)
Using slot 0 with a present token (0x1)

The issue is key dependant 

I tested using the code below and see that 192r1, 384r1, bp384 and 192k1 EC keys works?

---

#!/bin/bash

#Test to see if you can reach the card reader

KeyTypes="secp192r1 secp256r1 secp384r1 secp521r1 brainpoolP256r1 brainpoolP384r1 brainpoolP512r1 secp192k1 secp256k1"

for kt in $KeyTypes; do

    OPENSC_DEBUG=9
    
    python3 ~/M1/pico-hsm/tools/pico-hsm-tool.py initialize --so-pin 3537363231383830 
    pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so --login --pin 648219 --change-pin --new-pin 123456

    pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --pin 123456 --keypairgen --key-type EC:$kt --id 16 --label "MyECKey"

    pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --read-object --pin 123456 --id 16 --type pubkey > ec16.der

    openssl ec -inform DER -outform PEM -in ec16.der -pubin > ec16.pub

    echo "This is a test string. Be safe, be secure." > data

    openssl dgst -sha1 -binary -out data.sha1 data

    pkcs11-tool --module /usr/local/lib/libsc-hsm-pkcs11.so -l --id 16 --sign --pin 123456 --mechanism ECDSA -i data.sha1 -o data.sig --signature-format openssl

    openssl pkeyutl -verify -pubin -inkey ec16.pub -in data.sha1 -sigfile data.sig

    echo $kt

done

B00148917, 2023-04-05:
Sorry don't work...all the others work fine?

polhenarejos, 2024-06-26:
No activity. Feel free to reopen if still persists.