polhenarejos/pico-hsm #24: urllib.error.HTTPError: HTTP Error 500: Internal Server Error in "pico-hsm-tool.py" initialize command

Aadinthosh, 2023-04-10:
When executing the python command -  "python3 pico-hsm-tool.py initialize --so-pin 3537363231383830 "
I am facing an exception "urllib.error.HTTPError: HTTP Error 500: Internal Server"
PS D:\PICO\pico-hsm\tools> python3 pico-hsm-tool.py initialize --so-pin 3537363231383830

********************************
*   PLEASE READ IT CAREFULLY   *
********************************

This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
Pico HSM Tool v1.8
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

[Press enter to confirm]
Public Point: 0499220c3a32eaed086a775774cb4fde1942b4fc15118ae4470d60af41f72de24f57d405e9c0fbaffd156ab84bc6c602e4121db4ac934364a080ad5fa769468a7e
Traceback (most recent call last):
  File "D:\PICO\pico-hsm\tools\pico-hsm-tool.py", line 640, in <module>
    run()
  File "D:\PICO\pico-hsm\tools\pico-hsm-tool.py", line 637, in run
    main(args)
  File "D:\PICO\pico-hsm\tools\pico-hsm-tool.py", line 618, in main
    initialize(card, args)
  File "D:\PICO\pico-hsm\tools\pico-hsm-tool.py", line 277, in initialize
    j = get_pki_data('cvc', data=data)
  File "D:\PICO\pico-hsm\tools\pico-hsm-tool.py", line 199, in get_pki_data
    response = urllib.request.urlopen(req)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 525, in open
    response = meth(req, response)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 634, in http_response
    response = self.parent.error(
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 563, in error
    return self._call_chain(*args)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 496, in _call_chain
    result = func(*args)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 500: Internal Server Error

polhenarejos, 2023-04-27:
Does it happen yet?

Aadinthosh, 2023-04-27:
Yeah still I'm facing the same issue ...

Aadinthosh, 2023-04-27:
[image]

polhenarejos, 2023-04-27:
Fixed at the backend. Reopen if persists.

closed 2023-04-27 by polhenarejos