polhenarejos/pico-hsm #26: Strange behaviours betwwen Windows 10 & 11

fchaxel, 2023-06-22:
Hi Pol,

With the same pico device I have two radically different behaviours between Windows 10 and 11, with the same open-cs version.

`opensc-tool --version
OpenSC-0.23.0, rev: 5497519e, commit-time: 2022-11-29 09:34:43 +0100`

On Windows 11
`opensc-tool -na
Using reader with a card: Broadcom Corp Contacted SmartCard 0
Card not present.
Failed to connect to reader: Card not present`

On Windows 10
`opensc-tool -na
Using reader with a card: Pol Henarejos Pico Key 0
3b:fe:18:00:00:81:31:fe:45:80:31:81:54:48:53:4d:31:73:80:21:40:81:07:fa
SmartCard-HSM version 3.4`

... one can see version 3.4 using the uf2 firmware file named pico_hsm_pico-3.2.uf2.

 pkcs15-tool is OK on Windows 10 but give also a card not dectected message on Windows 11

`pkcs15-tool.exe -D
Using reader with a card: Broadcom Corp Contacted SmartCard 0
Card not present.`

But on both operating system using XCA with the dll (64bits) from CardContact I can create and use keys (not the dll coming from open/SC setup where the pin code is not requested when working with the card ... so fail bad PIN.

Bye.

polhenarejos, 2023-06-22:
Could you paste what `opensc-tool -l` displays in Windows 11?
It seems there is another reader attached to computer.

fchaxel, 2023-06-22:
Once again two differents behaviours.

On Windows 10

Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Pol Henarejos Pico Key 0

and on Windows 11

Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Broadcom Corp Contacted SmartCard 0
1    Yes             Pol Henarejos Pico Key 0

polhenarejos, 2023-06-22:
You have another reader.

Try to select reader `1` with `-r` flag: `opensc-tool -r 1 -an`

fchaxel, 2023-06-22:
Your right on my Windows 11 PC without any kind of smart card reader I can see the Broadcom Corp Contacted SmartCard.

So sorry for the spend time.

closed 2023-08-16 by polhenarejos