polhenarejos/pico-hsm #34: waveshare zero build fails.

nimbius, 2024-04-03:
after copying file for waveshare zero, the device blinks red.

polhenarejos, 2024-04-03:
The color is not important, but the blink pattern.

What `opensc-tool -an` reports?

nimbius, 2024-04-04:
debug output as requested:

sudo opensc-tool -an
No smart card readers found.
Failed to connect to reader: No readers found

device blinks steady red, every .5sec.

nimbius, 2024-04-07:
my apologies for not including this earlier.  version 3.6 specifically fails to launch on the waveshare zero.  version 3.0 of the code loads OK

polhenarejos, 2024-04-10:
Can you include any log from dmesg to see pcscd messages?

dev-area, 2024-04-10:
im working with version 3.6 on waveshare zero and its working ok 
red led blinking but working ok
BTW if i load the simple pico build to the board its also working ok with no blinking

nimbius, 2024-05-12:
interesting! ill test now

nimbius, 2024-05-12:
still failing 3.6.  opensc-tool doesnt see the device :\

dmesg output as follows after loading the pico-patched (nitrokey) 

[ 4304.635423] usb 1-6: new full-speed USB device number 22 using xhci_hcd
[ 4304.945226] usb 1-6: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[ 4304.957210] usb 1-6: New USB device found, idVendor=20a0, idProduct=4230, bcdDevice= 5.00
[ 4304.957216] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4304.957220] usb 1-6: Product: Pico Key
[ 4304.957223] usb 1-6: Manufacturer: Pol Henarejos
[ 4304.957226] usb 1-6: SerialNumber: E6625887D3345230
[ 4333.662700] usb 1-6: USB disconnect, device number 22
[ 4336.595592] usb 1-6: new full-speed USB device number 23 using xhci_hcd
[ 4336.905270] usb 1-6: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[ 4336.916851] usb 1-6: New USB device found, idVendor=20a0, idProduct=4230, bcdDevice= 5.00
[ 4336.916864] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4336.916870] usb 1-6: Product: Pico Key
[ 4336.916875] usb 1-6: Manufacturer: Pol Henarejos
[ 4336.916880] usb 1-6: SerialNumber: E6625887D3345230

nimbius, 2024-05-13:
dmesg output from the 3.0 build 
[ 1998.466955] usb 1-5: USB disconnect, device number 10
[ 2007.277538] usb 1-6: USB disconnect, device number 9
[ 2009.010705] usb 1-6: new full-speed USB device number 11 using xhci_hcd
[ 2009.334184] usb 1-6: New USB device found, idVendor=20a0, idProduct=4230, bcdDevice= 3.04
[ 2009.334199] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2009.334205] usb 1-6: Product: Pico HSM CCID
[ 2009.334210] usb 1-6: Manufacturer: Pol Henarejos
[ 2009.334215] usb 1-6: SerialNumber: E6625887D3345230

curiously enough, opensc-tool -an does not see a card reader here either.

polhenarejos, 2024-05-30:
I use earlier versions from opensc. 0.22 or 0.23 should work fine. Can you test it?

Does `pcsctest` provide useful info?

nimbius, 2024-05-31:
not sure i have pcsctest available (my OS is arch)

is there an OS i can try with?  I have a homelab, and the ability to present the dev environment to you remotely if you like.

polhenarejos, 2024-06-26:
v4.0 should work with OpenSC 0.25.  Feel free to reopen if still persists.

closed 2024-06-26 by polhenarejos

abu-matterize, 2024-11-13:
I've similar issues with Arch. My hardware is RPI Pico 2. Flashed with pico-hsm from https://www.picokeys.com/getting-started/

`dmesg` output
```sh
[  179.413262] usb 3-2: New USB device found, idVendor=1050, idProduct=0030, bcdDevice= 7.00
[  179.413279] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  179.413286] usb 3-2: Product: Pico Key
[  179.413291] usb 3-2: Manufacturer: Pol Henarejos
[  179.413296] usb 3-2: SerialNumber: 8D68603BBDE18CBE
[  190.054006] usb 3-2: USB disconnect, device number 4
[  193.884003] usb 3-2: new full-speed USB device number 5 using xhci_hcd
[  194.040743] usb 3-2: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[  194.052907] usb 3-2: New USB device found, idVendor=1050, idProduct=0030, bcdDevice= 7.00
[  194.052924] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  194.052930] usb 3-2: Product: Pico Key
[  194.052936] usb 3-2: Manufacturer: Pol Henarejos
[  194.052941] usb 3-2: SerialNumber: 8D68603BBDE18CBE
```
When I run `opensc-tool -v -sc`
```sh
[host user]~ opensc-tool -v -sc
No smart card readers found.
Failed to connect to reader: No readers found