polhenarejos/pico-hsm #41: Unable to restore PicoHSM Private key (Issue importing DKEK?)

lindenaar, 2024-06-16:
Hi,

I am following the instructions for [Backup and restore](https://github.com/polhenarejos/pico-hsm/blob/master/doc/backup-and-restore.md) to setup my Pico HSM with a known DKEK so that it is possible to backup/restore the keys in it. So far all steps seem successful except that I am unable to restore (unwrap) a backed up key. 

I suspect there is an issue with the import of the DKEK as when importing the DKEK with (sensitive data is stored in environment variables):

```
sc-hsm-tool --initialize --so-pin env:HSM_SOPIN  --pin env:HSM_PIN --dkek-shares 1
sc-hsm-tool --import-dkek-share dkek.pbe--password env:DKEK_PASSWORD
```

 I always get output like:

```
Using reader with a card: Pol Henarejos Pico Key CCID Interface
Deciphering DKEK share, please wait...
DKEK share imported
DKEK shares          : 1
DKEK key check value : 0000000000000000
```

This also happens when I initialise it with multiple DKEK files, the check value being `0000000000000000` does not seem to be correct (is at least not what I would expect). However, this is always the case with every `dkek.pbe` file I generate (and these files are not the same)

I have wrapped the a 2048 bit RSA key (tried with and without loadin the cert) with:

```
sc-hsm-tool --pin env:HSM_PIN --wrap-key hsm-wrapped-key.bin--key-reference 1
```

but when I try to load is back after I have reinitialised the PicoHSM) with:

```
sc-hsm-tool --pin env:HSM_PIN --unwrap-key hsm-wrapped-key.bin --key-reference 1 --force
```

I get the following output:
```
Using reader with a card: Pol Henarejos Pico Key CCID Interface
Wrapped key contains:
  Key blob
  Private Key Description (PRKD)
  Certificate
sc_card_ctl(*, SC_CARDCTL_SC_HSM_UNWRAP_KEY, *) failed with Reference data not usable
```

The device I am using is a WaveShare Pico RP2040 Zero and us  image `pico_hsm_waveshare_rp2040_zero-3.6.gnuk.uf2`.

I also tried to use `pico_hsm_waveshare_rp2040_zero-3.4.gnuk.uf2` but tht could not load a DKEK at all with my board

Is there something I am missing in my steps or is this a (known) issue? Any help or hints would be appreciated.

regards,

Frederik

lindenaar, 2024-06-16:
Dear @polhenarejos ,

Sure, though at this moment 3.6 seems to be the latest release unless you mean switching to the development branch and compule the image myself ?

regards,

Frederik

polhenarejos, 2024-06-17:
Sure, not a problem.

A `DKEK key check value : 0000000000000000` means that you have not been logged in. It is totally normal.

I cannot reproduce your problem. I use the following commands:
```
sc-hsm-tool --initialize --so-pin env:HSM_SOPIN  --pin env:HSM_PIN --dkek-shares 1
sc-hsm-tool --pin env:HSM_PIN --import-dkek-share dkek.pbe
pkcs11-tool -l --pin env:HSM_PIN --keypairgen --key-type ec:secp256r1 --id 1 --label "SECP256R1"
sc-hsm-tool --pin env:HSM_PIN --wrap-key hsm-wrapped-key.bin --key-reference 1
sc-hsm-tool --pin env:HSM_PIN --unwrap-key hsm-wrapped-key.bin --key-reference 2
```

Can you try them? Perhaps is a problem with RSA.

lindenaar, 2024-06-18:
I tried with the `ec:secp256r1` key as suggested though that gave the same result unfortunately:

```
Using reader with a card: Pol Henarejos Pico Key CCID Interface
Wrapped key contains:
  Key blob
  Private Key Description (PRKD)
  Certificate
sc_card_ctl(*, SC_CARDCTL_SC_HSM_UNWRAP_KEY, *) failed with Reference data not usable
```

would it make sense to build an image myself or perhaps use another board (e.g. a Raspberry Pi Pico - not sure if that could make a differentce) ?

polhenarejos, 2024-06-19:
Can you execute this code and upload `output.log`?
```
export OPENSC_DEBUG=9
{ 
sc-hsm-tool --initialize --so-pin env:HSM_SOPIN  --pin env:HSM_PIN --dkek-shares 1
sc-hsm-tool --pin env:HSM_PIN --import-dkek-share dkek.pbe
pkcs11-tool -l --pin env:HSM_PIN --keypairgen --key-type ec:secp256r1 --id 1 --label "SECP256R1"
sc-hsm-tool --pin env:HSM_PIN --wrap-key hsm-wrapped-key.bin --key-reference 1
sc-hsm-tool --pin env:HSM_PIN --unwrap-key hsm-wrapped-key.bin --key-reference 2
} > output.log 2>&1 
unset OPENSC_DEBUG
```

lindenaar, 2024-06-20:
Sure, please [output.log](https://github.com/user-attachments/files/15919191/output.log) attached

polhenarejos, 2024-06-20:
I see:
`Enter password to decrypt DKEK share : ./init_hsm.sh: line 114: 78462 Segmentation fault: 11  sc-hsm-tool --pin env:HSM_PIN --import-dkek-share dkek.pbe
`

which seems broken.

I see you are using OpenSC 0.25 and I remember it has some troubles. Can you try with OpenSC v0.22 or 0.23? These are the ones tested.

lindenaar, 2024-06-21:
Hi @polhenarejos ,

Thanks for diving into this and pointing this out. I can confirm that installing OpenSC 0.23.0 indeed resolves the problem.

Perhaps good to add this to the documentation that it is required to not use a higher version? I installed OpenSC using HomeBrew (brew.sh) and didn't pay attention to the version. It turns out that only version OpenSC 0.25 is available through that channel so I had to install the official OpenSC package 0.23.0 from their release and that worked.

I have closed this issue, thanks again for your help!

closed 2024-06-21 by lindenaar

polhenarejos, 2024-06-24:
Version 4.0 adds support for OpenSC 0.25.1 and should fix key import.

lindenaar, 2024-06-27:
@polhenarejos thanks for letting me know, I can confirm this works now both with OpenSC 0.25.1 as well as 0.23.0

commit d708158f (2024-06-22): Fix key unwrap with latest OpenSC. Fixes #41.

SC-HSM driver in OpenSC has changed the procedure for unwraping. It stores the wrap into a temporary file (0x2F10) and then it calls unwrap cmd.