polhenarejos/pico-hsm #44: USB not recognized

dev-area, 2024-06-25:
hi
Iv updated the code to release 4.0 , build it ok without errors (on debian running inside VMware)
but after flashing the firmware  it is no more recognized - im getting an error on windows (usb not recognized) and also cant mount it on linux
i also tried with a prebuild uf2 file from the repo here and the same problem
test it on pico simple board and also on seeed xiao

polhenarejos, 2024-06-25:
Which VID & PID?

dev-area, 2024-06-25:
Tried the one in the example 0x1234:0x5678
And also tried cardcontact hsm vid:pid 

also got the same error using a prebuilt uf2 from here

polhenarejos, 2024-06-25:
You have to use one of known vendors. `234b:0000` is from FSIJ and it's free to use. 
Try to use `pico-hsm-tool.py phy vidpid 234b:0000` and reboot.

dev-area, 2024-06-25:
not working 
i built it using the commands:
# PICO_SDK_PATH=../projects/pico/pico-sdk cmake .. -DDEBUG_APDU=1 -DPICO_BOARD=pico -DUSB_VID=0x234b -DUSB_PID=0x0000
# make

after flashing it - not recognized

polhenarejos, 2024-06-25:
What `lsusb` shows?

dev-area, 2024-06-26:
ok
found the problem
Its windows OS 
I built everything again on a linux computer (not VMWare image) and its ok
it is recognized and i can use it 
but
when connecting to windows 10 computer - windows os doesnt recognize it and not forward it to the VM
the error on the device manager is:
"The USB device returned an invalid USB configuration descriptor."

polhenarejos, 2024-06-26:
Usually VM deny SmartCard devices for security purposes. Like Nitrokey or Yubikey, we do not support VM.

dev-area, 2024-06-26:
that means we cant use the pico hsm on windows
Im using version 3.6 with several projects on windows

polhenarejos, 2024-06-26:
3.6 worked with VM?

dev-area, 2024-06-26:
Yes
Vm and windows host without a problem

polhenarejos, 2024-06-27:
Try this:

In line https://github.com/polhenarejos/pico-keys-sdk/blob/01d1de6074ab8be93319e469f1f706a7dc444a24/src/usb/usb_descriptors.c#L123

replace `U32_TO_U8S_LE(0x2)` by `U32_TO_U8S_LE(0x1 | 0x2)` , build the firmware and reflash

dev-area, 2024-06-27:
same problem

polhenarejos, 2024-06-27:
Try this:

In line https://github.com/polhenarejos/pico-keys-sdk/blob/01d1de6074ab8be93319e469f1f706a7dc444a24/src/usb/usb_descriptors.c#L37 

replace `0x0210` by `0x0200`

dev-area, 2024-06-27:
yes,
now its ok
with both changes
thanks

closed 2024-06-27 by dev-area

polhenarejos, 2024-06-27:
Good to know. Can you undo the first change and leave it to `U32_TO_U8S_LE(0x2)` ? I am not sure if both are fixes or just the second.

dev-area, 2024-06-27:
its ok,
only the second change is required (0x200)

thanks

commit bf2f961b (2024-06-27): Fix windows compatibility. Fixes #44.