polhenarejos/pico-hsm #47: esp32s3

ihavetenfingers, 2024-08-16:
ive somehow (no experience at all with anything like this) managed to get this running on a waveshare rp2040 zero clone, an esp32s3 zero clone from the precompiled files, and lately compiled it myself for the s3. 

Using the s3 is however super slow when generating rsa keys, not only 4096 but even anything lower than that.
I also cannot get openpgp running on any of my tries, and pkcs15-init and pkcs15-tools refuses to do much with the cards besides listing what is already there.

Love all of these pico-key projects so far even though ive had mixed luck getting it working.

Any idea what's up or how i would go about getting further in this?
And on another note, any reason why hsm isnt in the same project as fido and openpgp?

polhenarejos, 2024-08-16:
This is something I also encountered. I am afraid that this is due to the internal random number generator, which feeds the entropy really slow. My recommendation is to skip all RSA key stuff and go to ECC, it should be much much faster. 

When you say OpenPGP, are you referring to Pico OpenPGP with ESP32? I have not adapted the code to work with ESP32. Have you make it work? If you are using Pico HSM, it is not compatible with openpgp, since both use different applications (even both use the same CCID interface). 

The reason of having separate projects is twofold:
1. Have the a minimal footprint. When I started all three projects I did not know a priori how much space would occupy the firmware, so I decided to present them separately. 
2. Avoid object overlapping. There is a list of objects in the HSM, OpenPGP and FIDO that share the same reference internally. Cirvumenting that requires lots of modifications and is tricky to test it. Even with PIV and OpenPGP also happened, where there are two objects sharing the same reference that drove some issues.

Unify OpenPGP and HSM is perhaps the natural step, since both worlds share exchangable tools and share the same interface, but this is something that will happen in the mid-term. This is not the case for FIDO, since the use, tools and interfaces are totally different.

ihavetenfingers, 2024-08-18:
> This is something I also encountered. I am afraid that this is due to the internal random number generator, which feeds the entropy really slow. My recommendation is to skip all RSA key stuff and go to ECC, it should be much much faster.
> 
Gotcha! I will stop chasing that one then.

> When you say OpenPGP, are you referring to Pico OpenPGP with ESP32? I have not adapted the code to work with ESP32. Have you make it work? If you are using Pico HSM, it is not compatible with openpgp, since both use different applications (even both use the same CCID interface).
> 
I was referring to pico-hsm, I figured It would work since they share somr similarities. Pico-openpgp works great by itself on my rp2040 boards though!

> The reason of having separate projects is twofold:
>
>     1. Have the a minimal footprint. When I started all three projects I did not know a priori how much space would occupy the firmware, so I [...]

I really do appreciate these projects, even though I cannot contribute code of any value. Thank you (and any other contributors) for taking your time keeping something as neat and important as this open sourced. All around simply amazing!

Bless!

ihavetenfingers, 2024-08-18:
Closing since i got answers

closed 2024-08-18 by ihavetenfingers