polhenarejos/pico-hsm #54: fail to generate pico_hsm.uf2 when launching the project

RuixiangJiang, 2024-09-23:
Greetings! I'm pretty interested in this project, but I have some problems building it.

The first problem is that when I download the project (no matter the release version 4.2 or the latest version), the `pico-keys-sdk` folder is empty as it actually provides an external link to another project. So I have to move to some other projects to complete the folder, which is not convenient. I've tried downloading by `git clone` and simply clicking the `Download ZIP` provided by the GitHub website. I wonder if there are some easier ways to download, like using a single command, and everything would be done automatically then.

The second problem is my failure to generate `pico_hsm.uf2`. I downloaded the `pico-sdk` from https://github.com/raspberrypi/pico-sdk/releases/tag/2.0.0, and I used `PICO_SDK_PATH=my_own_path_to_pico_sdk cmake ..` in `build` folder to build it. The picture below shows the detailed output of the makefile script. However, as shown in the picture, there is neither `pico_hsm.uf2` in the build folder, nor any error message. That confuses me, so I raise this issue, looking for help.

[image]

For your review, my operating system is MacOS 14.6 (arm64). If there is any other information that I need to provide, please let me know. Thanks in advance!

RuixiangJiang, 2024-09-23:
By the way, I've used `pico_sdk_import.cmake` provided by https://github.com/raspberrypi/pico-sdk/releases/tag/2.0.0 to replace the file with the same name provided by your team. If I don't do the replacement, the result is still the same, that is, neither any error message nor `pico_hsm.uf2`.

polhenarejos, 2024-09-23:
`--recursive` should clone modules too.  It is also happening in your `pico-sdk` repo, as it says to use `git submodule` command.

I'll handle `pico_sdk_import.cmake`.

RuixiangJiang, 2024-09-24:
Thanks for being so patient and responsive! I've tried running `git submodule update --init --recursive` within `pico-hsm` folder, and everything looks fine. However, if I try to execute this command outside the `pico-hsm` folder, i.e., in the folder where I cloned your repository, it raises an error that the folder is not a git repo. So I use `git init` to initialize it and then use `git submodule update --init --recursive` to update it, but nothing happens... I mean, I think it should be updated recursively, so it will be okay to execute `git submodule update` first and then `cd pico-hsm`. However, on my laptop, it's okay only when I enter `pico-hsm` first and then execute `git submodule update`.

I've also executed `git submodule update` in the `pico-sdk-2.0.0` folder provided by raspberrypi, but nothing happened. I guess it doesn't contain submodules.

So now I move on to the final command, `make`. Here I encounter an error message shown in the figure:

[image]

In StackOverflow, I find that somebody has the same issue as I do. He said he swiched from `arm-none-eabi-gcc` to `gcc-arm-embedded` to solve the problem. I wonder what your development environment is, Linux, macOS, or Windows? Would it be better to run this project on Windows? Thanks in advance.

RuixiangJiang, 2024-09-24:
Wait, I use Windows for another try, and I find there are some submodules in the `pick-sdk` folder... I don't know why I fail to update such submodules on my macbook, but I'll fix it.

RuixiangJiang, 2024-09-24:
Now I've successfully generated the artifact `pico_hsm.uf2`. I wonder how to test it, as I don't have a development board. I mean, I notice that there is a `test` folder in your repo, and I wonder how to test without a board, maybe through Docker as mentioned in the readme file?

closed 2024-09-24 by RuixiangJiang