polhenarejos/pico-hsm #55: Can't create pico-hsm-builder docker image

fastchain, 2024-10-01:
Hi.

I run this command in git root
```
sudo docker build \
    --build-arg VERSION_PICO_SDK=1.5.0 \
    --build-arg VERSION_MAJOR=4 \
    --build-arg VERSION_MINOR=2 \
    --build-arg PICO_BOARD=pico \
    --build-arg USB_VID=0xfeff \
    --build-arg USB_PID=0xfcfd \
    -t pico-hsm-builder .
```

and then I get this 

```
....
2.619 [ 98%] Building C object CMakeFiles/pico_hsm.dir/home/builduser/Devel/pico/pico-sdk/lib/tinyusb/hw/bsp/rp2040/family.c.obj
2.619 [100%] Building C object CMakeFiles/pico_hsm.dir/home/builduser/Devel/pico/pico-sdk/src/rp2_common/hardware_pio/pio.c.obj
2.726 make[2]: Target 'CMakeFiles/pico_hsm.dir/build' not remade because of errors.
2.727 make[1]: *** [CMakeFiles/Makefile2:1767: CMakeFiles/pico_hsm.dir/all] Error 2
2.727 make[1]: Target 'all' not remade because of errors.
2.727 make: *** [Makefile:103: all] Error 2
2.727 make: Target 'default_target' not remade because of errors.
------
Dockerfile:50
--------------------
  49 |     ARG PICO_BOARD=pico
  50 | >>> RUN cd /home/builduser/pico-hsm \
  51 | >>>      && cd build_release \
  52 | >>>      && cmake .. -DPICO_BOARD=$PICO_BOARD -DUSB_VID=${USB_VID} -DUSB_PID=${USB_PID} \
  53 | >>>      && make -kj20
  54 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /home/builduser/pico-hsm \t&& cd build_release \t&& cmake .. -DPICO_BOARD=$PICO_BOARD -DUSB_VID=${USB_VID} -DUSB_PID=${USB_PID} \t&& make -kj20" did not complete successfully: exit code: 2

```

What am I doing wrong?

polhenarejos, 2024-10-01:
There's an error some lines above. Paste the entire log.

fastchain, 2024-10-01:
here is a full log:

```
sudo docker build \
    --build-arg VERSION_PICO_SDK=1.5.0 \
    --build-arg VERSION_MAJOR=4 \
    --build-arg VERSION_MINOR=2 \
    --build-arg PICO_BOARD=pico \
    --build-arg USB_VID=0xfeff \
    --build-arg USB_PID=0xfcfd \
    -t pico-hsm-builder .
[+] Building 3.1s (13/13) FINISHED                                                                                                                                                                                  docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                                          0.0s
 => => transferring dockerfile: 1.25kB                                                                                                                                                                                        0.0s
 => [internal] load metadata for docker.io/library/debian:bullseye                                                                                                                                                            0.9s
 => [internal] load .dockerignore                                                                                                                                                                                             0.0s
 => => transferring context: 2B                                                                                                                                                                                               0.0s
 => [ 1/10] FROM docker.io/library/debian:bullseye@sha256:152b9a5dc2a03f18ddfd88fbe7b1df41bd2b16be9f2df573a373caf46ce78c08                                                                                                    0.0s
 => CACHED [ 2/10] RUN apt-get update && apt-get install -y  build-essential  git  cmake  gcc-arm-none-eabi  libnewlib-arm-none-eabi  libstdc++-arm-none-eabi-newlib  python3  python3-pip                                    0.0s
 => CACHED [ 3/10] RUN useradd -m builduser                                                                                                                                                                                   0.0s
 => CACHED [ 4/10] WORKDIR /home/builduser                                                                                                                                                                                    0.0s
 => CACHED [ 5/10] RUN mkdir -p /home/builduser/Devel/pico                                                                                                                                                                    0.0s
 => CACHED [ 6/10] RUN cd /home/builduser/Devel/pico  && git clone https://github.com/raspberrypi/pico-sdk.git         && cd pico-sdk         && git checkout 1.5.0         && git submodule update --init --recursive        0.0s
 => CACHED [ 7/10] RUN pip install cryptography                                                                                                                                                                               0.0s
 => CACHED [ 8/10] RUN cd /home/builduser  && git clone https://github.com/polhenarejos/pico-hsm.git  && cd pico-hsm  && git checkout v4.2  && git submodule update --init --recursive  && mkdir build_release                0.0s
 => [ 9/10] RUN echo pico                                                                                                                                                                                                     0.2s
 => ERROR [10/10] RUN cd /home/builduser/pico-hsm  && cd build_release  && cmake .. -DPICO_BOARD=pico -DUSB_VID=0xfeff -DUSB_PID=0xfcfd  && make -kj20                                                                        2.0s
------
 > [10/10] RUN cd /home/builduser/pico-hsm      && cd build_release     && cmake .. -DPICO_BOARD=pico -DUSB_VID=0xfeff -DUSB_PID=0xfcfd         && make -kj20:
0.186 Using PICO_SDK_PATH from environment ('/home/builduser/Devel/pico/pico-sdk')
0.186 PICO_SDK_PATH is /home/builduser/Devel/pico/pico-sdk
0.186 Defaulting PICO_PLATFORM to rp2040 since not specified.
0.186 Defaulting PICO platform compiler to pico_arm_gcc since not specified.
0.186 -- Defaulting build type to 'Release' since not specified.
0.186 PICO compiler is pico_arm_gcc
0.222 -- The C compiler identification is GNU 8.3.1
0.256 -- The CXX compiler identification is GNU 8.3.1
0.262 -- The ASM compiler identification is GNU
0.263 -- Found assembler: /usr/bin/arm-none-eabi-gcc
0.265 -- Detecting C compiler ABI info
0.301 -- Detecting C compiler ABI info - done
0.307 -- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
0.307 -- Detecting C compile features
0.307 -- Detecting C compile features - done
0.310 -- Detecting CXX compiler ABI info
0.349 -- Detecting CXX compiler ABI info - done
0.354 -- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
0.354 -- Detecting CXX compile features
0.355 -- Detecting CXX compile features - done
0.356 Build type is Release
0.356 Using PICO_BOARD from environment ('pico')
0.356 Using board configuration from /home/builduser/Devel/pico/pico-sdk/src/boards/include/boards/pico.h
0.403 -- Found Python3: /usr/bin/python3.9 (found version "3.9.2") found components: Interpreter 
[... cut ...]
  49 |     ARG PICO_BOARD=pico
  50 | >>> RUN cd /home/builduser/pico-hsm \
  51 | >>>      && cd build_release \
  52 | >>>      && cmake .. -DPICO_BOARD=$PICO_BOARD -DUSB_VID=${USB_VID} -DUSB_PID=${USB_PID} \
  53 | >>>      && make -kj20
  54 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /home/builduser/pico-hsm \t&& cd build_release \t&& cmake .. -DPICO_BOARD=$PICO_BOARD -DUSB_VID=${USB_VID} -DUSB_PID=${USB_PID} \t&& make -kj20" did not complete successfully: exit code: 2

```

polhenarejos, 2024-10-01:
You must use `VERSION_PICO_SDK=2.0.0`

closed 2024-10-01 by fastchain