polhenarejos/pico-hsm #57: Secret key import

fastchain, 2024-10-20:
Hello,
is there any alternatives to SCS3 to import secret key?  SCS3 does not fit, since it requires interactions with GUI, so it's impossible to use for automation?

Is there is no, are there any low level description of DKEK wrapping mechanics, detailed enough to create wrapped container outside HSM?

Thank you

polhenarejos, 2024-10-20:
Keys can be imported using OpenSC sc-hsm-init tool, —wrap/—unwrap flags. 

https://github.com/OpenSC/OpenSC/wiki/SmartCardHSM

polhenarejos, 2024-10-21:
I guess you are referring a P12 container with a private key in it. In this case, only SCS3 can import these containers. SCS3 has a GUI but it can also be used with a CLI by creating a javascript.

In Python, PicoHSM has `import_key()` method which accepts `rsa.RSAPrivateKey`, `ec.EllipticCurvePrivateKey`, `x25519.X25519PrivateKey`, `x448.X448PrivateKey`, `ed25519.Ed25519PrivateKey` o `ed448.Ed448PrivateKey` private key argument. You could unwrap first the P12 container with `pkcs12.load_key_and_certificates` which will return the private key and then call `import_key()`.

Note that, in all cases, you will need all DKEK used in the initialization process.

fastchain, 2024-10-21:
> I guess you are referring a P12 container with a private key in it

No, it's just a raw bytes array. 
Anyway, thank you for the info, I'll try to play with import_key().

fastchain, 2024-10-22:
https://github.com/polhenarejos/pypicohsm/blob/2142e8ed51f8897d97db40b672f3c3e10db22868/picohsm/PicoHSM.py#L619

works just fine. the only minor problem is object id is ASCII encoded, so a new key on freshly inited device always gets id 31.

closed 2024-10-22 by fastchain

polhenarejos, 2024-10-22:
What do you mean?

fastchain, 2024-10-24:
@polhenarejos 
If  I run this or similar code against a clear just initialized device

```
... 
curve=ec.SECP256K1
    pkey = ec.derive_private_key(
        int.from_bytes(device_config.ETH_SECRET_KEY, byteorder='big'),
        curve(),  # Curve used in Ethereum
        default_backend()
    )
    device_config.KEY_ID = device.import_key(pkey,dkek=device_config.DKEK)
...
```
the key ID should be 1. 

But when I run **pkcs11-tool -O** I see this:

```
#pkcs11-tool -O

Using slot 2 with a present token (0x8)
Public Key Object; EC  EC_POINT 256 bits
  EC_POINT:   044104e42b7ee5c1357599b60f07c330324aaefed919e93b4bee1a029cc83930eb117f8a206acb460970ce99956a6e7e2c1b1f8417779e146f0576e673fb382a707164
  EC_PARAMS:  06082a8648ce3d030107
  label:      ESPICOHSMTR
  ID:         0000000000000000000000000000000000000000
  Usage:      verify
  Access:     none
Public Key Object; EC  EC_POINT 256 bits
  EC_POINT:   0441047e3fbe8bd79efb6765a0e13e9f4eae7d4db1c484a920dff76a2783720ec8a5b570493850c320294790abe0c4d0f2e521b68f39e056bd41f6dfac93bf60b2c0c1
  EC_PARAMS:  06052b8104000a
  label:      
  ID:         31
  Usage:      verify
  Access:     none
Profile object 1944761952
  profile_id:          CKP_PUBLIC_CERTIFICATES_TOKEN (4)
```
ID is 31.

I can't understand exactly why it works this way.  I have a hypothesis, that there is  datatype confusion somewhere and string "1"  saved as int.