polhenarejos/pico-hsm #58: pico-keys-sdk: fatal error: hardware/pio.h: No such file or directory

patvdleer, 2024-10-31:
Error;
```
/home/patrick/git/picokeys/pico-hsm/pico-keys-sdk/src/led/led_ws2812.c:22:10: fatal error: hardware/pio.h: No such file or directory
   22 | #include "hardware/pio.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/pico_hsm.dir/build.make:692: CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/led/led_ws2812.c.obj] Error 1
```

Should the pico-keys-sdk be updated to a different version? The other projects, FIDO and OpenGPG, do compile.

Commands ran;
```bash
mkdir picokeys
cd picokeys
git clone --recursive --branch 2.0.0 https://github.com/raspberrypi/pico-sdk.git

git clone --recursive https://github.com/polhenarejos/pico-hsm.git
cd pico-hsm/
mkdir build
cd build
PICO_SDK_PATH=../../pico-sdk cmake .. -DPICO_BOARD=waveshare_rp2040_one -DVIDPID=GnuPG 
make -j 16
```

Complete log:
```
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys$ git clone --recursive https://github.com/polhenarejos/pico-hsm.git
cd pico-hsm/
mkdir build
cd build
PICO_SDK_PATH=../../pico-sdk cmake .. -DPICO_BOARD=waveshare_rp2040_one -DVIDPID=GnuPG 
make -j 16
Cloning into 'pico-hsm'...
remote: Enumerating objects: 5220, done.
remote: Counting objects: 100% (500/500), done.
remote: Compressing objects: 100% (170/170), done.
remote: Total 5220 (delta 329), reused 462 (delta 307), pack-reused 4720 (from 1)
Receiving objects: 100% (5220/5220), 2.08 MiB | 13.01 MiB/s, done.
Resolving deltas: 100% (3371/3371), done.
Submodule 'pico-keys-sdk' (https://github.com/polhenarejos/pico-keys-sdk) registered for path 'pico-keys-sdk'
Cloning into '/home/patrick/git/picokeys/pico-hsm/pico-keys-sdk'...
remote: Enumerating objects: 2654, done.        
remote: Counting objects: 100% (776/776), done.        
remote: Compressing objects: 100% (320/320), done.        
remote: Total 2654 (delta 586), reused 626 (delta 452), pack-reused 1878 (from 1)        
Receiving objects: 100% (2654/2654), 834.97 KiB | 9.38 MiB/s, done.
Resolving deltas: 100% (1782/1782), done.
Submodule path 'pico-keys-sdk': checked out '1bf323c36789e7c1a9273ca7ae5f3ad221fcbef5'
Submodule 'mbedtls' (https://github.com/ARMmbed/mbedtls) registered for path 'pico-keys-sdk/mbedtls'
Submodule 'tinycbor' (https://github.com/intel/tinycbor.git) registered for path 'pico-keys-sdk/tinycbor'
Cloning into '/home/patrick/git/picokeys/pico-hsm/pico-keys-sdk/mbedtls'...
remote: Enumerating objects: 262065, done.        
remote: Counting objects: 100% (244/244), done.        
remote: Compressing objects: 100% (70/70), done.        
remote: Total 262065 (delta 198), reused 198 (delta 174), pack-reused 261821 (from 1)        
Receiving objects: 100% (262065/262065), 125.20 MiB | 34.99 MiB/s, done.
Resolving deltas: 100% (204585/204585), done.
Cloning into '/home/patrick/git/picokeys/pico-hsm/pico-keys-sdk/tinycbor'...
remote: Enumerating objects: 3191, done.        
remote: Counting objects: 100% (392/392), done.        
remote: Compressing objects: 100% (167/167), done.        
remote: Total 3191 (delta 241), reused 339 (delta 216), pack-reused 2799 (from 1)        
Receiving objects: 100% (3191/3191), 1.46 MiB | 13.73 MiB/s, done.
Resolving deltas: 100% (2136/2136), done.
Submodule path 'pico-keys-sdk/mbedtls': checked out '71c569d44bf3a8bd53d874c81ee8ac644dd6e9e3'
Submodule 'framework' (https://github.com/Mbed-TLS/mbedtls-framework) registered for path 'pico-keys-sdk/mbedtls/framework'
Cloning into '/home/patrick/git/picokeys/pico-hsm/pico-keys-sdk/mbedtls/framework'...
remote: Enumerating objects: 200487, done.        
remote: Counting objects: 100% (12216/12216), done.        
[... 152 lines cut ...]
[ 27%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/rng/random.c.obj
[ 27%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/rng/hwrng.c.obj
[ 28%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/eac.c.obj
[ 28%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/crypto_utils.c.obj
[ 29%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/asn1.c.obj
[ 29%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/apdu.c.obj
[ 30%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/led/led.c.obj
[ 30%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/led/led_ws2812.c.obj
[ 31%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/led/led_pico.c.obj
[ 31%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/usb/ccid/ccid.c.obj
[ 32%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/usb/usb_descriptors.c.obj
[ 32%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/mbedtls/library/aes.c.obj
[ 34%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/mbedtls/library/asn1parse.c.obj
[ 35%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/mbedtls/library/bignum.c.obj
[ 35%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/mbedtls/library/asn1write.c.obj
[ 35%] Building C object CMakeFiles/pico_hsm.dir/pico-keys-sdk/mbedtls/library/bignum_core.c.obj
/home/patrick/git/picokeys/pico-hsm/pico-keys-sdk/src/led/led_ws2812.c:22:10: fatal error: hardware/pio.h: No such file or directory
   22 | #include "hardware/pio.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/pico_hsm.dir/build.make:692: CMakeFiles/pico_hsm.dir/pico-keys-sdk/src/led/led_ws2812.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1748: CMakeFiles/pico_hsm.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

polhenarejos, 2024-10-31:
Can you try `development` branch?

patvdleer, 2024-11-01:
Fixed!

```
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm$ rm -rf build
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm$ git checkout development
M	pico-keys-sdk
Branch 'development' set up to track remote branch 'development' from 'origin' by rebasing.
Switched to a new branch 'development'
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm$ git pull
Already up to date.
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm$ git submodule update
Submodule path 'pico-keys-sdk': checked out '62c3d0c3603e673ac4a338450db27fd3356468ac'
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm$ mkdir build
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm$ cd build/
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm/build$ PICO_SDK_PATH=../../pico-sdk cmake .. -DPICO_BOARD=waveshare_rp2040_one -DVIDPID=GnuPG 
Using PICO_SDK_PATH from environment ('../../pico-sdk')
PICO_SDK_PATH is /home/patrick/git/picokeys/pico-sdk
Target board (PICO_BOARD) is 'waveshare_rp2040_one'.
Using board configuration from /home/patrick/git/picokeys/pico-sdk/src/boards/include/boards/waveshare_rp2040_one.h
Defaulting platform (PICO_PLATFORM) to 'rp2040' based on PICO_BOARD setting.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
CMake Warning at /home/patrick/git/picokeys/pico-sdk/tools/Findpicotool.cmake:28 (message):
  No installed picotool with version 2.0.0 found - building from source

  It is recommended to build and install picotool separately, or to set
  PICOTOOL_FETCH_FROM_GIT_PATH to a common directory for all your SDK
  projects
Call Stack (most recent call first):
  /home/patrick/git/picokeys/pico-sdk/tools/CMakeLists.txt:138 (find_package)
  /home/patrick/git/picokeys/pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  /home/patrick/git/picokeys/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  /home/patrick/git/picokeys/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)

Downloading Picotool
-- Found Python3: /usr/bin/python3.12 (found version "3.12.3") found components: Interpreter 
TinyUSB available at /home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/patrick/git/picokeys/pico-sdk/lib/btstack
cyw43-driver available at /home/patrick/git/picokeys/pico-sdk/lib/cyw43-driver
lwIP available at /home/patrick/git/picokeys/pico-sdk/lib/lwip
mbedtls available at /home/patrick/git/picokeys/pico-sdk/lib/mbedtls
-- VIDPID:			 'GnuPG'
-- Delayed boot:		 disabled
-- USB CCID Interface:		 enabled
-- USB WebCCID Interface:	 enabled
-- USB VID/PID:			 0x1209:0x2440
-- LED driver:			 led_ws2812
-- Found version:		 4.2
-- Configuring done
-- Generating done
-- Build files have been written to: /home/patrick/git/picokeys/pico-hsm/build
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/picokeys/pico-hsm/build$ make -j 16
[  0%] Built target picotoolForceReconfigure
[... 201 lines cut ...]
[ 87%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/src/rp2_common/hardware_rtc/rtc.c.obj
[ 88%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c.obj
[ 88%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[ 89%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/device/usbd.c.obj
[ 89%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/device/usbd_control.c.obj
[ 90%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/audio/audio_device.c.obj
[ 90%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/cdc/cdc_device.c.obj
[ 91%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/dfu/dfu_device.c.obj
[ 91%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/dfu/dfu_rt_device.c.obj
[ 92%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c.obj
[ 92%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/midi/midi_device.c.obj
[ 93%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c.obj
[... cut ...]
[ 96%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/src/common/tusb_fifo.c.obj
[ 97%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[ 98%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/lib/tinyusb/hw/bsp/rp2040/family.c.obj
[ 98%] Building C object CMakeFiles/pico_hsm.dir/home/patrick/git/picokeys/pico-sdk/src/rp2_common/hardware_pio/pio.c.obj
[100%] Linking CXX executable pico_hsm.elf
[100%] Built target pico_hsm
```

closed 2024-11-01 by patvdleer