polhenarejos/pico-hsm #60: Cannot connect to the remote server

EdXian, 2024-11-12:
Hi @polhenarejos ,
I'm intereting to pico-hsm. Now, I'm trying to setup the pico-hsm on my pico2.
Howerver, I can't connect to the server when i executed the command `python3 tools/pico-hsm-tool.py --pin 648219 initialize --so-pin 57621880`. I have seen the similar issue in the history. Is it caused by the remote server is turn-off?  Can you help me  restart the remote server. thanks a lot!

```
Pico HSM Tool v2.0
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

********************************
*   PLEASE READ IT CAREFULLY   *
********************************

This tool will erase and reset your device. It will delete all private and secret keys.
Are you sure?
[Press enter to confirm]
Public Point: 00
Traceback (most recent call last):
  File "C:\Users\edwu\Desktop\pico-hsm\tools\pico-hsm-tool.py", line 543, in <module>
    run()
  File "C:\Users\edwu\Desktop\pico-hsm\tools\pico-hsm-tool.py", line 540, in run
    main(args)
  File "C:\Users\edwu\Desktop\pico-hsm\tools\pico-hsm-tool.py", line 518, in main
    initialize(picohsm, args)
  File "C:\Users\edwu\Desktop\pico-hsm\tools\pico-hsm-tool.py", line 230, in initialize
    j = get_pki_data('cvc', data=data)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\edwu\Desktop\pico-hsm\tools\pico-hsm-tool.py", line 165, in get_pki_data
    response = urllib.request.urlopen(req)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2032.0_x64__qbz5n2kfra8p0\Lib\urllib\request.py", line 215, in urlopen
    return opener.open(url, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2032.0_x64__qbz5n2kfra8p0\Lib\urllib\request.py", line 521, in open
    response = meth(req, response)
               ^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2032.0_x64__qbz5n2kfra8p0\Lib\urllib\request.py", line 630, in http_response
    response = self.parent.error(
               ^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2032.0_x64__qbz5n2kfra8p0\Lib\urllib\request.py", line 559, in error
    return self._call_chain(*args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2032.0_x64__qbz5n2kfra8p0\Lib\urllib\request.py", line 492, in _call_chain
    result = func(*args)
             ^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2032.0_x64__qbz5n2kfra8p0\Lib\urllib\request.py", line 639, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 500: Internal Server Error
```

polhenarejos, 2024-11-12:
It seems work to me. Can you try again?

EdXian, 2024-11-12:
Sorry ... I still got the same error. 
Should i set something before initialize the pico-hsm?
[image]

EdXian, 2024-11-12:
Hello @polhenarejos ,
I use the [pico2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/)(rp2350) released from raspberry pi official. what nuke version should i program for this board?
[image]

I have trird some nuke.uf2. However, the public point is always 00.
Thank you!

EdXian, 2024-11-12:
Update
1. I tried to program with [pico_nuke_pico2-1.2.uf2](https://github.com/polhenarejos/pico-nuke/releases/download/v1.2/pico_nuke_pico2-1.2.uf2) . the onboard led blinked 3 times and then return to bootloader mode. I think it works.

2. then I program  pico_hsm_pico2-5.0.uf2 which is downloaded from [here ](https://www.picokeys.com/getting-started/)

3. run `python tools/pico-hsm-tool.py --pin 648219 initialize --so-pin 57621880` 
4. got the same error...
[image]

polhenarejos, 2024-11-12:
Do you have the official Raspberry Pico 2 then?

polhenarejos, 2024-11-12:
`pkcs11-tool -O`

What does return?

closed 2024-11-12 by polhenarejos

polhenarejos, 2024-11-12:
I uploaded a new release that fixes this issue.

You have to update:
- The firmware
- `pico-hsm-tool.py`
- `pypicohsm` (`pip install -U pypicohsm`).

EdXian, 2024-11-13:
It works now! 👍 👍 👍

commit 1ea0a91b (2024-11-12): Fix initialization when initializing a RP2350 board for first time.

Fixes #60.