polhenarejos/pico-hsm #61: Any production ready hardware available?

abu-matterize, 2024-11-13:
Hello people,

I was wondering is there any production ready hardware available to use?

Any suggestions appreciated.

polhenarejos, 2024-11-13:
What do you mean? There are plenty of boards mounting RP2350, RP2040 or ESP32-S3 for selling from different vendors.

polhenarejos, 2024-11-13:
I did not find it. 
I use Tiny2350 which is not USB key but is really tiny.

abu-matterize, 2024-11-13:
Thanks for sharing your findings @polhenarejos

abu-matterize, 2024-11-25:
So, after searching for sometime, I decided to build my own USB-C key which can accommodate RP2350/RP2354. And of-course decided publish all of the resources.

Here is the initial version of the schematic: https://github.com/abu-matterize/pico-keychain/blob/main/hardware/schematics/picokey-schematic-v1.0.pdf

Suggestions and scrutinizes welcomed.

abu-matterize, 2024-11-25:
> What GPIO24 will be used for?

An additional button if in case required.

> I would prefer a RGBW neopixel rather than a plain green led. Most of the boards equip a neopixel.

Of-course. Will change it to WS.

closed 2025-01-15 by polhenarejos