polhenarejos/pico-hsm #62: RPI Pico 2 - `No smart card readers found`

abu-matterize, 2024-11-13:
I've similar issues with Arch. My hardware is RPI Pico 2. Flashed with pico-hsm from https://www.picokeys.com/getting-started/

`dmesg` output
```sh
[  179.413262] usb 3-2: New USB device found, idVendor=1050, idProduct=0030, bcdDevice= 7.00
[  179.413279] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  179.413286] usb 3-2: Product: Pico Key
[  179.413291] usb 3-2: Manufacturer: Pol Henarejos
[  179.413296] usb 3-2: SerialNumber: 8D68603BBDE18CBE
[  190.054006] usb 3-2: USB disconnect, device number 4
[  193.884003] usb 3-2: new full-speed USB device number 5 using xhci_hcd
[  194.040743] usb 3-2: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[  194.052907] usb 3-2: New USB device found, idVendor=1050, idProduct=0030, bcdDevice= 7.00
[  194.052924] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  194.052930] usb 3-2: Product: Pico Key
[  194.052936] usb 3-2: Manufacturer: Pol Henarejos
[  194.052941] usb 3-2: SerialNumber: 8D68603BBDE18CBE
```
When I run `opensc-tool -v -sc`
```sh
[host user]~ opensc-tool -v -sc
No smart card readers found.
Failed to connect to reader: No readers found

_Originally posted by @abu-matterize in https://github.com/polhenarejos/pico-hsm/issues/34#issuecomment-2473077417_

abu-matterize, 2024-11-13:
Originally posted in #34 https://github.com/polhenarejos/pico-hsm/issues/34#issuecomment-2473077417

polhenarejos, 2024-11-13:
Yubico HSM is not supported by OpenSC. Try with FISJ or Nitrokey HSM.

abu-matterize, 2024-11-13:
Let me give it a try.

abu-matterize, 2024-11-13:
It's the same. No changes. `opensc-tool -i` returns
```sh
> opensc-tool -i
OpenSC 0.25.1 [gcc  13.2.1 20230801]
Enabled features: locking zlib readline openssl pcsc(libpcsclite.so.1)
```

polhenarejos, 2024-11-13:
Paste dmesg log

abu-matterize, 2024-11-13:
Sure.

Here is the steps followed after copying `flash_nuke.uf2`
1. `cp Downloads/pico_hsm_pico2-5.0-eddsa1.uf2 /run/media/amac/RP2350/`
2. `lsusb` - `Bus 001 Device 042: ID feff:fcfd Pol Henarejos Pico Key`
3. `dmesg`
```sh
[ 9679.039775] usb 1-2: new full-speed USB device number 42 using xhci_hcd
[ 9679.181077] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[ 9679.181539] usb 1-2: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 7.00
[ 9679.181542] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9679.181544] usb 1-2: Product: Pico Key
[ 9679.181546] usb 1-2: Manufacturer: Pol Henarejos
[ 9679.181548] usb 1-2: SerialNumber: 8D68603BBDE18CBE
~ >                                                            
```
4. Visited https://www.picokeys.com/pico-commissioner/ and set vendor to `Nitrokey HSM` [following screenshot]
[image]
5. Power cycled Pico
6. `dmesg`
```sh
[ 9679.039775] usb 1-2: new full-speed USB device number 42 using xhci_hcd
[ 9679.181077] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[ 9679.181539] usb 1-2: New USB device found, idVendor=feff, idProduct=fcfd, bcdDevice= 7.00
[ 9679.181542] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9679.181544] usb 1-2: Product: Pico Key
[ 9679.181546] usb 1-2: Manufacturer: Pol Henarejos
[ 9679.181548] usb 1-2: SerialNumber: 8D68603BBDE18CBE
[10189.395786] usb 1-2: USB disconnect, device number 42
[10196.786623] usb 1-2: new full-speed USB device number 43 using xhci_hcd
[10196.928324] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x82 has an invalid bInterval 0, changing to 10
[10196.928817] usb 1-2: New USB device found, idVendor=20a0, idProduct=4230, bcdDevice= 7.00
[10196.928827] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10196.928834] usb 1-2: Product: Pico Key
[10196.928840] usb 1-2: Manufacturer: Pol Henarejos
[10196.928845] usb 1-2: SerialNumber: 8D68603BBDE18CBE
~ > 
```
7. `lsusb` - `Bus 001 Device 043: ID 20a0:4230 Clay Logic Nitrokey HSM`
8. `opensc-tool -an`
```sh
> opensc-tool -an
No smart card readers found.
Failed to connect to reader: No readers found
~ > 
```

polhenarejos, 2024-11-13:
Seems a problem with OpenSC+Linux, perhaps a permission issue. Does `sudo` work?

abu-matterize, 2024-11-13:
No.

polhenarejos, 2024-11-13:
Then you should try to recover pcsclite. Ensure that `pcscd` service is running properly, start `pcscd` manual to get debug, `opensc-tool -l` should return the list of found readers or `pcsc_scan`. 
If you prepend `OPENSC_DEBUG=9 opensc_command` it will output verbose debug for this command that will help.
It could also be a problem of libraries or similar.

abu-matterize, 2024-11-13:
`opensc-tool -l` with debug
```sh
> OPENSC_DEBUG=9 opensc-tool -l
P:8203; T:0x128690580975168 18:43:09.958 [opensc-tool] ctx.c:753:process_config_file: Used configuration file '/etc/opensc.conf'
P:8203; T:0x128690580975168 18:43:09.958 [opensc-tool] ctx.c:981:sc_context_create: ===================================
P:8203; T:0x128690580975168 18:43:09.958 [opensc-tool] ctx.c:982:sc_context_create: OpenSC version: 0.25.1
P:8203; T:0x128690580975168 18:43:09.958 [opensc-tool] ctx.c:983:sc_context_create: Configured for opensc-tool (/usr/bin/opensc-tool)
P:8203; T:0x128690580975168 18:43:09.959 [opensc-tool] reader-pcsc.c:890:pcsc_init: PC/SC options: connect_exclusive=0 disconnect_action=0 transaction_end_action=0 reconnect_action=0 enable_pinpad=1 enable_pace=1
P:8203; T:0x128690580975168 18:43:09.959 [opensc-tool] reader-pcsc.c:1397:pcsc_detect_readers: called
P:8203; T:0x128690580975168 18:43:09.959 [opensc-tool] reader-pcsc.c:1410:pcsc_detect_readers: Probing PC/SC readers
P:8203; T:0x128690580975168 18:43:09.959 [opensc-tool] reader-pcsc.c:1463:pcsc_detect_readers: Establish PC/SC context
P:8203; T:0x128690580975168 18:43:09.973 [opensc-tool] reader-pcsc.c:1458:pcsc_detect_readers: SCardListReaders failed: 0x8010002e
P:8203; T:0x128690580975168 18:43:09.973 [opensc-tool] reader-pcsc.c:1579:pcsc_detect_readers: returning with: -1101 (No readers found)
No smart card readers found.
P:8203; T:0x128690580975168 18:43:09.973 [opensc-tool] ctx.c:1066:sc_release_context: called
P:8203; T:0x128690580975168 18:43:09.973 [opensc-tool] reader-pcsc.c:978:pcsc_finish: called
```

abu-matterize, 2024-11-13:
Done a quick check with macOS and it's finding it.
```sh
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Nitrokey Nitrokey HSM
```

closed 2025-01-15 by polhenarejos