polhenarejos/pico-hsm #63: Commissioning failed: pico/pico-hsm/cvc Error 500

rrottmann, 2024-11-22:
ESP32S3 freshly flashed cannot be commisioned. Web console shows communication hence and forth to the Pico Key with last line Public Point: 04dff...36e

POST https://www.picokeys.com/pico/pico-ham/cvc yields error 500 in wcid.js?v10:42

SyntaXError: Unexpected token '<', "<!DOCTYPE " is not valid JSON

polhenarejos, 2024-11-22:
Perhaps is a temporary problem with the backend. Can you try again?

polhenarejos, 2024-11-22:
A bug in the frontend affecting only ESP32.

If you refresh the page F5 should be fixed. If not, try deleting the cache.

closed 2024-11-22 by polhenarejos

polhenarejos, 2024-11-22:
Please paste line 142 of wcid.js

rrottmann, 2024-11-22:
[wcid.js](https://www.picokeys.com/js/wcid.js?v10)

```140   const pbk = base64UrlEncode(Y);
141   let params = { pubkey: pbk }
142   if (platform == PICO_PLATFORM.RP2350) {
143       params.curve = 'secp256k1';
144   }
```

polhenarejos, 2024-11-22:
Ok, you are using the unpatched js, probably due to some cache. I've updated the versioning so you should see the corrected one.

rrottmann, 2024-11-22:
Works like a charm now.