polhenarejos/pico-hsm #64: Tutorial on using Pico-HSM with AWS IAM Roles Anywhere

1oh1, 2024-11-23:
Is there a set of instructions I can follow to get pico-hsm working with [aws_signing_helper](https://github.com/aws/rolesanywhere-credential-helper) of [IAM Roles Anywhere](https://docs.aws.amazon.com/rolesanywhere/latest/userguide/introduction.html)?

I'm looking for a guide similar to this one from AWS on using their aws_signing_helper tool with a Yubikey series 5: https://aws.amazon.com/blogs/security/how-to-implement-cryptographic-modules-to-secure-private-keys-used-with-iam-roles-anywhere/

Can the pico-hsm be used instead of a Yubikey series 5 for such a setup?

polhenarejos, 2024-11-23:
It works like any PKCS11 token. It shouldn't be any difference.

1oh1, 2024-11-29:
I tried following the instructions to generate keys and created RSA2K and EC521 keys but I'm not able to use them to generate certificates. Referring to the keys using `0:1` doesn't seem to work.

Am I missing something?
```
Microsoft Windows [Version 10.0.22631.4541]
(c) Microsoft Corporation. All rights reserved.

C:\Users\1oh1>pkcs15-tool -D
Using reader with a card: Pol Henarejos Pico Key CCID Interface 0
PKCS#15 Card [Pico-HSM]:
        Version        : 5
        Serial number  : ESPICOHSMTR
        Manufacturer ID: Pol Henarejos
        Flags          : PRN generation, EID compliant

PIN [UserPIN]
        Object Flags   : [0x03], private, modifiable
        Auth ID        : 02
        ID             : 01
        Flags          : [0x812], local, initialized, exchangeRefData
        Length         : min_len:6, max_len:15, stored_len:0
        Pad char       : 0x00
        Reference      : 129 (0x81)
        Type           : ascii-numeric
        Path           : e82b0601040181c31f0201::
        Tries left     : 3

PIN [SOPIN]
        Object Flags   : [0x01], private
        ID             : 02
        Flags          : [0x9A], local, unblock-disabled, initialized, soPin
        Length         : min_len:16, max_len:16, stored_len:0
        Pad char       : 0x00
        Reference      : 136 (0x88)
        Type           : bcd
        Path           : e82b0601040181c31f0201::
        Tries left     : 15

Private EC Key [EC521]
        Object Flags   : [0x03], private, modifiable
        Usage          : [0x10C], sign, signRecover, derive
        Access Flags   : [0x1D], sensitive, alwaysSensitive, neverExtract, local
        Algo_refs      : 0
        FieldLength    : 521
        Key ref        : 1 (0x01)
        Native         : yes
        Auth ID        : 01
        ID             : 02
        MD:guid        : 47900a69-0461-ca27-4ccb-78889969f725

Private RSA Key [RSA2K]
        Object Flags   : [0x03], private, modifiable
        Usage          : [0x0E], decrypt, sign, signRecover
        Access Flags   : [0x1D], sensitive, alwaysSensitive, neverExtract, local
        Algo_refs      : 0
        ModLength      : 2048
        Key ref        : 2 (0x02)
        Native         : yes
        Auth ID        : 01
        ID             : 01
        MD:guid        : 9e1f2fc5-d47c-5c45-645f-ae84f3a1a073

Public EC Key [EC521]
        Object Flags   : [0x00]
        Usage          : [0x140], verify, derive
        Access Flags   : [0x02], extract
        FieldLength    : 528
        Key ref        : 0 (0x00)
        Native         : no
        ID             : 02
        DirectValue    : <present>

Public RSA Key [RSA2K]
        Object Flags   : [0x00]
        Usage          : [0x51], encrypt, wrap, verify
        Access Flags   : [0x02], extract
        ModLength      : 2048
        Key ref        : 0 (0x00)
        Native         : no
        ID             : 01
        DirectValue    : <present>

C:\Users\1oh1>pkcs11-tool -l --pin 648219 -O
Using slot 0 with a present token (0x0)
Private Key Object; RSA
  label:      RSA2K
  ID:         01
  Usage:      decrypt, sign, signRecover
  Access:     sensitive, always sensitive, never extractable, local
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%01;object=RSA2K;type=private
Public Key Object; RSA 2048 bits
  label:      RSA2K
  ID:         01
  Usage:      encrypt, verify, wrap
  Access:     none
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%01;object=RSA2K;type=public
Private Key Object; EC
  label:      EC521
  ID:         02
  Usage:      sign, signRecover, derive
  Access:     sensitive, always sensitive, never extractable, local
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%02;object=EC521;type=private
Public Key Object; EC  EC_POINT 528 bits
  EC_POINT:   0481850400a7e1e04cf860b39154273e79676e0c36c53f6cee3565a7d330c1699bee5b662327496eccbf5919b3e627fb5e31828486de1999d5fe9c3dad4f40bdb958b43d19a801b34d375b6e638cd6e41a1ab1daa49569537c43c0e2d7c1971b621a4bff5358a9a1fac81fa2feb194a6df4ea8ab55317159cd8bcee67142c7641d11af4dafa960a5
  EC_PARAMS:  06052b81040023 (OID 1.3.132.0.35)
  label:      EC521
  ID:         02
  Usage:      verify, derive
  Access:     none
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%02;object=EC521;type=public
Profile object 2586793264
  profile_id:          CKP_PUBLIC_CERTIFICATES_TOKEN (4)

C:\Users\1oh1>
C:\Users\1oh1>openssl req -engine pkcs11 -new -key 0:1 -keyform engine -out cert.pem -text -x509 -days 365
Engine "pkcs11" set.
Unable to load module C:\Program Files\OpenSC Project\OpenSC\pkcs11\libp11.dll
PKCS11_get_private_key returned NULL
Could not find private key from org.openssl.engine:pkcs11:0:1
8C3B0000:error:41000401:libp11:ERR_P11_error:Unable to load PKCS#11 module:p11_load.c:103:
8C3B0000:error:40000067:pkcs11 engine:ERR_ENG_error:invalid parameter:eng_back.c:616:
8C3B0000:error:13000080:engine routines:ENGINE_load_private_key:failed loading private key:crypto\engine\eng_pkey.c:79:
```

closed 2025-01-15 by polhenarejos