polhenarejos/pico-hsm #67: Can't reInit unlocked device

fastchain, 2024-11-26:
When device is [unlocked](https://github.com/polhenarejos/pico-hsm/blob/master/tools/pico-hsm-tool.py#L377)
init procedure ends up with this: 

```
hw  |     device.initialize(pin=device_config.PIN,
hw  |   File "/usr/local/lib/python3.9/dist-packages/picohsm/PicoHSM.py", line 340, in initialize
hw  |     self.send(cla=0x80, command=0x50, data=data)
hw  |   File "/usr/local/lib/python3.9/dist-packages/picohsm/PicoHSM.py", line 277, in send
hw  |     raise APDUResponse(sw1, sw2)
hw  | picohsm.APDU.APDUResponse: SW:6400
```

to init device, I have to disconnect from power before init operation.

Btw, is there any way to reset hsm without  disconnect or lock it back?

polhenarejos, 2024-11-26:
Did you enable securelock? Which board do you use?

fastchain, 2024-11-27:
@polhenarejos 
I use RPI Pico and this  code snippet for  Init
 
 ```
 def InitDevice(device_config,force=False):
    device = PicoHSM()
    if len(device.list_keys()) > 2 and force==False:
        print("There are keys on HW. Init procedure stopped.")
        return 0

    device.initialize(pin=device_config.PIN,
                      sopin=device_config.SOPIN,
                      options=device_config.OPTIONS,
                      retries=device_config.RETRIES,
                      dkek_shares=device_config.DKEK_SHARES)

    device.import_dkek(device_config.DKEK)

    curve=ec.SECP256K1
    pkey = ec.derive_private_key(
        int.from_bytes(device_config.ETH_SECRET_KEY, byteorder='big'),
        curve(),  
        default_backend()
    )
    device_config.KEY_ID = device.import_key(pkey,dkek=device_config.DKEK)
    device_config.safeConfig(device_config.CONFIGFILENAME)

    slck = SecureLock2(device,device_config.HSM_ACCESS_KEY)
    slck.enable_device_aut()

    return device_config
```

and this snippet for unlock before use

```
device = PicoHSM(device_config.PIN)
slck = SecureLock2(device,device_config.HSM_ACCESS_KEY)
slck.unlock_device()
```

polhenarejos, 2024-11-27:
I cannot reproduce it. I do this:
1. Initialize: `pico-hsm-tool.py initialize`
2. Enable secure: `pico-hsm-tool.py secure enable`
3. Reset
4. Unlock: `pico-hsm-tool.py secure unlock` 
5. Initialize: `pico-hsm-tool.py initialize`

And every step works. Am I missing something?

fastchain, 2024-11-27:
@polhenarejos  
you are doing everything right.
The question is about step 3. Is it possible to reset with a software?

```usbreset``` is an option, but according to my tests it's not stable: sometimes it help, sometimes I get the same exception ``` picohsm.APDU.APDUResponse: SW:6400 ```  and have to reset manually anyway. Also a device ID must be hardcoded in scripts and I would like to avoid this if possible.

polhenarejos, 2024-11-27:
Yes, it can be easily implemented. But why do you need to reboot the key?

fastchain, 2024-11-28:
@polhenarejos 

> why do you need to reboot the key?

During software testing that involves the key, it needs to be sequentially initialized few times with different predefined keys. Tests automation is currently not possible as the key must be manually reset in each init.

polhenarejos, 2024-12-04:
I added a command to reboot the pico board. It is also available in `pypicohsm`.

closed 2024-12-04 by polhenarejos