polhenarejos/pico-hsm #68: Device freeze on device  unlock

fastchain, 2024-11-28:
Board: rpi Pico
pico-hsm version: 5.0

Here are the steps to reproduce the issue:

1.     Initialize the device.
2.     Enable SecureLock.
3.     Unlock the device.

On step 3, the green light starts blinking rapidly, and no further action occurs.

polhenarejos, 2024-11-28:
Is ir reproducible with pico-hsm-tool?

fastchain, 2024-11-29:
Hi, 
No, because the pico-hsm-tool requires a keyring, and I'm on Linux, which doesn't have a compatible keyring. Instead, I use a slightly modified version of the SecureLock2 class :

```
class SecureLock2:
    def __init__(self, picohsm, secretkey):
        self.picohsm = picohsm
        self.secretkey = secretkey

    def mse(self):
        sk = ec.generate_private_key(ec.SECP256R1())
        pn = sk.public_key().public_numbers()
        self.__pb = sk.public_key().public_bytes(Encoding.X962, PublicFormat.UncompressedPoint)

        ret = self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x01, data=list(self.__pb))

        pk = ec.EllipticCurvePublicKey.from_encoded_point(ec.SECP256R1(), bytes(ret))
        shared_key = sk.exchange(ec.ECDH(), pk)

        xkdf = HKDF(
            algorithm=hashes.SHA256(),
            length=12+32,
            salt=None,
            info=self.__pb
        )
        kdf_out = xkdf.derive(shared_key)
        self.__key_enc = kdf_out[12:]
        self.__iv = kdf_out[:12]

    def encrypt_chacha(self, data):
        chacha = ChaCha20Poly1305(self.__key_enc)
        ct = chacha.encrypt(self.__iv, data, self.__pb)
        return ct

    def unlock_device(self):

        ct = self.get_skey()

        self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x03, data=list(ct))

    def _get_key_device(self):
        return self.secretkey

    def get_skey(self):
        self.mse()
        ct = self.encrypt_chacha(self._get_key_device())
        return ct

    def enable_device_aut(self):
        ct = self.get_skey()
        self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x02, data=list(ct))

    def disable_device_aut(self):
        ct = self.get_skey()
        self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x04, p3=list(ct))
```

The main difference here is that the key is provided as a variable during initialization, rather than being retrieved from a keyring.

I use the following function to initialize the device:

```
def InitDevice(device_config,force=False):
    device = PicoHSM()
    if len(device.list_keys()) > 2 and force==False:
        print("There are keys on HW. Init procedure stopped.")
        return 0

    device.initialize(pin=device_config.PIN,
                      sopin=device_config.SOPIN,
                      options=device_config.OPTIONS,
                      retries=device_config.RETRIES,
                      dkek_shares=device_config.DKEK_SHARES)

    device.import_dkek(device_config.DKEK)

    curve=ec.SECP256K1
    pkey = ec.derive_private_key(
        int.from_bytes(device_config.SECRET_KEY, byteorder='big'),
        curve(),  # Curve used in Ethereum
        default_backend()
    )
    device_config.KEY_ID = device.import_key(pkey,dkek=device_config.DKEK)
    # pubkey = device.public_key(keyid, param=curve().name)
    device_config.safeConfig(device_config.CONFIGFILENAME)

    #enabling encryption for HSM storage
    slck = SecureLock2(device,device_config.HSM_ACCESS_KEY)
    slck.enable_device_aut()
    slck.unlock_device()

    return device_config
```

And device_config.HSM_ACCESS_KEY derivation function is 

```HSM_ACCESS_KEY = secrets.token_bytes(32)```

According to tests, key freeze on line

```
self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x03, data=list(ct))
```

It seems like something had changed in firmware, because works fine with version 3.6.

polhenarejos, 2024-11-29:
Sounds strange. Which board?

fastchain, 2024-11-29:
@polhenarejos  RPI Pico

fastchain, 2024-12-02:
Here is log from ```pcsd```

```
hw  | 00008916 eventhandler.c:494:EHStatusHandlerThread() powerState: POWER_STATE_POWERED
hw  | 00172477 winscard_msg_srv.c:256:ProcessEventsServer() Common channel packet arrival
hw  | 00000025 winscard_msg_srv.c:267:ProcessEventsServer() ProcessCommonChannelRequest detects: 15
hw  | 00000001 pcscdaemon.c:133:SVCServiceRunLoop() A new context thread creation is requested: 15
hw  | 00000157 winscard_svc.c:340:ContextThread() Authorized PC/SC client
hw  | 00000016 winscard_svc.c:343:ContextThread() Thread is started: dwClientID=15, threadContext @0x5a05538e8150
hw  | 00000007 winscard_svc.c:361:ContextThread() Received command: CMD_VERSION from client 15
hw  | 00000001 winscard_svc.c:373:ContextThread() Client is protocol version 4:4
hw  | 00000001 winscard_svc.c:396:ContextThread() CMD_VERSION rv=0x0 for client 15
hw  | 00000163 winscard_svc.c:361:ContextThread() Received command: ESTABLISH_CONTEXT from client 15
hw  | 00000005 winscard.c:215:SCardEstablishContext() Establishing Context: 0x7D113133
hw  | 00000001 winscard_svc.c:461:ContextThread() ESTABLISH_CONTEXT rv=0x0 for client 15
hw  | 00000179 winscard_svc.c:361:ContextThread() Received command: CMD_GET_READERS_STATE from client 15
hw  | 00000181 winscard_svc.c:361:ContextThread() Received command: CMD_GET_READERS_STATE from client 15
hw  | 00000130 winscard_svc.c:361:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 15
hw  | 00000005 winscard_svc.c:840:MSGSendReaderStates() Send reader states: 15
hw  | 00000015 winscard_svc.c:361:ContextThread() Received command: CMD_STOP_WAITING_READER_STATE_CHANGE from client 15
hw  | 00000002 winscard_svc.c:442:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 15
hw  | 00000118 winscard_msg_srv.c:256:ProcessEventsServer() Common channel packet arrival
hw  | 00000009 winscard_msg_srv.c:267:ProcessEventsServer() ProcessCommonChannelRequest detects: 16
hw  | 00000002 pcscdaemon.c:133:SVCServiceRunLoop() A new context thread creation is requested: 16
hw  | 00000232 winscard_svc.c:340:ContextThread() Authorized PC/SC client
hw  | 00000017 winscard_svc.c:343:ContextThread() Thread is started: dwClientID=16, threadContext @0x5a0553946960
hw  | 00000013 winscard_svc.c:361:ContextThread() Received command: CMD_VERSION from client 16
hw  | 00000002 winscard_svc.c:373:ContextThread() Client is protocol version 4:4
hw  | 00000001 winscard_svc.c:396:ContextThread() CMD_VERSION rv=0x0 for client 16
hw  | 00000174 winscard_svc.c:361:ContextThread() Received command: ESTABLISH_CONTEXT from client 16
hw  | 00000005 winscard.c:215:SCardEstablishContext() Establishing Context: 0x3F07672A
hw  | 00000002 winscard_svc.c:461:ContextThread() ESTABLISH_CONTEXT rv=0x0 for client 16
hw  | 00000098 winscard_svc.c:361:ContextThread() Received command: CONNECT from client 16
hw  | 00000002 winscard_svc.c:499:ContextThread() Authorized client for 'Yubico YubiKey OTP+FIDO+CCID [Pico Key CCID Interface] (E6612483CB1F932D) 00 00'
hw  | 00000001 winscard.c:258:SCardConnect() Attempting Connect to Yubico YubiKey OTP+FIDO+CCID [Pico Key CCID Interface] (E6612483CB1F932D) 00 00 using protocol: 3
hw  | 00000002 readerfactory.c:825:RFReaderInfo() RefReader() count was: 1
hw  | 00000001 winscard.c:352:SCardConnect() powerState: POWER_STATE_IN_USE
hw  | 00000001 winscard.c:430:SCardConnect() Active Protocol: T=1
hw  | 00000002 winscard.c:456:SCardConnect() hCard Identity: 60f2cf9c
hw  | 00000019 winscard.c:518:SCardConnect() UnrefReader() count was: 2
hw  | 00000002 winscard_svc.c:519:ContextThread() CONNECT rv=0x0 for client 16
hw  | 00000118 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 16
hw  | 00000004 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1
hw  | 00000002 winscard.c:1595:SCardTransmit() Send Protocol: T=1
hw  | 00000001 APDU: 00 A4 04 04 08 A0 58 3F C1 9B 7E 4F 21 00 
hw  | 00000002 ifdhandler.c:1332:IFDHTransmitToICC() usb:1050/0407:libudev:0:/dev/bus/usb/001/060 (lun: 0)
hw  | 00058790 SW: 00 01 07 00 90 00 
hw  | 00000019 winscard.c:1648:SCardTransmit() UnrefReader() count was: 2
hw  | 00000006 winscard_svc.c:691:ContextThread() TRANSMIT rv=0x0 for client 16
hw  | 00000327 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 16
hw  | 00000027 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1
hw  | 00000006 winscard.c:1595:SCardTransmit() Send Protocol: T=1
hw  | 00000010 APDU: 00 A4 04 00 0B E8 2B 06 01 04 01 81 C3 1F 02 01 00 
hw  | 00000202 ifdhandler.c:1332:IFDHTransmitToICC() usb:1050/0407:libudev:0:/dev/bus/usb/001/060 (lun: 0)
hw  | 00059981 SW: 62 22 81 02 00 00 82 01 01 83 02 00 00 84 0B 2B 06 01 04 01 81 C3 1F 02 01 49 8A 01 05 85 05 04 01 FF 05 00 90 00 
hw  | 00000020 winscard.c:1648:SCardTransmit() UnrefReader() count was: 2
hw  | 00000008 winscard_svc.c:691:ContextThread() TRANSMIT rv=0x0 for client 16
hw  | 00000506 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 16
hw  | 00000031 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1
hw  | 00000007 winscard.c:1595:SCardTransmit() Send Protocol: T=1
hw  | 00000009 APDU: 00 B1 2F 02 00 00 04 54 02 00 00 00 00 
hw  | 00000007 ifdhandler.c:1332:IFDHTransmitToICC() usb:1050/0407:libudev:0:/dev/bus/usb/001/060 (lun: 0)
[... cut ...]
hw  | 00000023 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1
hw  | 00000005 winscard.c:1595:SCardTransmit() Send Protocol: T=1
hw  | 00000012 APDU: 80 64 3A 03 00 00 30 B8 37 22 69 BE D6 39 12 11 4D E3 C7 A2 C3 02 4D A2 BE 45 60 D4 AF AE F1 A8 F4 3C 27 D2 3B CB C4 A4 46 08 06 FD 6E 3D BB 36 CD 0C 06 92 D5 6F 19 00 00 
hw  | 00000004 ifdhandler.c:1332:IFDHTransmitToICC() usb:1050/0407:libudev:0:/dev/bus/usb/001/060 (lun: 0)

```
this is a log of init + unlock procedure until the freeze

polhenarejos, 2024-12-02:
Here's the problem:
```
hw  | 00000012 APDU: 80 64 3A 03 00 00 30 B8 37 22 69 BE D6 39 12 11 4D E3 C7 A2 C3 02 4D A2 BE 45 60 D4 AF AE F1 A8 F4 3C 27 D2 3B CB C4 A4 46 08 06 FD 6E 3D BB 36 CD 0C 06 92 D5 6F 19 00 00 
```
You are sending an encrypted key of 48 bytes, but it must be 32 bytes. I guess that `HSM_ACCESS_KEY = secrets.token_bytes(32)` does not return 32 bytes, but 48. Check it.

fastchain, 2024-12-03:
@polhenarejos 
According to specs and tests it returns exactly 32 bytes
```
Python 3.12.4 (main, Jun  7 2024, 06:33:07) [GCC 14.1.1 20240522] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import secrets
>>> len(secrets.token_bytes(32))
32
>>> secrets.token_bytes(32)
b'\xfe\x8aZ96R\xd7\xbd:\xa3\x94L\xc7\xa4\x10\xa4\x96\xf7O\xc5\xfb\xeb\x12\x1b\xf5\x84\xf4\xb0E\x9f\xd7<'
>>> 
```
https://docs.python.org/3/library/secrets.html

polhenarejos, 2024-12-03:
There was a buffer overflow in the code and it has been fixed.

You can test it tomorrow with nightly development build or build it yourself.

fastchain, 2024-12-03:
@polhenarejos 

Many thanks! By the way, which versions should be set to build from the nightly with this command

```
sudo docker build \                                                                             
    --build-arg VERSION_PICO_SDK=2.0.0 \
    --build-arg VERSION_MAJOR=5 \
    --build-arg VERSION_MINOR=0 \
    --build-arg PICO_BOARD=pico  \
    --build-arg USB_VID=0x.. \
    --build-arg USB_PID=0x.. \
    -t pico-hsm-builder . --no-cache
  ```
  ?

polhenarejos, 2024-12-03:
Nightly builds are built automatically during the midnight. So you can wait until tomorrow and download a fresh built.

If you prefer not waiting, then you can build it. `development` branch uses Pico SDK 2.1.0.

fastchain, 2024-12-03:
@polhenarejos Thanks. Just tested: initialization with SecureLock and unlock works smoothly. But there is problem with further use. I created a new ticket https://github.com/polhenarejos/pico-hsm/issues/69

commit f88aad1e (2024-12-03): Fixed buffer overflow when unlocking the device.

Fixes #68.
closed 2024-12-03 by fastchain