polhenarejos/pico-hsm #69: CKR_GENERAL_ERROR (0x5) on object read

fastchain, 2024-12-03:
Version: current state of development branch 
Board:  Pico

Pico-hsm  with enabled [SecureLock](https://github.com/polhenarejos/pico-hsm/blob/master/tools/pico-hsm-tool.py#L316) and unlocked.
  
Output on ```/usr/local/bin/pkcs11-tool -O```
```
hw  | Using slot 2 with a present token (0x8)
hw  | Public Key Object; EC  EC_POINT 256 bits
hw  |   EC_POINT:   044104d2ab4fd55170dc931ef40c5e21e2bc74a4ab99fad358ffe5cc580dbe226d1b9de310e0c47903be3b1ea1c9b27977ae9ae2084451d85532b527df21ad13ca8c5b
hw  |   EC_PARAMS:  06082a8648ce3d030107 (OID 1.2.840.10045.3.1.7)
hw  |   label:      ESPICOHSMTR
hw  |   ID:         0000000000000000000000000000000000000000
hw  |   Usage:      verify, derive
hw  |   Access:     none
hw  | Public Key Object; EC  EC_POINT 256 bits
hw  |   EC_POINT:   04410420b871f3ced029e14472ec4ebc3c0448164942b123aa6af91a3386c1c403e0ebd3b4a5752a2b6c49e574619e6aa0549eb9ccd036b9bbc507e1f7f9712a236092
hw  |   EC_PARAMS:  06052b8104000a (OID 1.3.132.0.10)
hw  |   label:      
hw  |   ID:         01
hw  |   Usage:      verify, derive
hw  |   Access:     none
hw  | Profile object 1949634128
hw  |   profile_id:          CKP_PUBLIC_CERTIFICATES_TOKEN (4)

```

when I try to read public key with 
```
pkcs11-tool --read-object --pin 3760328958 --id 1 --type pubkey > tmp/1pub.der
```
I get this (log with APDU commands)

```
hw  | pkcs11-tool --read-object --pin 3760328958 --id 1 --type pubkey > tmp/1pub.der
hw  | 00941928 ifdwrapper.c:477:IFDControl() Card not transacted: 606
hw  | 00000104 ifdwrapper.c:477:IFDControl() Card not transacted: 606
hw  | 00006039 APDU: 00 A4 04 00 07 62 76 01 FF 00 00 00 
hw  | 00000500 SW: 6A 82 
hw  | 00000052 APDU: 00 A4 04 00 06 A0 00 00 00 01 01 
hw  | 00000280 SW: 6A 82 
hw  | 00000042 APDU: 00 A4 04 00 0B E8 2B 06 01 04 01 81 C3 1F 02 01 00 
hw  | 00060565 SW: 62 22 81 02 00 00 82 01 01 83 02 00 00 84 0B 2B 06 01 04 01 81 C3 1F 02 01 49 8A 01 05 85 05 04 01 FF 05 00 90 00 
hw  | 00000310 APDU: 00 A4 08 00 02 2F 00 00 
hw  | 00000577 SW: 62 0E 81 02 00 19 82 01 01 83 02 2F 00 8A 01 05 90 00 
hw  | 00000457 APDU: 00 B1 00 00 04 54 02 00 00 19 
hw  | 00000493 SW: 61 17 4F 0B E8 2B 06 01 04 01 81 C3 1F 02 01 50 08 50 69 63 6F 2D 48 53 4D 90 00 
hw  | 00000278 APDU: 00 A4 00 00 02 2F 02 00 
hw  | 00000352 SW: 62 0E 81 02 03 AC 82 01 01 83 02 2F 02 8A 01 05 90 00 
hw  | 00000072 APDU: 00 B1 00 00 00 00 04 54 02 00 00 04 00 
hw  | 00002522 SW: 67 82 01 ED 7F 21 82 01 93 7F 4E 82 01 4B 5F 29 01 00 42 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 7F 49 82 01 1D 06 0A 04 00 7F 00 07 02 02 02 02 03 81 20 FF FF FF FF 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF 82 20 FF FF FF FF 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FC 83 20 5A C6 35 D8 AA 3A 93 E7 B3 EB BD 55 76 98 86 BC 65 1D 06 B0 CC 53 B0 F6 3B CE 3C 3E 27 D2 60 4B 84 41 04 6B 17 D1 F2 E1 2C 42 47 F8 BC E6 E5 63 A4 40 F2 77 03 7D 81 2D EB 33 A0 F4 A1 39 45 D8 98 C2 96 4F E3 42 E2 FE 1A 7F 9B 8E E7 EB 4A 7C 0F 9E 16 2B CE 33 57 6B 31 5E CE CB B6 40 68 37 BF 51 F5 85 20 FF FF FF FF 00 00 00 00 FF FF FF FF FF FF FF FF BC E6 FA AD A7 17 9E 84 F3 B9 CA C2 FC 63 25 51 86 41 04 D2 AB 4F D5 51 70 DC 93 1E F4 0C 5E 21 E2 BC 74 A4 AB 99 FA D3 58 FF E5 CC 58 0D BE 22 6D 1B 9D E3 10 E0 C4 79 03 BE 3B 1E A1 C9 B2 79 77 AE 9A E2 08 44 51 D8 55 32 B5 27 DF 21 AD 13 CA 8C 5B 87 01 01 5F 20 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 5F 37 40 B4 60 AD 65 46 D8 AA FF F0 67 EC 76 56 21 AF 84 B0 4C 8E A8 7B 3B F6 1E 05 2F C7 67 72 EC 6F 54 7A DD 9B CE EB 92 35 2F 40 70 3A C5 00 47 B8 D6 E8 E4 FE 64 16 BE 81 6F 3F EE BA AB 07 85 47 89 42 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 5F 37 40 78 06 15 48 04 4E E6 0D B9 CA 7C 2F 5C BF 23 4A E8 FE DE 33 4C FB 58 82 16 C9 3A 8C 67 6C 31 15 80 75 53 0E DF 10 44 3B 8E E8 B2 0F 91 8B 5B F1 3D 01 A8 19 CF 5E 6F 02 0E F2 13 5B 7B B8 FE 4C 7F 21 82 01 B6 7F 4E 82 01 6E 5F 29 01 00 42 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 7F 49 82 01 1D 06 0A 04 00 7F 00 07 02 02 02 02 03 81 20 FF FF FF FF 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF 82 20 FF FF FF FF 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FC 83 20 5A C6 35 D8 AA 3A 93 E7 B3 EB BD 55 76 98 86 BC 65 1D 06 B0 CC 53 B0 F6 3B CE 3C 3E 27 D2 60 4B 84 41 04 6B 17 D1 F2 E1 2C 42 47 F8 BC E6 E5 63 A4 40 F2 77 03 7D 81 2D EB 33 A0 F4 A1 39 45 D8 98 C2 96 4F E3 42 E2 FE 1A 7F 9B 8E E7 EB 4A 7C 0F 9E 16 2B CE 33 57 6B 31 5E CE CB B6 40 68 37 BF 51 F5 85 20 FF FF FF FF 00 00 00 00 FF FF FF FF FF FF FF FF BC E6 FA AD A7 17 9E 84 F3 B9 CA C2 FC 63 25 51 86 41 04 D2 AB 4F D5 51 70 DC 93 1E F4 0C 5E 21 E2 BC 74 A4 AB 99 FA D3 58 FF E5 CC 58 0D BE 22 6D 1B 9D E3 10 E0 C4 79 03 BE 3B 1E A1 C9 B2 79 77 AE 9A E2 08 44 51 D8 55 32 B5 27 DF 21 AD 13 CA 8C 5B 87 01 01 5F 20 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 7F 4C 0E 06 09 04 00 7F 00 07 03 01 02 02 53 01 00 5F 25 06 02 03 00 03 02 01 5F 24 06 07 00 01 02 03 01 5F 37 40 0C EB C6 5F 63 1B 52 1B 34 EC 61 BB 10 50 71 E8 0A F6 54 B5 E6 06 49 9F D2 8F 2E 6C EA 1D AC F7 07 F9 F3 08 EE 3E 91 C8 BF 9B 32 B6 80 F6 B7 7A 5D AA 35 61 D3 CC 90 C6 10 FD E8 32 39 67 B4 3B 90 00 
hw  | 00000109 APDU: 00 B1 00 00 04 54 02 03 AC 54 
hw  | 00000340 SW: 90 00 
hw  | 00000021 APDU: 00 A4 00 00 02 2F 03 00 
hw  | 00000310 SW: 62 0E 81 02 00 2C 82 01 01 83 02 2F 03 8A 01 05 90 00 
hw  | 00000009 APDU: 00 B1 00 00 00 00 04 54 02 00 00 02 00 
hw  | 00000371 SW: 30 2A 02 01 05 04 08 E6 61 24 83 CB 1F 93 2D 0C 0D 50 6F 6C 20 48 65 6E 61 72 65 6A 6F 73 80 08 50 69 63 6F 2D 48 53 4D 03 02 04 30 90 00 
hw  | 00000011 APDU: 00 B1 00 00 00 00 04 54 02 00 2C 01 D4 
hw  | 00000462 SW: 90 00 
hw  | 00000021 APDU: 00 20 00 81 
hw  | 00000288 SW: 63 C3 
hw  | 00000013 APDU: 00 20 00 88 
hw  | 00000313 SW: 63 CF 
hw  | 00000010 APDU: 00 20 00 85 
hw  | 00000355 SW: 90 00 
hw  | 00000071 APDU: 80 58 00 00 00 00 00 
hw  | 00000348 SW: C4 00 CC 00 CC 01 C4 01 90 00 
hw  | 00000064 APDU: 00 A4 00 00 02 C4 00 00 
hw  | 00000559 SW: 62 0E 81 02 00 3A 82 01 08 83 02 C4 00 8A 01 05 90 00 
hw  | 00000020 APDU: 00 B1 00 00 00 00 04 54 02 00 00 10 00 
[... 81 lines cut ...]
hw  | 00000383 SW: 62 0E 81 02 01 B2 82 01 01 83 02 CE 01 8A 01 05 90 00 
hw  | 00000182 APDU: 00 B1 00 00 00 00 04 54 02 00 00 10 00 
hw  | 00001428 SW: 67 82 01 AE 7F 21 82 01 54 7F 4E 82 01 0C 5F 29 01 00 42 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 7F 49 81 DF 06 0A 04 00 7F 00 07 02 02 02 02 03 81 20 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FE FF FF FC 2F 82 01 00 83 01 07 84 41 04 79 BE 66 7E F9 DC BB AC 55 A0 62 95 CE 87 0B 07 02 9B FC DB 2D CE 28 D9 59 F2 81 5B 16 F8 17 98 48 3A DA 77 26 A3 C4 65 5D A4 FB FC 0E 11 08 A8 FD 17 B4 48 A6 85 54 19 9C 47 D0 8F FB 10 D4 B8 85 20 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FE BA AE DC E6 AF 48 A0 3B BF D2 5E 8C D0 36 41 41 86 41 04 20 B8 71 F3 CE D0 29 E1 44 72 EC 4E BC 3C 04 48 16 49 42 B1 23 AA 6A F9 1A 33 86 C1 C4 03 E0 EB D3 B4 A5 75 2A 2B 6C 49 E5 74 61 9E 6A A0 54 9E B9 CC D0 36 B9 BB C5 07 E1 F7 F9 71 2A 23 60 92 87 01 01 5F 20 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 5F 37 40 70 9D A6 F8 D6 B3 72 95 E7 C4 F3 F7 97 46 42 5C D5 9A 66 80 D5 AF B5 F0 77 8A 38 B0 23 63 2A 58 28 F7 17 4F 5B E7 97 55 5C EF 29 82 4C FE AF 26 53 66 C5 21 86 76 57 DE FD 99 0F E9 BF 18 34 3B 42 10 45 53 50 49 43 4F 48 53 4D 54 52 30 30 30 30 31 5F 37 40 CB BE 2D 03 39 6F F6 17 60 07 39 8F 01 DC 8F 71 57 16 41 BC 43 9F EB C8 47 61 86 F1 72 D7 AB 21 3B 3C E1 8A 5C 3C 50 BE 8D 2F 21 15 32 DA 8A 9D DD 76 91 2B AB FF AC 33 B3 A1 02 55 85 C7 66 C0 90 00 
hw  | 00000300 APDU: 00 B1 00 00 00 00 04 54 02 01 B2 0E 4E 
hw  | 00000372 SW: 90 00 
hw  | Using slot 3 with a present token (0xc)
hw  | 00036790 APDU: 00 20 00 81 
hw  | 00000542 SW: 63 C3 
hw  | 00000434 APDU: 00 20 00 81 
hw  | 00000438 SW: 63 C3 
hw  | 00000217 APDU: 00 20 00 81 0A 33 37 36 30 33 32 38 39 35 38 
hw  | 00084738 SW: 90 00 
hw  | Using signature algorithm ECDSA
hw  | 00000911 APDU: 80 68 01 70 00 00 20 33 F4 34 0C E7 2D CB 30 E5 E9 11 49 F6 22 C5 CA 5A A5 8C E8 FF A3 6A 75 8D 31 2C 72 73 9A CD 71 02 00 
hw  | 00001648 SW: 64 00 
hw  | 00000139 APDU: 80 68 01 70 00 00 20 33 F4 34 0C E7 2D CB 30 E5 E9 11 49 F6 22 C5 CA 5A A5 8C E8 FF A3 6A 75 8D 31 2C 72 73 9A CD 71 02 00 
hw  | 00001047 SW: 64 00 
hw  | 00000204 APDU: 80 68 01 70 00 00 20 33 F4 34 0C E7 2D CB 30 E5 E9 11 49 F6 22 C5 CA 5A A5 8C E8 FF A3 6A 75 8D 31 2C 72 73 9A CD 71 02 00 
hw  | 00001099 SW: 64 00 
hw  | 00000071 APDU: 80 68 01 70 00 00 20 33 F4 34 0C E7 2D CB 30 E5 E9 11 49 F6 22 C5 CA 5A A5 8C E8 FF A3 6A 75 8D 31 2C 72 73 9A CD 71 02 00 
hw  | 00000819 SW: 64 00 
hw  | error: PKCS11 function C_SignFinal failed: rv = CKR_GENERAL_ERROR (0x5)
hw  | Aborting.

```

polhenarejos, 2024-12-03:
How is the key generated?

fastchain, 2024-12-04:
@polhenarejos 

> How is the key generated?

It was imported following the method described in this [comment](https://github.com/polhenarejos/pico-hsm/issues/68#issuecomment-2507476113)

Here is key itself, if needed
```
7c852118294e51e653712a81e05800f419141751be58f605c371e15141b007a6
```

> Edit: seems an outdated version of OpenSC. Try to use version 0.26

didn't help.

polhenarejos, 2024-12-04:
I am using this script and it works:
```
from picohsm import PicoHSM
from cryptography.hazmat.primitives.asymmetric import ec
from binascii import unhexlify

curve=ec.SECP256K1
secret_key=unhexlify('7c852118294e51e653712a81e05800f419141751be58f605c371e15141b007a6')
pkey = ec.derive_private_key(
    int.from_bytes(secret_key, byteorder='big'),
    curve(),  # Curve used in Ethereum
)
DEFAULT_DKEK = bytes([0x1] * 32)
print(pkey)
device = PicoHSM()
device.initialize(dkek_shares=1)

device.import_dkek(DEFAULT_DKEK)
key_id = device.import_key(pkey, dkek=DEFAULT_DKEK)
print(key_id)
pubkey = device.public_key(key_id, param=curve().name)
print(pubkey)
```

fastchain, 2024-12-04:
@polhenarejos interesting. 
Was the  [SecureLock](https://github.com/polhenarejos/pico-hsm/blob/master/tools/pico-hsm-tool.py#L316)  enabled on init and the and unlocked before read?

polhenarejos, 2024-12-04:
No, it wasn't. Can you try the snippet? To isolate the problem. Perhaps it's the securelock.

polhenarejos, 2024-12-04:
I updated the script with SecureLock2 and still works. Can you try it? It is the same as the other comment.
```python
from picohsm import PicoHSM
from cryptography.hazmat.primitives.asymmetric import ec
from binascii import unhexlify
from cryptography.hazmat.primitives.kdf.hkdf import HKDF
from cryptography.hazmat.primitives import hashes
from cryptography.hazmat.primitives.ciphers.aead import ChaCha20Poly1305
from cryptography.hazmat.primitives.serialization import Encoding, PublicFormat
import platform
import sys

class SecureLock2:
    def __init__(self, picohsm, secretkey):
        self.picohsm = picohsm
        self.secretkey = secretkey

    def mse(self):
        sk = ec.generate_private_key(ec.SECP256R1())
        pn = sk.public_key().public_numbers()
        self.__pb = sk.public_key().public_bytes(Encoding.X962, PublicFormat.UncompressedPoint)

        ret = self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x01, data=list(self.__pb))

        pk = ec.EllipticCurvePublicKey.from_encoded_point(ec.SECP256R1(), bytes(ret))
        shared_key = sk.exchange(ec.ECDH(), pk)

        xkdf = HKDF(
            algorithm=hashes.SHA256(),
            length=12+32,
            salt=None,
            info=self.__pb
        )
        kdf_out = xkdf.derive(shared_key)
        self.__key_enc = kdf_out[12:]
        self.__iv = kdf_out[:12]

    def encrypt_chacha(self, data):
        chacha = ChaCha20Poly1305(self.__key_enc)
        ct = chacha.encrypt(self.__iv, data, self.__pb)
        return ct

    def unlock_device(self):

        ct = self.get_skey()

        self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x03, data=list(ct))

    def _get_key_device(self):
        return self.secretkey

    def get_skey(self):
        self.mse()
        ct = self.encrypt_chacha(self._get_key_device())
        return ct

    def enable_device_aut(self):
        ct = self.get_skey()
        self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x02, data=list(ct))

    def disable_device_aut(self):
        ct = self.get_skey()
        self.picohsm.send(cla=0x80, command=0x64, p1=0x3A, p2=0x04, p3=list(ct))

curve=ec.SECP256K1
secret_key=unhexlify('7c852118294e51e653712a81e05800f419141751be58f605c371e15141b007a6')
pkey = ec.derive_private_key(
    int.from_bytes(secret_key, byteorder='big'),
    curve(),  # Curve used in Ethereum
)
DEFAULT_DKEK = bytes([0x1] * 32)
print(pkey)
device = PicoHSM()
device.initialize(dkek_shares=1)

device.import_dkek(DEFAULT_DKEK)
key_id = device.import_key(pkey, dkek=DEFAULT_DKEK)
print(key_id)
pubkey = device.public_key(key_id, param=curve().name)
print(pubkey)
slck = SecureLock2(device,secret_key)
slck.enable_device_aut()
slck.unlock_device()
pubkey = device.public_key(key_id, param=curve().name)
print(pubkey)
```

fastchain, 2024-12-06:
Hi! 
I executed this Python code without encountering any exceptions, but the pkcs11-tool still returned the same error.   
Here is output for 
```
OPENSC_DEBUG=9 pkcs11-tool --read-object --pin 648219 --id 31 --type pubkey
```
[pkcs11-tool.output.log](https://github.com/user-attachments/files/18033741/pkcs11-tool.output.log)

btw, what version of pkcs11-tool do you use?

fastchain, 2024-12-06:
Hi! 
I executed this Python code without encountering any exceptions, but the pkcs11-tool still returned the same error.   
Here is output for 
```
OPENSC_DEBUG=9 pkcs11-tool --read-object --pin 648219 --id 31 --type pubkey
```
[pkcs11-tool.output.log](https://github.com/user-attachments/files/18033741/pkcs11-tool.output.log)

what version of pkcs11-tool do you use? 

Btw, when I do [RAW ECDSA](https://github.com/polhenarejos/pypicohsm/blob/203777f318ff7818a4efc97283e7c3a0a73dd5a5/picohsm/Algorithm.py#L28) 

```
....
signature=device.sign(keyid=key_id,data=secret_key,scheme=0x70)
```
At the end of your code, I get this

```
Traceback (most recent call last):
  File "/hw/test_sign.py", line 85, in <module>
    signature=device.sign(keyid=key_id,data=secret_key,scheme=0x70)
  File "/pypicohsm/picohsm/PicoHSM.py", line 470, in sign
    resp = self.send(cla=0x80, command=0x68, p1=keyid, p2=scheme or 0x00, data=data)
  File "/pypicohsm/picohsm/PicoHSM.py", line 277, in send
    raise APDUResponse(sw1, sw2)
picohsm.APDU.APDUResponse: SW:6400
```

what am I doing wrong?

polhenarejos, 2024-12-06:
I use OpenSC v0.26

EC_RAW requires data previously hashed. The used hash is derived from the length of data. Is the data 32 bytes length?

fastchain, 2024-12-06:
@polhenarejos 
I use secret_key  variable  as data-to-be-signed, as far as I can see it's 256 bits (32 bytes).

Just updated to OpenSC v0.26, with no luck.  Btw, does your script  interact with Pico from OS (which one do you use?) directly  or from docker?

polhenarejos, 2024-12-06:
After inspecting the log, I see the problem but not the cause. Your PCSC is disconnecting the session once it recovers all keys and then try log in. But since it has been disconnected, log in fails and raises the error.
Try without PIN:
```
OPENSC_DEBUG=9 pkcs11-tool --read-object --id 31 --type pubkey
```
and with login first
```
OPENSC_DEBUG=9 pkcs11-tool --login --pin 648219 --read-object --id 31 --type pubkey
```
I guess it won't be any difference, but let's try.

About signature, I confirm it fails. I'll check why later.

polhenarejos, 2024-12-08:
Can you try latest nightly development build? It should be fixed. You would need to nuke it first.

fastchain, 2024-12-14:
@polhenarejos 

> Can you try latest nightly development build? It should be fixed. You would need to nuke it first.

Patched fixed  [this issue](https://github.com/polhenarejos/pico-hsm/issues/69#issuecomment-2522344973). 

>OPENSC_DEBUG=9 pkcs11-tool --read-object --id 31 --type pubkey

[nopin.txt](https://github.com/user-attachments/files/18134976/nopin.txt)

works fine

>OPENSC_DEBUG=9 pkcs11-tool --login --pin 123456 --read-object --id 31 --type pubkey

[withpin.txt](https://github.com/user-attachments/files/18134972/withpin.txt)

same error.  (key was initialized with pin 123456, so here is valid pin)

btw, during signing signing, when pin is important, I get the same error.

polhenarejos, 2024-12-14:
There's always a default PIN (648219). 
Can you try the second command with with `648219` instead of `123456`? Perhaps it is initialized badly.

commit 3d960b93 (2024-12-08): Fix MKEK masking order. Fixes #69.