polhenarejos/pico-hsm #72: SecureLock unlock is needed  for reinitialization

fastchain, 2024-12-18:
Hello,
When SecureLock feature is enabled and device is in locked state,  device.initialize(...) returns SW:63C1. 

How to reproduce:

1. Init a device
2. Enable SecureLock and unlock device
3. Disconnect device
4. Init a device

On step 4 you should get 

```
hw  |   File "/pypicohsm/picohsm/PicoHSM.py", line 360, in initialize
hw  |     self.send(cla=0x80, command=0x50, data=data)
hw  |   File "/pypicohsm/picohsm/PicoHSM.py", line 297, in send
hw  |     raise APDUResponse(sw1, sw2)
hw  | picohsm.APDU.APDUResponse: SW:63C1
```

The only  fix I found  is to erase device with the nuke firmware and then upload pico-hsm firmware and init.

If device is unlocked before step 4 , reinitialization  goes smoothly.

Board: pico
pico-hsm version: development branch

polhenarejos, 2024-12-18:
This message corresponds to the number of retries missing before locking, in this case 1 retry more. What is strange is getting this error when you send the INIT command.

fastchain, 2024-12-19:
@polhenarejos 

> Failing is expected, but it should return 0x6982.

Am I correct in understanding that initialization should fail if the key is locked? I initially thought it was a bug, as I assumed reinitialization should still be possible even if the device is locked. Is there any documentation worth reading about this kind of logic ?

fastchain, 2024-12-23:
@polhenarejos 
I see, thank you for the explanation. 
Is there a way to reinitialize the device without reflashing it if SecureLock is enabled and the SecureLock key is lost?

polhenarejos, 2024-12-23:
The fastest is to nuke it.

closed 2024-12-24 by fastchain