polhenarejos/pico-hsm #73: pico2 ecdsa download does not install

juliadin, 2025-01-01:
I am trying to install the EdDSA release of PicoHSM to an authentic RPi Pico2 module but the module won't flash it after copying.
The non-EdDSA release works just fine

SHA256 of the downloads:
```
e3f0aa5fbd1b186be812c86b205778ddd5bf5510d936156b42818d97bae411cd  Downloads/pico_hsm_pico2-5.0-eddsa1.uf2
b1dcaab4495bbe4aff24290b9cbd7ea3be99eaad5bf28ae35c0526c0dd776e3f  Downloads/pico_hsm_pico2-5.0.uf2
```

While trying to build the modules myself I failed to find instructions about the EdDSA release - also the release build tool doesn't seem to build them and there are no nightly or development builds to try a few things. 

Since ed25519 and ed448 are listed on the main feature page, is it supported by the official builds at this time?

I must admit I am a bit confused :)

polhenarejos, 2025-01-01:
Can you describe what are you experiencing and what brings you to say “the module won't flash it after copying.”? I don’t understand this sentence. 

The EdDSA release uses a fork of the official MbedTLS that introduces EdDSA. I sent a PR to ARM but it is under review since 3 years. 
Since it is no fully approved, I leave this fork as a beta and thus, as a separate release. There’s no difference between main and edda branches, except for EdDSA support. 

The core is exactly the same, so both should boot normally. 

Steps to see whether is recognized are
1. Use `lsusb` to see whether is found by OS. 
2. Use `opensc-tool -an` to see whether is recognized by PSCS. Note that it has to be previously commissioned with a valid VID/PID. 

If both work, then it’s a problem with LED gpio number.

juliadin, 2025-01-01:
What I meant with that was, when in BOOTSEL mode, i can copy the non-eddsa image to the module just fine, it disconnects and flashes it, disconnects and boots the flashed application.
When I do that with the eddsa build, nothing happens (i.e. it doesn't disconnect, it doesn't reboot, it just sits there in BOOTSEL mode waiting for an image.

I went to the parts drawer though and it seems to be a problem with the Pico2 since another one works.

Anyway, since I have a few less vanilla RP2350 and RP2040 boards here that I would like to write header files for to have them supported: Is there more to building the EdDSA version than checking out the eddsa branch, reinitializing all the submodules so the correct mbedtls is used and building?

polhenarejos, 2025-01-01:
It's hard to say. I do not own a Pico2 but I tried with my Tiny2040 and Tiny2350 and both worked with EdDSA release.

You do not need nothing special for EdDSA branch. Just clone, checkout that branch, build and flash.

closed 2025-01-15 by polhenarejos