polhenarejos/pico-hsm #74: PicoHSM Slot Behavior Inconsistency Between Ubuntu and Windows

Otmane65, 2025-01-15:
When using PicoHSM on Ubuntu, I've encountered several inconsistencies compared to Windows behavior:

## Slot Detection
- On Ubuntu: PicoHSM appears as two slots
- On Windows: PicoHSM appears as a single slot
[image]

## Initialization and Login Issues
- Initial setup works correctly
- Direct token generation fails
## Login behavior:

- Fails when using slot 0
- Works when explicitly specifying slot ID 4
[image]

## Impact on OpenSSL Operations
The slot ID discrepancy (slot 4 vs slot 0) causes issues with subsequent OpenSSL operations, affecting the overall workflow.

polhenarejos, 2025-01-15:
Can you try with v5.2?

Otmane65, 2025-01-16:
yes the same issue with v5.2

polhenarejos, 2025-01-16:
Which OpenSC version you use? I got different outputs. I'm using v0.26. Check it and upgrade it if needed.

Otmane65, 2025-01-16:
I have successfully upgraded OpenSC to version 0.26 by compiling it manually. However, I am still facing the same issue. I
[image]

polhenarejos, 2025-01-16:
No, it's not. Only ESP32-S3 is supported.

Can you `opensc-tool -l` ?

Disappear9, 2025-01-16:
Same issue here
```
Ubuntu 22.04
OpenSC 0.26.1 manually compiled
pico_hsm_pico-5.2.uf2
```

$ pkcs11-tool -L
```
Available slots:
Slot 0 (0x0): Nitrokey Nitrokey HSM [Pico Key CCID OTP FIDO Interfac] (E660...
  token label        : Pico-HSM
  token manufacturer : Pol Henarejos
  token model        : PKCS#15 emulated
  token flags        : login required, PIN pad present, rng, token initialized, PIN initialized
  hardware version   : 24.13
  firmware version   : 5.2
  serial num         : ESPICOHSMTR
  pin min/max        : 6/15
  uri                : pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM
Slot 1 (0x4): Nitrokey Nitrokey HSM [Pico Key WebCCID Interface] (E660...
  token label        : Pico-HSM
  token manufacturer : Pol Henarejos
  token model        : PKCS#15 emulated
  token flags        : login required, PIN pad present, rng, token initialized, PIN initialized
  hardware version   : 24.13
  firmware version   : 5.2
  serial num         : ESPICOHSMTR
  pin min/max        : 6/15
  uri                : pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM

```

```
$ pkcs11-tool --login --pin 648219 --test --slot 0
error: PKCS11 function C_Login failed: rv = CKR_GENERAL_ERROR (0x5)
Aborting.
$ pkcs11-tool --login --pin 648219 --test --slot 4
C_SeedRandom() and C_GenerateRandom():
  seeding (C_SeedRandom) not supported
  seems to be OK
Digests:
  all 4 digest functions seem to work
  MD5: OK
  RIPEMD160: OK
  SHA-1: OK
  SHA256: OK
Ciphers: not implemented
Signatures (currently only for RSA)
  testing key 0 (root)  -- non-RSA, skipping
  testing key 1 (intermediate)  -- non-RSA, skipping
Signatures: no private key found in this slot
Verify (currently only for RSA)
  testing key 0 (root) -- non-RSA, skipping
  testing key 1 (intermediate) with 1 mechanism -- non-RSA, skipping
Decryption (currently only for RSA)
  testing key 0 (root) -- non-RSA, skipping
  testing key 1 (intermediate) -- non-RSA, skipping
No errors
```

$ opensc-tool -l
```
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Nitrokey Nitrokey HSM [Pico Key CCID OTP FIDO Interfac] (E66XXXXXX) 00 00
1    Yes             Nitrokey Nitrokey HSM [Pico Key WebCCID Interface] (E66XXXXXX) 01 00
```

polhenarejos, 2025-01-17:
Seems a bug with CCID 1.5 and above. It looks through the different interfaces and try loading them if the descriptor is a Smart Card class. CCID and WebCCID are Smart Card descriptors, so both are loaded. However, since both belong to the same, only the last is opened and this is why you can communicate only with slot 4 (webCCID).

I pushed a tweak by changing maxSlot of WebCCID to a wrong value to cause a timeout in LibCCID and deactivate this interface, leaving only 1. This still works with WebUSB Commissioner since it does not check maxSlot value.

Can you try with the development branch? A [nightly build](https://github.com/polhenarejos/pico-hsm/releases/tag/nightly-development) will be available tomorrow if you don't want to build it.

Disappear9, 2025-01-18:
i nuked the flash & reflashed nightly build `pico_hsm_pico-5.0.uf2` still the same.

i paste logs to gist to prevent someone watching this repo get a extremely long email:

$ OPENSC_DEBUG=9 pkcs11-tool --login --pin 648219 --test --slot 0
https://gist.github.com/Disappear9/d4aea5e643637f91aab921c6eeee5a73#file-slot0-log

$ OPENSC_DEBUG=9 pkcs11-tool --login --pin 648219 --test --slot 4
https://gist.github.com/Disappear9/d4aea5e643637f91aab921c6eeee5a73#file-slot4-log

polhenarejos, 2025-01-18:
You have to use v5.2 from the development nightly build.

Disappear9, 2025-01-18:
i have tested v5.2 nightly build for a while, all function seems working fine, just need to wait more time for that timeout.
and why `pico-hsm-tool.py phy wcid disable` didn't disable WebCCID interface?

polhenarejos, 2025-01-18:
I forgot to remove it from the tool. WebCCID is used by Pico Commissioner through WebUSB, so it cannot be disables anymore. 
The thing is that this is a bug fron LibCCID that happens on newer versions. For instance, it doesn’t happen in macOS. So apparently the linux user has to decide between the slot timeout or Pico Commissioner.

geekman, 2025-02-11:
I'm also facing the same problem, as multiple slots/cards are appearing. 

I have downloaded the latest nightly (seems like it was built 12 hours ago) and flashed it, but instead of reducing to a single slot, another slot appeared:

```
$ opensc-tool -l
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             PicoHSM [Pico Key CCID OTP FIDO Interfac] (E663xxxx) 00 00
1    Yes             PicoHSM [Pico Key WebCCID Interface] (E663xxxx) 01 00
2    Yes             PicoHSM [Pico Key WebCCID Interface] (E663xxxx) 01 01
```

opensc is version 0.21, on Debian 11. Let me know what other informatin is required.

polhenarejos, 2025-02-11:
It was corrected in the development nightly build. Are you using stable or development?

polhenarejos, 2025-02-11:
I’d need the log of pcscd. 

1. Terminate pcscd service. 
2. Execute `sudo LIBCCID_ifdLogLevel=0x000F pcscd --foreground --debug --apdu --color | tee -i log.txt`
3. Wait for 5-10 seconds and kill it by ctrl-c. 
4. Upload log.txt file.

geekman, 2025-02-12:
I've gathered the log from pcscd as requested. I kept the original PID/VID and added it to `/etc/libccid_Info.plist` to get pcscd to recognize the device and use the generic driver.

[pcscd-log.txt](https://github.com/user-attachments/files/18763249/pcscd-log.txt)

polhenarejos, 2025-02-12:
I pushed some changes in the development branch. Can you check them?

geekman, 2025-02-12:
Looks like it's working now. It's down to 1 card only. Thanks!

commit 8eb4669c (2025-01-17): Fixes #74.

LibCCID allows propietary class for CCID (ALLOW_PROPIETARY_CLASS in LibCCID). Then both interfaces are loaded but since both belong to same device, only the last is opened.

We change maxSlot value even it breaks the ICCD spec to cause a timeout in LibCCID and deactivate WebCCID interface.
commit 0f3bb824 (2025-02-12): Fix multiple interfaces in older PCSC versions.

Fixes #74 again.
commit 64d8dafa (2025-02-12): Fix multiple interfaces in older PCSC versions.

Fixes #74 again.
closed 2025-02-19 by polhenarejos