polhenarejos/pico-hsm #75: pico-hsm-tool.py tough to build on linux?

nimbius, 2025-01-25:
Hi,
thanks for working on pihohsm, its pretty cool!

the pico-hsm-tool.py has a dependency on pycvc, which doesnt exist in debian, ubuntu, or arch repos.  building it from pypi is discouraged by most
major distros that centrally manage their python libs.  

is it possible to add this to releases somehow?  is there an equivalent to pycvc that can be used?

polhenarejos, 2025-01-25:
Pycvc is made by me. Why you cannot just `pip install pycvc`?

nimbius, 2025-02-09:
Pip install in most major distros comes with a warning and error...Ubuntu for example really seems to hate pip outside its apt controlled ecosystem.

I guess you could build in venv though?

jouellnyc, 2025-04-23:
$~/gitrepos/pico-hsm/tools$ python3 -m venv ~/gitrepos/myhsmvenv
$ source  ~/gitrepos/myhsmvenv/bin/activate
(myhsmvenv) :~/gitrepos/pico-hsm/tools$ python3 pico-hsm-tool.py initialize
ERROR: cvc module not found! Install pycvc package.
Try with `pip install pycvc`
(myhsmvenv) john@user-OptiPlex-7050:~/gitrepos/pico-hsm/tools$ pip install pycvc
<snip>
Successfully installed cffi-1.17.1 cryptography-44.0.2 pycparser-2.22 pycvc-1.5.0 setuptools-79.0.0
(myhsmvenv) :~/gitrepos/pico-hsm/tools$ python3 pico-hsm-tool.py initialize
ERROR: picohsm module not found! Install picohsm package.
Try with `pip install pypicohsm`

(myhsmvenv) :~/gitrepos/pico-hsm/tools$ pip install pypicohsm
Collecting pypicohsm
  Using cached pypicohsm-1.4-py3-none-any.whl.metadata (2.0 kB)
Requirement already satisfied: setuptools in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from pypicohsm) (79.0.0)
Requirement already satisfied: cryptography>=3.3 in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from pypicohsm) (44.0.2)
Collecting base58 (from pypicohsm)
  Using cached base58-2.1.1-py3-none-any.whl.metadata (3.1 kB)
Collecting pyusb (from pypicohsm)
  Using cached pyusb-1.3.1-py3-none-any.whl.metadata (2.5 kB)
Requirement already satisfied: pycvc in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from pypicohsm) (1.5.0)
Collecting pyscard (from pypicohsm)
  Using cached pyscard-2.2.2.tar.gz (156 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: cffi>=1.12 in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from cryptography>=3.3->pypicohsm) (1.17.1)
Requirement already satisfied: pycparser in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from cffi>=1.12->cryptography>=3.3->pypicohsm) (2.22)
Using cached pypicohsm-1.4-py3-none-any.whl (22 kB)
Using cached base58-2.1.1-py3-none-any.whl (5.6 kB)
Using cached pyusb-1.3.1-py3-none-any.whl (58 kB)
Building wheels for collected packages: pyscard
  Building wheel for pyscard (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pyscard (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      Package libpcsclite was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libpcsclite.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'libpcsclite', required by 'virtual:world', not found
      /tmp/pip-build-env-branwe8o/overlay/lib/python3.12/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!
      
              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:
      
              License :: OSI Approved :: GNU Lesser General Public License v2 or later (LGPLv2+)
      
              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************
      
      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      running build_py
      running build_ext
      building 'smartcard.scard._scard' extension
      swigging src/smartcard/scard/scard.i to src/smartcard/scard/scard_wrap.c
      swig -python -outdir src/smartcard/scard -DPCSCLITE -o src/smartcard/scard/scard_wrap.c src/smartcard/scard/scard.i
      creating build/temp.linux-x86_64-cpython-312/src/smartcard/scard
      x86_64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -DVER_PRODUCTVERSION=2,2,2,0000 -DVER_PRODUCTVERSION_STR=2.2.2 -DPCSCLITE=1 -Isrc/smartcard/scard/ -I/usr/include/PCSC -I/usr/local/include/PCSC -I/home/john/gitrepos/myhsmvenv/include -I/usr/include/python3.12 -c src/smartcard/scard/helpers.c -o build/temp.linux-x86_64-cpython-312/src/smartcard/scard/helpers.o
      src/smartcard/scard/helpers.c:28:10: fatal error: winscard.h: No such file or directory
         28 | #include <winscard.h>
            |          ^~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyscard
Failed to build pyscard                                                                                                                                                                       
ERROR: Could not build wheels for pyscard, which is required to install pyproject.toml-based projects

jouellnyc, 2025-04-23:
Hmmm...I have a similar issue:

```
$~/gitrepos/pico-hsm/tools$ python3 -m venv ~/gitrepos/myhsmvenv

$ source  ~/gitrepos/myhsmvenv/bin/activate
(myhsmvenv) :~/gitrepos/pico-hsm/tools$ **python3 pico-hsm-tool.py initialize**
ERROR: cvc module not found! Install pycvc package.
Try with `pip install pycvc`

(myhsmvenv) john@user-OptiPlex-7050:~/gitrepos/pico-hsm/tools$ **pip install pycvc**
<snip>
Successfully installed cffi-1.17.1 cryptography-44.0.2 pycparser-2.22 pycvc-1.5.0 setuptools-79.0.0
(myhsmvenv) :~/gitrepos/pico-hsm/tools$ python3 pico-hsm-tool.py initialize
ERROR: picohsm module not found! Install picohsm package.
Try with `pip install pypicohsm`

(myhsmvenv) :~/gitrepos/pico-hsm/tools$ **pip install pypicohsm**
Collecting pypicohsm
  Using cached pypicohsm-1.4-py3-none-any.whl.metadata (2.0 kB)
Requirement already satisfied: setuptools in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from pypicohsm) (79.0.0)
Requirement already satisfied: cryptography>=3.3 in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from pypicohsm) (44.0.2)
Collecting base58 (from pypicohsm)
  Using cached base58-2.1.1-py3-none-any.whl.metadata (3.1 kB)
Collecting pyusb (from pypicohsm)
  Using cached pyusb-1.3.1-py3-none-any.whl.metadata (2.5 kB)
Requirement already satisfied: pycvc in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from pypicohsm) (1.5.0)
Collecting pyscard (from pypicohsm)
  Using cached pyscard-2.2.2.tar.gz (156 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: cffi>=1.12 in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from cryptography>=3.3->pypicohsm) (1.17.1)
Requirement already satisfied: pycparser in /home/john/gitrepos/myhsmvenv/lib/python3.12/site-packages (from cffi>=1.12->cryptography>=3.3->pypicohsm) (2.22)
Using cached pypicohsm-1.4-py3-none-any.whl (22 kB)
Using cached base58-2.1.1-py3-none-any.whl (5.6 kB)
Using cached pyusb-1.3.1-py3-none-any.whl (58 kB)
Building wheels for collected packages: pyscard
  Building wheel for pyscard (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pyscard (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      Package libpcsclite was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libpcsclite.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'libpcsclite', required by 'virtual:world', not found
      /tmp/pip-build-env-branwe8o/overlay/lib/python3.12/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!
      
              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:
      
              License :: OSI Approved :: GNU Lesser General Public License v2 or later (LGPLv2+)
      
              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************
      
      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      running build_py
      running build_ext
      building 'smartcard.scard._scard' extension
      swigging src/smartcard/scard/scard.i to src/smartcard/scard/scard_wrap.c
      swig -python -outdir src/smartcard/scard -DPCSCLITE -o src/smartcard/scard/scard_wrap.c src/smartcard/scard/scard.i
      creating build/temp.linux-x86_64-cpython-312/src/smartcard/scard
      x86_64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -DVER_PRODUCTVERSION=2,2,2,0000 -DVER_PRODUCTVERSION_STR=2.2.2 -DPCSCLITE=1 -Isrc/smartcard/scard/ -I/usr/include/PCSC -I/usr/local/include/PCSC -I/home/john/gitrepos/myhsmvenv/include -I/usr/include/python3.12 -c src/smartcard/scard/helpers.c -o build/temp.linux-x86_64-cpython-312/src/smartcard/scard/helpers.o
      src/smartcard/scard/helpers.c:28:10: fatal error: winscard.h: No such file or directory
         28 | #include <winscard.h>
            |          ^~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyscard
Failed to build pyscard                                                                                                                                                                       
ERROR: Could not build wheels for pyscard, which is required to install pyproject.toml-based projects

```

Tried to install swig and various other ideas from the web.

```
$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble

```

polhenarejos, 2025-04-23:
See https://github.com/LudovicRousseau/pyscard/issues/78

grayawa, 2025-12-18:
Is pypicohsm still exist?I can't find it on PyPI,nor github.I could only find it on google cached search results...

meteokr, 2025-12-31:
I'm also experiencing the same issue of trying to use this repo on Linux.

```
python3 pico-hsm/tools/pico-hsm-tool.py
ERROR: picohsm module not found! Install picohsm package.
Try with `pip install pypicohsm`
```
```
pip install pypicohsm
ERROR: Could not find a version that satisfies the requirement pypicohsm (from versions: none)
ERROR: No matching distribution found for pypicohsm
```

[image]

grayawa, 2026-01-01:
> I'm also experiencing the same issue of trying to use this repo on Linux.
> 
> ```
> python3 pico-hsm/tools/pico-hsm-tool.py
> ERROR: picohsm module not found! Install picohsm package.
> Try with `pip install pypicohsm`
> ```
> 
> ```
> pip install pypicohsm
> ERROR: Could not find a version that satisfies the requirement pypicohsm (from versions: none)
> ERROR: No matching distribution found for pypicohsm
> ```
> 
> <img alt="Image" width="1180" height="263" src="https://private-user-images.githubusercontent.com/17933684/531296134-e17ee122-59d5-4c84-af84-b02c953d0f83.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjcyNjQ2NzcsIm5iZiI6MTc2NzI2NDM3NywicGF0aCI6Ii8xNzkzMzY4NC81MzEyOTYxMzQtZTE3ZWUxMjItNTlkNS00Yzg0LWFmODQtYjAyYzk1M2QwZjgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTAxVDEwNDYxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3NTg4OGQ1MGZlOWVhMDhlYzE4NTYyMDhmMGFmYzU4MGNhOTM3MGFhMzMyMGIzZDg4M2M1OTkxOWIyNGY1NDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6309uwy5MHHlMPeYYEVuXguemBGg5udkSnA9WWJxG9s">

I do think the author deleted this repo.
He also deleted the pico-conmissioner for some reason?

polhenarejos, 2026-08-21:
Closed due to inactivity. Reopen if needed.

closed 2026-08-21 by polhenarejos