polhenarejos/pico-hsm #76: Raspberry pi pico 2 not seens by commissionner after provisionning HSM u2f

Cabaalist, 2025-01-31:
Download u2f for pico 2 - copy it on pico 2, pico restart 
open commissionner web page in Google Chrome, clic webUSB, Popup open but pico 2 dont appear.

If doing same process for pico 1 , every thing work

polhenarejos, 2025-01-31:
It's strange because there's nothing different between pico 1 and pico 2 in the code, a part from the RPi's Pico SDK. I do not have a Pico 2 but a Tiny2350 and it works well too.

Does an error appears in the Chrome's console?

Cabaalist, 2025-01-31:
For info, lsusb return : Bus 003 Device 011: ID 1050:0030 Yubico.com Pico Key

Here the errors from chrome console : 

pico-commissioner/:95 
        
        
       GET https://www.googletagmanager.com/gtag/js?id=G-K3H7HREJGY net::ERR_CONNECTION_REFUSED
jquery-migrate.min.js?ver=3.4.1:2 JQMIGRATE: Migrate is installed, version 3.4.1
text-limit.es5.min.js?ver=1.9.3.2:1 Uncaught ReferenceError: wpforms_settings is not defined
    at text-limit.es5.min.js?ver=1.9.3.2:1:1962
    at Array.map (<anonymous>)
    at Object.initHint (text-limit.es5.min.js?ver=1.9.3.2:1:1733)
    at a (text-limit.es5.min.js?ver=1.9.3.2:1:1617)
    at Object.<anonymous> (text-limit.es5.min.js?ver=1.9.3.2:1:3209)
    at s (text-limit.es5.min.js?ver=1.9.3.2:1:216)
    at i (text-limit.es5.min.js?ver=1.9.3.2:1:382)
    at text-limit.es5.min.js?ver=1.9.3.2:1:399
(anonymous) @ text-limit.es5.min.js?ver=1.9.3.2:1
initHint @ text-limit.es5.min.js?ver=1.9.3.2:1
a @ text-limit.es5.min.js?ver=1.9.3.2:1
(anonymous) @ text-limit.es5.min.js?ver=1.9.3.2:1
s @ text-limit.es5.min.js?ver=1.9.3.2:1
i @ text-limit.es5.min.js?ver=1.9.3.2:1
(anonymous) @ text-limit.es5.min.js?ver=1.9.3.2:1
wcid.js?v11:204 NotFoundError: Failed to execute 'requestDevice' on 'USB': No device selected.
pico-commissioner/:1 Uncaught (in promise) NotFoundError: Failed to execute 'requestDevice' on 'USB': No device selected.

polhenarejos, 2025-01-31:
How did you burn this VIDPID? Do not use Yubico's VIDPID, as they are blocked from WebUSB.

Cabaalist, 2025-01-31:
Hum, Not sure, was from a previous firmware I guess

Cabaalist, 2025-01-31:
Ok my bad, sorry for the annoyance, the pico is burned or I did something wrong with it, I just tested with my second pico 2 and work fine.

closed 2025-01-31 by Cabaalist