polhenarejos/pico-hsm #80: Fails to sign with 4096 bit RSA

daniel-starke, 2025-03-03:
I am using a Xiao RP2350 with PicoHSM FW 5.4 as provided in the release notes here.
I am executing the following commands on Windows using OpenSC 0.26.1:
`pkcs11-tool -l --pin <mypin> --keypairgen --key-type rsa:4096 --id 1 --label 001`  
`openssl req -x509 -new -nodes -engine pkcs11 -key 31 -keyform engine -out 001.crt -days 9125 -config openssl.cnf -subj "%SUBJ%" -reqexts ca_ext`

This, however, fails after sending the APDU for signing:
```
P:8052; T:2440 2025-03-03 22:28:52.364 [opensc-pkcs11] framework-pkcs15.c:4443:pkcs15_prkey_sign: Selected flags 140. Now computing signature for 51 bytes. 512 bytes reserved.
P:8052; T:2440 2025-03-03 22:28:52.368 [opensc-pkcs11] pkcs15-sec.c:615:sc_pkcs15_compute_signature: called
P:8052; T:2440 2025-03-03 22:28:52.371 [opensc-pkcs11] pkcs15-sec.c:668:sc_pkcs15_compute_signature: supported algorithm flags 0x80E0E211, private key usage 0x2E
P:8052; T:2440 2025-03-03 22:28:52.375 [opensc-pkcs11] padding.c:701:sc_get_encoding_flags: called
P:8052; T:2440 2025-03-03 22:28:52.377 [opensc-pkcs11] padding.c:705:sc_get_encoding_flags: iFlags 0x140, card capabilities 0x80E0E211
P:8052; T:2440 2025-03-03 22:28:52.380 [opensc-pkcs11] padding.c:766:sc_get_encoding_flags: pad flags 0x140, secure algorithm flags 0x1
P:8052; T:2440 2025-03-03 22:28:52.384 [opensc-pkcs11] padding.c:767:sc_get_encoding_flags: returning with: 0 (Success)
P:8052; T:2440 2025-03-03 22:28:52.387 [opensc-pkcs11] pkcs15-sec.c:739:sc_pkcs15_compute_signature: DEE flags:0x00000140 alg_info->flags:0x80e0e211 pad:0x00000140 sec:0x00000001
P:8052; T:2440 2025-03-03 22:28:52.391 [opensc-pkcs11] padding.c:624:sc_pkcs1_encode: called
P:8052; T:2440 2025-03-03 22:28:52.393 [opensc-pkcs11] padding.c:630:sc_pkcs1_encode: hash algorithm 0x100, pad algorithm 0x40
P:8052; T:2440 2025-03-03 22:28:52.396 [opensc-pkcs11] padding.c:654:sc_pkcs1_encode: returning with: 0 (Success)
P:8052; T:2440 2025-03-03 22:28:52.399 [opensc-pkcs11] card.c:471:sc_lock: called
P:8052; T:2440 2025-03-03 22:28:52.401 [opensc-pkcs11] card.c:513:sc_lock: returning with: 0 (Success)
P:8052; T:2440 2025-03-03 22:28:52.404 [opensc-pkcs11] sec.c:108:sc_set_security_env: called
P:8052; T:2440 2025-03-03 22:28:52.406 [opensc-pkcs11] card-sc-hsm.c:1069:sc_hsm_set_security_env: returning with: 0 (Success)
P:8052; T:2440 2025-03-03 22:28:52.409 [opensc-pkcs11] sec.c:112:sc_set_security_env: returning with: 0 (Success)
P:8052; T:2440 2025-03-03 22:28:52.412 [opensc-pkcs11] sec.c:62:sc_compute_signature: called
P:8052; T:2440 2025-03-03 22:28:52.415 [opensc-pkcs11] apdu.c:550:sc_transmit_apdu: called
P:8052; T:2440 2025-03-03 22:28:52.417 [opensc-pkcs11] card.c:471:sc_lock: called
P:8052; T:2440 2025-03-03 22:28:52.419 [opensc-pkcs11] card.c:513:sc_lock: returning with: 0 (Success)
P:8052; T:2440 2025-03-03 22:28:52.422 [opensc-pkcs11] apdu.c:515:sc_transmit: called
P:8052; T:2440 2025-03-03 22:28:52.424 [opensc-pkcs11] apdu.c:363:sc_single_transmit: called
P:8052; T:2440 2025-03-03 22:28:52.426 [opensc-pkcs11] apdu.c:367:sc_single_transmit: CLA:80, INS:68, P1:1, P2:20, data(512) 0000000000330000
P:8052; T:2440 2025-03-03 22:28:52.430 [opensc-pkcs11] reader-pcsc.c:327:pcsc_transmit: reader 'Pol Henarejos Pico Key CCID OTP FIDO Interfac 0'
P:8052; T:2440 2025-03-03 22:28:52.433 [opensc-pkcs11] reader-pcsc.c:328:pcsc_transmit:
Outgoing APDU (521 bytes):
80 68 01 20 00 02 00 00 01 FF FF FF FF FF FF FF .h. ............
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF 00 30 31 30 0D 06 09 60 86 48 01 65 03 ....010...`.H.e.
04 02 01 05 00 04 20 D7 D4 B4 55 0B 3D 08 E7 CA ...... ...U.=...
0E E4 50 97 50 52 15 0B 5D 9E 00 A3 7C 91 DC 27 ..P.PR..]...|..'
EA 0A 84 7C 33 22 C8 02 00                      ...|3"...

P:8052; T:2440 2025-03-03 22:28:52.479 [opensc-pkcs11] reader-pcsc.c:245:pcsc_internal_transmit: called
P:8052; T:2440 2025-03-03 22:28:55.666 [opensc-pkcs11] reader-pcsc.c:334:pcsc_transmit: unable to transmit
P:8052; T:2440 2025-03-03 22:28:55.670 [opensc-pkcs11] apdu.c:380:sc_single_transmit: unable to transmit APDU: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.674 [opensc-pkcs11] apdu.c:518:sc_transmit: transmit APDU failed: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.677 [opensc-pkcs11] card.c:523:sc_unlock: called
P:8052; T:2440 2025-03-03 22:28:55.679 [opensc-pkcs11] card-sc-hsm.c:1121:sc_hsm_compute_signature: APDU transmit failed: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.683 [opensc-pkcs11] sec.c:66:sc_compute_signature: returning with: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.687 [opensc-pkcs11] card.c:523:sc_unlock: called
P:8052; T:2440 2025-03-03 22:28:55.689 [opensc-pkcs11] pkcs15-sec.c:169:use_key: returning with: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.692 [opensc-pkcs11] pkcs15-sec.c:779:sc_pkcs15_compute_signature: use_key() failed: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.698 [opensc-pkcs11] pkcs15-sec.c:792:sc_pkcs15_compute_signature: returning with: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.702 [opensc-pkcs11] card.c:523:sc_unlock: called
P:8052; T:2440 2025-03-03 22:28:55.704 [opensc-pkcs11] reader-pcsc.c:734:pcsc_unlock: called
P:8052; T:2440 2025-03-03 22:28:55.707 [opensc-pkcs11] framework-pkcs15.c:4451:pkcs15_prkey_sign: Sign complete. Result -1213.
P:8052; T:2440 2025-03-03 22:28:55.710 [opensc-pkcs11] misc.c:72:sc_to_cryptoki_error_common: libopensc return value: -1213 (Unknown data received from card)
P:8052; T:2440 2025-03-03 22:28:55.714 [opensc-pkcs11] mechanism.c:608:sc_pkcs11_signature_final: returning with: 5
P:8052; T:2440 2025-03-03 22:28:55.716 [opensc-pkcs11] mechanism.c:465:sc_pkcs11_sign_final: returning with: 5
P:8052; T:2440 2025-03-03 22:28:55.719 [opensc-pkcs11] pkcs11-object.c:745:C_Sign: C_Sign() = CKR_GENERAL_ERROR
```

In Wireshark I can see two things I would expect different (without checking the actual specification):
```
21:39:22.514397 USB CCID
    Message Type: RDR_to_PC_DataBlock (0x80)
    Packet Length: 0
    Slot: 0
    Sequence: 170
    Status: 128, Status: An ICC is present and active, Status: Time Extension is requested 
    Error: 0
    Chain Parameter: 0

21:39:24.015537	USB CCID
    Message Type: RDR_to_PC_DataBlock (0x80)
    Packet Length: 0
    Slot: 0
    Sequence: 170
    Status: 128, Status: An ICC is present and active, Status: Time Extension is requested 
    Error: 0
    Chain Parameter: 0

21:39:24.193537 USB CCID
    Message Type: RDR_to_PC_DataBlock (0x80)
    Packet Length: 514
    Slot: 0
    Sequence: 170
    Status: 0, Status: An ICC is present and active, Status: Processed without error 
    Error: 0
    Chain Parameter: 0

21:39:24.193537 USB CCID
    Message Type: Unknown (0xbd)
```
First: The sequence number stays the same. Is that correct?  
Second: There is some unknown message type from Pico HSM to PC. What means 0xBD? I can see a lot of 0x55 in other parts, but that does not seems to be an issue.

polhenarejos, 2025-03-03:
Try prepending `OPENSC_DEBUG=9`

```
OPENSC_DEBUG=9 pkcs11-tool -l --pin <mypin> --keypairgen --key-type rsa:4096 --id 1 --label 001
openssl req -x509 -new -nodes -engine pkcs11 -key 31 -keyform engine -out 001.crt -days 9125 -config openssl.cnf -subj "%SUBJ%" -reqexts ca_ext
```
and paste the log.

polhenarejos, 2025-03-03:
Can you try with a shorter key like 1024?

daniel-starke, 2025-03-04:
1024: OK
2048: OK
3072: FAIL
4096: FAIL

polhenarejos, 2025-03-11:
Try this snippet and report the result:

```
echo "This is a test string. Be safe, be secure." > data
pkcs11-tool --read-object --pin 648219 --id 1 --type pubkey > 1.der
openssl rsa -inform DER -outform PEM -in 1.der -pubin > 1.pub
openssl dgst -sha1 -binary -out data.sha1 data
pkcs11-tool --id 1 --sign --pin 648219 --mechanism SHA1-RSA-PKCS -i data -o data.sig
openssl pkeyutl -verify -in data.sha1 -sigfile data.sig -pubin -inkey 1.pub -pkeyopt digest:sha1
```

daniel-starke, 2025-03-11:
```
> pkcs11-tool -l --pin 381180295 --keypairgen --key-type rsa:1024 --id 3 --label test
Using slot 0 with a present token (0x0)
error: PKCS11 function C_GenerateKeyPair failed: rv = CKR_GENERAL_ERROR (0x5)
Aborting.
> pkcs11-tool --read-object --pin 381180295 --id 33 --type pubkey > 1.der
Using slot 0 with a present token (0x0)
> openssl rsa -inform DER -outform PEM -in 1.der -pubin > 1.pub
writing RSA key
> openssl dgst -sha1 -binary -out data.sha1 data
> pkcs11-tool --id 33 --sign --pin 381180295 --mechanism SHA1-RSA-PKCS -i data -o data.sig
Using slot 0 with a present token (0x0)
Using signature algorithm SHA1-RSA-PKCS
> openssl pkeyutl -verify -in data.sha1 -sigfile data.sig -pubin -inkey 1.pub -pkeyopt digest:sha1
Signature Verified Successfully
```

[test.zip](https://github.com/user-attachments/files/19195771/test.zip)

polhenarejos, 2025-03-11:
Is the key 33 a rsa 4096?

daniel-starke, 2025-03-11:
No, 1024 because 4096 fails.
Btw, I am not sure why but keypairgen id 3 is turned into id 33.

polhenarejos, 2025-03-11:
So you tried the snippet with a rsa 4096 and failed, right? In which step? Which error?

daniel-starke, 2025-03-12:
If you ask me that way... key generation fails for every RSA key size:
```
P:1388; T:8160 2025-03-12 18:08:00.295 [opensc-pkcs11] pkcs15-sc-hsm.c:116:sc_hsm_create_key: called
P:1388; T:8160 2025-03-12 18:08:00.298 [opensc-pkcs11] pkcs15-sc-hsm.c:117:sc_hsm_create_key: returning with: 0 (Success)
P:1388; T:8160 2025-03-12 18:08:00.301 [opensc-pkcs11] pkcs15-sc-hsm.c:260:sc_hsm_generate_key: called
P:1388; T:8160 2025-03-12 18:08:00.303 [opensc-pkcs11] pkcs15-sc-hsm.c:137:sc_hsm_determine_free_id: called
P:1388; T:8160 2025-03-12 18:08:00.306 [opensc-pkcs11] card.c:559:sc_list_files: called
P:1388; T:8160 2025-03-12 18:08:00.309 [opensc-pkcs11] apdu.c:550:sc_transmit_apdu: called
P:1388; T:8160 2025-03-12 18:08:00.311 [opensc-pkcs11] card.c:471:sc_lock: called
P:1388; T:8160 2025-03-12 18:08:00.314 [opensc-pkcs11] card.c:513:sc_lock: returning with: 0 (Success)
P:1388; T:8160 2025-03-12 18:08:00.316 [opensc-pkcs11] apdu.c:515:sc_transmit: called
P:1388; T:8160 2025-03-12 18:08:00.319 [opensc-pkcs11] apdu.c:363:sc_single_transmit: called
P:1388; T:8160 2025-03-12 18:08:00.321 [opensc-pkcs11] apdu.c:367:sc_single_transmit: CLA:80, INS:58, P1:0, P2:0, data(0) 0000000000000000
P:1388; T:8160 2025-03-12 18:08:00.325 [opensc-pkcs11] reader-pcsc.c:327:pcsc_transmit: reader 'Pol Henarejos Pico Key CCID OTP FIDO Interfac 0'
P:1388; T:8160 2025-03-12 18:08:00.328 [opensc-pkcs11] reader-pcsc.c:328:pcsc_transmit:
Outgoing APDU (7 bytes):
80 58 00 00 00 00 00 .X.....

P:1388; T:8160 2025-03-12 18:08:00.332 [opensc-pkcs11] reader-pcsc.c:245:pcsc_internal_transmit: called
P:1388; T:8160 2025-03-12 18:08:00.336 [opensc-pkcs11] reader-pcsc.c:337:pcsc_transmit:
Incoming APDU (14 bytes):
C4 00 CC 00 CC 01 CC 02 C4 01 C4 02 90 00 ..............

P:1388; T:8160 2025-03-12 18:08:00.341 [opensc-pkcs11] apdu.c:382:sc_single_transmit: returning with: 0 (Success)
P:1388; T:8160 2025-03-12 18:08:00.344 [opensc-pkcs11] apdu.c:539:sc_transmit: returning with: 0 (Success)
P:1388; T:8160 2025-03-12 18:08:00.347 [opensc-pkcs11] card.c:523:sc_unlock: called
P:1388; T:8160 2025-03-12 18:08:00.349 [opensc-pkcs11] card-sc-hsm.c:961:sc_hsm_list_files: returning with: 12
P:1388; T:8160 2025-03-12 18:08:00.352 [opensc-pkcs11] card.c:565:sc_list_files: returning with: 12
P:1388; T:8160 2025-03-12 18:08:00.355 [opensc-pkcs11] pkcs15-sc-hsm.c:149:sc_hsm_determine_free_id: returning with: 3
P:1388; T:8160 2025-03-12 18:08:00.357 [opensc-pkcs11] pkcs15-sc-hsm.c:160:sc_hsm_encode_gakp_rsa: called
P:1388; T:8160 2025-03-12 18:08:00.360 [opensc-pkcs11] pkcs15-sc-hsm.c:172:sc_hsm_encode_gakp_rsa: returning with: 0 (Success)
P:1388; T:8160 2025-03-12 18:08:00.363 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry: encoding 'certificate'
P:1388; T:8160 2025-03-12 18:08:00.367 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry: type=129, tag=0x60001f21, parm=000000000030d110, len=0
P:1388; T:8160 2025-03-12 18:08:00.370 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:  encoding 'certificateBody'
P:1388; T:8160 2025-03-12 18:08:00.374 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry:  type=129, tag=0x60001f4e, parm=000000000030d190, len=0
P:1388; T:8160 2025-03-12 18:08:00.378 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:   encoding 'certificateProfileIdentifier'
P:1388; T:8160 2025-03-12 18:08:00.381 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry:   type=2, tag=0x40001f29, parm=000000000030d510, len=0
P:1388; T:8160 2025-03-12 18:08:00.385 [opensc-pkcs11] asn1.c:2026:asn1_encode_entry:   length of encoded item=4
P:1388; T:8160 2025-03-12 18:08:00.388 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:   encoding 'certificationAuthorityReference'
P:1388; T:8160 2025-03-12 18:08:00.391 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry:   type=19, tag=0x40000002, parm=000000000030d514, len=9
P:1388; T:8160 2025-03-12 18:08:00.395 [opensc-pkcs11] asn1.c:2026:asn1_encode_entry:   length of encoded item=11
P:1388; T:8160 2025-03-12 18:08:00.398 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:   encoding 'publicKey'
P:1388; T:8160 2025-03-12 18:08:00.401 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry:   type=129, tag=0x60001f49, parm=000000000030d260, len=0
P:1388; T:8160 2025-03-12 18:08:00.405 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:    encoding 'publicKeyOID'
P:1388; T:8160 2025-03-12 18:08:00.408 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry:    type=6, tag=0x06, parm=000000000030d530, len=0
P:1388; T:8160 2025-03-12 18:08:00.411 [opensc-pkcs11] asn1.c:2026:asn1_encode_entry:    length of encoded item=12
P:1388; T:8160 2025-03-12 18:08:00.414 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:    encoding 'primeOrModulus' (not present)
P:1388; T:8160 2025-03-12 18:08:00.417 [opensc-pkcs11] asn1.c:2026:asn1_encode_entry:    length of encoded item=0
P:1388; T:8160 2025-03-12 18:08:00.420 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:    encoding 'coefficientAorExponent'
P:1388; T:8160 2025-03-12 18:08:00.423 [opensc-pkcs11] asn1.c:1833:asn1_encode_entry:    type=4, tag=0x80000002, parm=0000000001ea70a0, len=3
P:1388; T:8160 2025-03-12 18:08:00.427 [opensc-pkcs11] asn1.c:2026:asn1_encode_entry:    length of encoded item=5
P:1388; T:8160 2025-03-12 18:08:00.430 [opensc-pkcs11] asn1.c:1828:asn1_encode_entry:    encoding 'coefficientB' (not present)
P:1388; T:8160 2025-03-12 18:08:00.433 [opensc-pkcs11] asn1.c:2026:asn1_encode_entry:    length of encoded item=0
[... cut ...]
Same each (example above for 1024 bit):
- `pkcs11-tool -l --pin 0815 --keypairgen --key-type rsa:1024 --id 3 --label test`
- `pkcs11-tool -l --pin 0815 --keypairgen --key-type rsa:2048 --id 3 --label test`
- `pkcs11-tool -l --pin 0815 --keypairgen --key-type rsa:3072 --id 3 --label test`
- `pkcs11-tool -l --pin 0815 --keypairgen --key-type rsa:4096 --id 3 --label test`

However, it completes in the background after a while. But usually, the device needs to be plugged again after that to work correctly.

daniel-starke, 2025-03-12:
Just for reference:
If I do all the steps [here](https://github.com/polhenarejos/pico-hsm/issues/80#issuecomment-2715242135) with the 4096 bit RSA key it fails at:
`pkcs11-tool --id 31 --sign --pin 0815 --mechanism SHA1-RSA-PKCS -i data -o data.sig`
with the following error:
```
P:4060; T:2008 2025-03-12 21:09:55.863 [opensc-pkcs11] framework-pkcs15.c:4297:pkcs15_prkey_sign: Initiating signing operation, mechanism 0x6.
P:4060; T:2008 2025-03-12 21:09:55.863 [opensc-pkcs11] card.c:471:sc_lock: called
P:4060; T:2008 2025-03-12 21:09:55.863 [opensc-pkcs11] reader-pcsc.c:692:pcsc_lock: called
P:4060; T:2008 2025-03-12 21:09:55.863 [opensc-pkcs11] card.c:513:sc_lock: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.863 [opensc-pkcs11] framework-pkcs15.c:4443:pkcs15_prkey_sign: Selected flags 240. Now computing signature for 20 bytes. 512 bytes reserved.
P:4060; T:2008 2025-03-12 21:09:55.864 [opensc-pkcs11] pkcs15-sec.c:615:sc_pkcs15_compute_signature: called
P:4060; T:2008 2025-03-12 21:09:55.864 [opensc-pkcs11] pkcs15-sec.c:668:sc_pkcs15_compute_signature: supported algorithm flags 0x80E0E211, private key usage 0x2E
P:4060; T:2008 2025-03-12 21:09:55.864 [opensc-pkcs11] padding.c:701:sc_get_encoding_flags: called
P:4060; T:2008 2025-03-12 21:09:55.864 [opensc-pkcs11] padding.c:705:sc_get_encoding_flags: iFlags 0x240, card capabilities 0x80E0E211
P:4060; T:2008 2025-03-12 21:09:55.864 [opensc-pkcs11] padding.c:766:sc_get_encoding_flags: pad flags 0x240, secure algorithm flags 0x1
P:4060; T:2008 2025-03-12 21:09:55.864 [opensc-pkcs11] padding.c:767:sc_get_encoding_flags: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.865 [opensc-pkcs11] pkcs15-sec.c:739:sc_pkcs15_compute_signature: DEE flags:0x00000240 alg_info->flags:0x80e0e211 pad:0x00000240 sec:0x00000001
P:4060; T:2008 2025-03-12 21:09:55.865 [opensc-pkcs11] padding.c:624:sc_pkcs1_encode: called
P:4060; T:2008 2025-03-12 21:09:55.865 [opensc-pkcs11] padding.c:630:sc_pkcs1_encode: hash algorithm 0x200, pad algorithm 0x40
P:4060; T:2008 2025-03-12 21:09:55.865 [opensc-pkcs11] padding.c:654:sc_pkcs1_encode: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.865 [opensc-pkcs11] card.c:471:sc_lock: called
P:4060; T:2008 2025-03-12 21:09:55.865 [opensc-pkcs11] card.c:513:sc_lock: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] sec.c:108:sc_set_security_env: called
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] card-sc-hsm.c:1069:sc_hsm_set_security_env: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] sec.c:112:sc_set_security_env: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] sec.c:62:sc_compute_signature: called
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] apdu.c:550:sc_transmit_apdu: called
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] card.c:471:sc_lock: called
P:4060; T:2008 2025-03-12 21:09:55.866 [opensc-pkcs11] card.c:513:sc_lock: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:09:55.867 [opensc-pkcs11] apdu.c:515:sc_transmit: called
P:4060; T:2008 2025-03-12 21:09:55.867 [opensc-pkcs11] apdu.c:363:sc_single_transmit: called
P:4060; T:2008 2025-03-12 21:09:55.867 [opensc-pkcs11] apdu.c:367:sc_single_transmit: CLA:80, INS:68, P1:1, P2:20, data(512) 0000000000100000
P:4060; T:2008 2025-03-12 21:09:55.867 [opensc-pkcs11] reader-pcsc.c:327:pcsc_transmit: reader 'Pol Henarejos Pico Key CCID OTP FIDO Interfac 0'
P:4060; T:2008 2025-03-12 21:09:55.867 [opensc-pkcs11] reader-pcsc.c:328:pcsc_transmit: 
Outgoing APDU (521 bytes):
80 68 01 20 00 02 00 00 01 FF FF FF FF FF FF FF .h. ............
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
[... cut ...]
P:4060; T:2008 2025-03-12 21:10:02.271 [opensc-pkcs11] reader-pcsc.c:677:pcsc_disconnect: Pol Henarejos Pico Key CCID OTP FIDO Interfac 0:SCardDisconnect returned: 0x00000000
P:4060; T:2008 2025-03-12 21:10:02.272 [opensc-pkcs11] card.c:434:sc_disconnect_card: returning with: 0 (Success)
P:4060; T:2008 2025-03-12 21:10:02.272 [opensc-pkcs11] ctx.c:1066:sc_release_context: called
P:4060; T:2008 2025-03-12 21:10:02.272 [opensc-pkcs11] reader-pcsc.c:971:pcsc_finish: called
error: PKCS11 function C_SignFinal failed: rv = CKR_GENERAL_ERROR (0x5)
Aborting.
```

polhenarejos, 2025-04-22:
Really strange. Also in macOS and Linux works fine too. I don’t know what changed between windows builds.

daniel-starke, 2025-04-22:
P.S.: RSA 4096 creation took around 300s on that device.

daniel-starke, 2025-05-12:
I suggest closing this issue with a remark in the readme for the minimal supported/tested Windows version.
Or do you plan to fix it?

polhenarejos, 2025-05-12:
To me it’s strange that in windows doesn’t work 4k, specially since 1k or 2k work well and all them work in macos/linux too. There’s nothing special different between 2k and 4k.

daniel-starke, 2025-05-12:
As mentioned before, Windows 7 seems to handle a couple of things differently.
Neither did the timeout function work properly nor the labeling.
Nevertheless, the question remains whether this project wants to support Windows 7 or not.

daniel-starke, 2025-05-23:
Hard to say. Sounds a bit different.

polhenarejos, 2025-05-23:
Not the problem but the cause, an underlaying issue with returned content.

polhenarejos, 2026-08-21:
Closed due to inactivity. Reopen if needed.

closed 2026-08-21 by polhenarejos