polhenarejos/pico-hsm #83: pico-hsm-tool.py & windows 11

squat0001, 2025-03-22:
The toot works fine with linux, but not with windows.. 

C:\Users\user01\Downloads\hsm\pico-hsm\tools>venv\Scripts\python.exe pico-hsm-tool.py keygen ed25519
Pico HSM Tool v2.2
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

Traceback (most recent call last):
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\pico-hsm-tool.py", line 570, in <module>
    run()
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\pico-hsm-tool.py", line 567, in run
    main(args)
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\pico-hsm-tool.py", line 537, in main
    picohsm = PicoHSM(args.pin)
              ^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\picohsm\PicoHSM.py", line 226, in __init__
    resp, sw1, sw2 = self.select_applet(rescue=True)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\picohsm\PicoHSM.py", line 243, in select_applet
    return self.__card.transmit([0x00, 0xA4, 0x04, 0x04, 0x08, 0xA0, 0x58, 0x3F, 0xC1, 0x9B, 0x7E, 0x4F, 0x21, 0x00])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\smartcard\CardConnectionDecorator.py", line 86, in transmit
    return self.component.transmit(command, protocol)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\smartcard\CardConnection.py", line 174, in transmit
    data, sw1, sw2 = self.doTransmit(command, protocol)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\smartcard\pcsc\PCSCCardConnection.py", line 269, in doTransmit
    raise CardConnectionException(
smartcard.Exceptions.CardConnectionException: Failed to transmit with protocol T1. Falscher Parameter. : Falscher Parameter.  (0x00000057)

polhenarejos, 2025-03-22:
Do you use th Eddsa firmware?

squat0001, 2025-03-22:
Yes
I need the ed25519 curve.

And ... Thank you for this project 😄

polhenarejos, 2025-03-22:
Are you able to do anything else in Windows or it’s just this curve?

squat0001, 2025-03-22:
i can access it nearly normal.. but.. i wanted to use the ed25519, and without the tools script its complicated to use

here an example..

C:\Program Files\OpenSC Project\OpenSC\tools>pkcs11-tool --slot-index 2 -l --pin  xxxxxxxxxxx -k
Using slot with index 2 (0x8)
Key pair generated:
Private Key Object; RSA
  label:      Private Key
  ID:         946c1592f383183d33c3b97aefcbda475868cbd0
  Usage:      sign
  Access:     sensitive, always sensitive, never extractable, local
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%946c1592f383183d33c3b97aefcbda475868cbd0;object=Private%20Key;type=private
Public Key Object; RSA 1024 bits
  label:      Private Key
  ID:         946c1592f383183d33c3b97aefcbda475868cbd0
  Usage:      verify
  Access:     none
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%946c1592f383183d33c3b97aefcbda475868cbd0;object=Private%20Key;type=public

C:\Program Files\OpenSC Project\OpenSC\tools>pkcs11-tool --slot-index 2 -l --pin 9039655317 -k
Using slot with index 2 (0x8)
Key pair generated:
Private Key Object; RSA
  label:      Private Key
  ID:         5ceecde2a52823d284ef704e0dbd6a4f27e63e24
  Usage:      sign
  Access:     sensitive, always sensitive, never extractable, local
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%5ceecde2a52823d284ef704e0dbd6a4f27e63e24;object=Private%20Key;type=private
Public Key Object; RSA 1024 bits
  label:      Private Key
  ID:         5ceecde2a52823d284ef704e0dbd6a4f27e63e24
  Usage:      verify
  Access:     none
  uri:        pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM;id=%5ceecde2a52823d284ef704e0dbd6a4f27e63e24;object=Private%20Key;type=public

squat0001, 2025-03-22:
no all commads do end in the same error

C:\Users\user01\Downloads\hsm\pico-hsm\tools>venv\Scripts\python.exe pico-hsm-tool.py reboot
Pico HSM Tool v2.2
Author: Pol Henarejos
Report bugs to https://github.com/polhenarejos/pico-hsm/issues

Traceback (most recent call last):
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\pico-hsm-tool.py", line 570, in <module>
    run()
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\pico-hsm-tool.py", line 567, in run
    main(args)
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\pico-hsm-tool.py", line 537, in main
    picohsm = PicoHSM(args.pin)
              ^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\picohsm\PicoHSM.py", line 226, in __init__
    resp, sw1, sw2 = self.select_applet(rescue=True)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\picohsm\PicoHSM.py", line 243, in select_applet
    return self.__card.transmit([0x00, 0xA4, 0x04, 0x04, 0x08, 0xA0, 0x58, 0x3F, 0xC1, 0x9B, 0x7E, 0x4F, 0x21, 0x00])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\smartcard\CardConnectionDecorator.py", line 86, in transmit
    return self.component.transmit(command, protocol)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\smartcard\CardConnection.py", line 174, in transmit
    data, sw1, sw2 = self.doTransmit(command, protocol)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user01\Downloads\hsm\pico-hsm\tools\venv\Lib\site-packages\smartcard\pcsc\PCSCCardConnection.py", line 269, in doTransmit
    raise CardConnectionException(
smartcard.Exceptions.CardConnectionException: Failed to transmit with protocol T1. Falscher Parameter. : Falscher Parameter.  (0x00000057)

polhenarejos, 2025-03-22:
Can you try with a console with Admin privileges?

squat0001, 2025-03-22:
same error, i just tested it, but i also tried it before opening the case
i do have in windows multiple devices, is it possible that i have to select one manually? does the script have an parameter selecting the slot?

C:\Program Files\OpenSC Project\OpenSC\tools>pkcs11-tool.exe -L
Available slots:
Slot 0 (0x0): Alcor Micro USB Smart Card Reader 0
  (empty)
Slot 1 (0x4): Microsoft UICC ISO Reader 30ae70ca 0
  (token not recognized)
Slot 2 (0x8): Pol Henarejos Pico Key CCID OTP FIDO Interfac 0
  token label        : Pico-HSM
  token manufacturer : Pol Henarejos
  token model        : PKCS#15 emulated
  token flags        : login required, PIN pad present, rng, token initialized, PIN initialized
  hardware version   : 24.13
  firmware version   : 5.4
  serial num         : ESPICOHSMTR
  pin min/max        : 6/15
  uri                : pkcs11:model=PKCS%2315%20emulated;manufacturer=Pol%20Henarejos;serial=ESPICOHSMTR;token=Pico-HSM
Slot 3 (0xc): Windows Hello for Business 1
  token label        : GIDS card
  token manufacturer : www.mysmartlogon.com
  token model        : PKCS#15 emulated
  token flags        : login required, token initialized, PIN initialized
  hardware version   : 0.0
  firmware version   : 0.0
  serial num         : afc1469fb2b9074a
  pin min/max        : 4/15
  uri                : pkcs11:model=PKCS%2315%20emulated;manufacturer=www.mysmartlogon.com;serial=afc1469fb2b9074a;token=GIDS%20card

polhenarejos, 2025-03-22:
Paste the output of `opensc-tool -l`

squat0001, 2025-03-22:
C:\Program Files\OpenSC Project\OpenSC\tools>opensc-tool -l
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Alcor Micro USB Smart Card Reader 0
1    Yes             Microsoft UICC ISO Reader 30ae70ca 0
2    Yes             Pol Henarejos Pico Key CCID OTP FIDO Interfac 0
3    Yes             Windows Hello for Business 1

polhenarejos, 2025-03-22:
I'd suggest to remove all others devices to see if this is the problem. I'll see how to select the slot in the Pico HSM Tool btw.

squat0001, 2025-03-22:
ok .. i disabled all, and the problem device is the UICC ISO which is ihe card from the LTE modem in the notebook, .. and its working :-) 
so if its possible to select the slot.. it would be really nice
or catch the error in the script probing the devices

commit 3e1b29f7 (2025-03-23): Add support for slot selection.

Fixes #83.
closed 2025-04-10 by polhenarejos