polhenarejos/pico-hsm #84: xca can not input pin

IsayIsee, 2025-04-03:
firmware:[3e1b29f](https://github.com/polhenarejos/pico-hsm/commit/3e1b29f72a1366d5b68803cfd75236ab58f27250)
xca version: 2.9

using pico-hsm, xca does not prompt for a PIN. I checked the xca manual（https://hohnstaedt.de/xca-doc/html/smartcard.html#） and found that when xca thinks the token has a built-in keyboard, it will use the built-in keyboard by default without asking for a PIN. I don't know if this is the reason why it does not prompt for a PIN.

polhenarejos, 2025-04-06:
Related with #81 

What `pkcs11-tool -L` returns?

IsayIsee, 2025-04-07:
[image]

IsayIsee, 2025-04-07:
i tested [611b762](https://github.com/IsayIsee/pico-hsm/commit/611b762a1fd48be2a617178e485eca4af9742aee), xca can prompt input pin now

thks

closed 2025-04-10 by polhenarejos