polhenarejos/pico-hsm #90: OpenSC 0.26.1 Init issue

IsayIsee, 2025-05-13:
install opensc 0.26.1 and use pico-hsm-tool init

```
C:\Program Files\OpenSC Project\OpenSC\tools>pkcs15-tool.exe -D
Using reader with a card: Pol Henarejos Pico Key CCID OTP FIDO Interfac 0
PKCS#15 Card [Pico-HSM]:
        Version        : 5
        Serial number  : ESPICOHSMTR
        Manufacturer ID: Pol Henarejos
        Flags          : PRN generation, EID compliant

PIN [UserPIN]
        Object Flags   : [0x03], private, modifiable
        Auth ID        : 02
        ID             : 01
        Flags          : [0x812], local, initialized, exchangeRefData
        Length         : min_len:6, max_len:15, stored_len:0
        Pad char       : 0x00
        Reference      : 129 (0x81)
        Type           : ascii-numeric
        Path           : e82b0601040181c31f0201::
        Tries left     : 3

PIN [SOPIN]
        Object Flags   : [0x01], private
        ID             : 02
        Flags          : [0x9A], local, unblock-disabled, initialized, soPin
        Length         : min_len:16, max_len:16, stored_len:0
        Pad char       : 0x00
        Reference      : 136 (0x88)
        Type           : bcd
        Path           : e82b0601040181c31f0201::
        Tries left     : 15
```

install opensc 0.25.1 and init 

```
C:\Program Files\OpenSC Project\OpenSC\tools>pkcs15-tool.exe -D
Using reader with a card: Pol Henarejos Pico Key CCID OTP FIDO Interfac 0
PKCS#15 Card [Pico-HSM]:
        Version        : 5
        Serial number  : ESPICOHSMTR
        Manufacturer ID: Pol Henarejos
        Flags          : PRN generation, EID compliant

PIN [UserPIN]
        Object Flags   : [0x03], private, modifiable
        Auth ID        : 02
        ID             : 01
        Flags          : [0x812], local, initialized, exchangeRefData
        Length         : min_len:6, max_len:15, stored_len:0
        Pad char       : 0x00
        Reference      : 129 (0x81)
        Type           : ascii-numeric
        Path           : e82b0601040181c31f0201::
        Tries left     : 3

PIN [SOPIN]
        Object Flags   : [0x01], private
        ID             : 02
        Flags          : [0x9A], local, unblock-disabled, initialized, soPin
        Length         : min_len:16, max_len:16, stored_len:0
        Pad char       : 0x00
        Reference      : 136 (0x88)
        Type           : bcd
        Path           : e82b0601040181c31f0201::
        Tries left     : 15

Private EC Key [ESPICOHSMTR]
        Object Flags   : [0x01], private
        Usage          : [0x104], sign, derive
        Access Flags   : [0x1D], sensitive, alwaysSensitive, neverExtract, local
        Algo_refs      : 0
        FieldLength    : 256
        Key ref        : 0 (0x00)
        Native         : yes
        Auth ID        : 01
        ID             : 0000000000000000000000000000000000000000
        MD:guid        : a3c05bfd-1d3c-7a25-75f7-618480b9d10e

Public EC Key [ESPICOHSMTR]
        Object Flags   : [0x00]
        Usage          : [0x140], verify, derive
        Access Flags   : [0x02], extract
        FieldLength    : 256
        Key ref        : 0 (0x00)
        Native         : no
        ID             : 0000000000000000000000000000000000000000
        DirectValue    : <present>
```

use opensc 0.26.1 no device cert.

in addition whether it is 0.25.1 or 0.26.1, the device certificate CA issuer identified in schs after initialization is ESPICOHSMCA00002, inconsistent with the patch, resulting in the inability to use schs

```
Device Certificate    : CVC id-AT Terminal CAR=ESPICOHSMDV00002 CHR=ESPICOHSMTRHBTVQ CED= 13, 2025 CXD= 13, 2026 
Device Issuer CA      : CVC id-AT DV (official domestic) CAR=ESPICOHSMCA00002 CHR=ESPICOHSMDV00002 CED= 6, 2025 CXD= 6, 2026 
Unknown or disabled root CA ESPICOHSMCA00002

````

polhenarejos, 2025-05-18:
I cannot reproduce it. I tested 0.26.1 and it works, at least in macOS.

For the last issue, try adding `ESPICOHSMCA00002` to `SmartCardHSM.rootCerts`:

```
ESPICOHSMCA00002: new CVC(new ByteString("7F218201BA7F4E8201725F290100421045535049434F48534D434130303030327F4982011D060A04007F000702020202038120FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFEFFFFFC2F822000000000000000000000000000000000000000000000000000000000000000008320000000000000000000000000000000000000000000000000000000000000000784410479BE667EF9DCBBAC55A06295CE870B07029BFCDB2DCE28D959F2815B16F81798483ADA7726A3C4655DA4FBFC0E1108A8FD17B448A68554199C47D08FFB10D4B88520FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFEBAAEDCE6AF48A03BBFD25E8CD0364141864104FD4DDCD165DD64494F5F1DB76CDAE4E6BD10DA59BF6D48B684E9D50BCC4C2114D2A8BC7C9C1ACAA24C982B1383D4D02E0F32323F657E146E8A051DEB7F6CD40C8701015F201045535049434F48534D434130303030327F4C12060904007F0007030102025305C0000000045F25060205000400065F24060308000302065F3740472AA7A7945F6352F8A65388C2810AEFB657356FFD31CBB2FCC516BC4ECC953D0BACA935B22D73F49802DD9E9DE6439FD8AEA18BD71E1439ABF458C085BD70C8", HEX))
```

IsayIsee, 2025-05-19:
> I cannot reproduce it. I tested 0.26.1 and it works, at least in macOS.
>
> For the last issue, try adding `ESPICOHSMCA00002` to `SmartCardHSM.rootCerts`:
>
> ```
> ESPICOHSMCA00002: new CVC(new [...]

It seems that only the device certificate cannot be seen, and other functions are normal. After adding the ESPICOHSMDV00002 certificate, schs can recognize pico-hsm normally. thanks

closed 2025-05-22 by IsayIsee