polhenarejos/pico-hsm #94: DKEK Key Check Value Shows 0000000000000000 After Import, Fails to Update

yuyulazybones, 2025-05-22:
Problem Description
When using sc-hsm-tool to initialize and import a DKEK key, the DKEK key check value remains 0000000000000000 even after successfully importing the DKEK share.

Environment Information

OpenSC Version: 0.26.1, 0.25.1
Hardward Board Model: RPi Pico 2040, waveshare 2350 one, xiao 2350
Operating System: Windows10 LTSC 2021, Linux(Ubuntu 24.04.2 LTS)

Steps to Reproduce
1. Create a single share DKEK
`sc-hsm-tool --create-dkek-share dkek.pbe`

2. Initialize with DKEK:
`sc-hsm-tool.exe --initialize --so-pin 3537363231383830 --pin 648219 --dkek-shares 1`

3. Check status (shows pending import):
`sc-hsm-tool.exe`
Output:  
DKEK shares          : 1  
DKEK import pending, 1 share(s) still missing  

4. Import DKEK share:
`sc-hsm-tool.exe --import-dkek-share dkek.pbe`

5. Enter correct password, shows successful import  

6. Check status again:
`sc-hsm-tool.exe`
Output:  
DKEK shares          : 1  
DKEK key check value : 0000000000000000

IsayIsee, 2025-05-22:
is also 0000000000000000, but using scsh after login, the value is ok

polhenarejos, 2025-05-23:
This is intended. DKEK is unlocked only when the PIN succeeds.

closed 2025-05-23 by polhenarejos